mbentley/docker-omada-controller #281: [Bug]: Something causing Watchtower to update container daily

threeonparfive, 2023-02-10:
### Controller Version

v5.8.4

### Describe the Bug

I run Watchtower to automatically update my containers when new images are found.  I just started using this Omada docker image a few days ago and Watchtower has updated it every day since despite no real new version of the controller being released.  It appears as if the image on Docker hub keeps slightly changing daily.

### Expected Behavior

The image should only change when there is a new controller version or an image bug is found and fixed.

### Steps to Reproduce

1. Install Omada docker image
2. Install Watchtower and let it run
3. Observe Watchtower upgrading the Omada container daily

### How You're Launching the Container

```plain
docker run -d --name=omada-controller \
-e PUID=1026 \
-e PGID=100 \
-e TZ=America/New_York \
-e MANAGE_HTTP_PORT=8088 \
-e MANAGE_HTTPS_PORT=8043 \
-e PORTAL_HTTP_PORT=8088 \
-e PORTAL_HTTPS_PORT=8043 \
-e SHOW_SERVER_LOGS=true \
-e SHOW_MONGODB_LOGS=false \
-e SSL_CERT_NAME="tls.crt" \
-e SSL_KEY_NAME="tls.key" \
-v /volume1/docker/omada/data:/opt/tplink/EAPController/data \
-v /volume1/docker/omada/logs:/opt/tplink/EAPController/logs \
-v /volume1/docker/omada/work:/opt/tplink/EAPController/work \
--net host \
--restart always \
mbentley/omada-controller
```

### Container Logs

```plain
These are the logs from Watchtower:
time="2023-02-08T10:30:40Z" level=info msg="Found new mbentley/omada-controller:latest image (d026b614bd3a)"
time="2023-02-09T10:30:41Z" level=info msg="Found new mbentley/omada-controller:latest image (b4223adfc5ca)"
time="2023-02-10T10:30:21Z" level=info msg="Found new mbentley/omada-controller:latest image (3ecc05279bf9)"
```

mbentley, 2023-02-10:
I build all of the images that I publish through some Jenkins pipelines that I've created + using buildkit.  One of the reasons that I am doing so is so that I can have greater control over the layer caching features of buildkit.  In this process, every morning around 3am Eastern time, these jobs kick off that will build any base images, where necessary, while using the buildkit caches.  This means that if there are not any changes to the base images, they're cached.  The same thing goes for the actual images, like the Omada Controller images which are triggered to be built after the base images are built, or in what's typical, they're just cached and no changes actually are present.  The exception to this process is when there are vulnerabilities discovered in either the base image or in the image itself.  So if there is a vulnerability in the base image that can be patched, I force that image to be re-built which will then also cause the downstream image (like the Omada Controller image) to be built due to the cache being busted so that vulnerabilities are not shipped.

So there are a few triggers:
1. The base image changes on Docker Hub
2. My refactored base image changed or needs a vulnerability patched
3. My image for Omada Controller changed or needs a vulnerability patched

And any of these can also be triggered from the cache busting.

I will say that this many updates isn't typical.  Here are the build jobs from the last week:

```
02/03 - 02/05   23-25 rebuilt due to buildkit update; these images are the same
02/06 - 02/08   26-28 rebuilt due to buildkit rollback; these images are the same
02/08           29 - vuln
02/08           30 - vuln
02/09           31 - no vuln; base image changed
02/10           32 - no vuln; base image changed
```

In this case, last week I upgraded a buildkit version and had to roll back the changes due to an issue with manifests changing due to the new provenance feature in buildkit 0.11.x.  Since then, I've setup a sandbox to test buildkit changes before I roll them out to all of my builds as it can be disruptive for many people such as yourself running tools like watchtower.  Then there were vulnerabilities that were patched in the images and then also in the base images.  The two builds on the 8th were due to me re-triggering some builds on accident due to a scanning failure.

While all of the times that the container uptime was reset wasn't just because of image updates (host reboot for updates, Docker updates for itself), here is a trend from the past 90 days of controller uptime I've personally had:

[image]

So while you're seeing a lot of updates, more than you'll probably see for other images published, I will have images that have all of the fixable vulnerabilities in them patched on a daily basis.  The current scan results are actually [available online](https://home.mbentley.net/trivy/image_scan_results_human.txt) for all of the images I build.

threeonparfive, 2023-02-10:
Thank you for this great explanation!  I understand completely and do respect maintaining a quality product above all else.

closed 2023-02-10 by threeonparfive

mbentley, 2023-02-10:
No problem!  I should actually do a proper writeup of this sometime as it's come up with a few other images I maintain as well.  I am really hoping I can narrow down some of the issues I do see though where occasionally and randomly manifests are changing for no reason that I can explain.