mbentley/docker-omada-controller #288: [Feature]: Portal URL

IndiGP, 2023-03-03:
### What problem are you looking to solve?

Hello,

we are currently facing the issue, that its not possible to change the Portal URL for voucher reediming. Problem is that we are hosting our omada container behind a reverse proxy, where ip and port 8088 is not reachable.
I read that its being created, while starting up the container.

### Describe the solution that you have in mind

It would be nice, if there were an option to set the portal URL + port as environment variable. So evertime someone is trying to access our network will redirected to the portal, where the controller is located.

mbentley, 2023-03-03:
I will have to look and see if there is any sort of setting available in the properties file but I am pretty sure it's not exposed to be able to set easily.

IndiGP, 2023-03-03:
Thanks for your support! Will look forward for your answer.

mbentley, 2023-03-13:
Yeah, so looking at the default values of `omada.properties`, I don't see any setting to be able to set the portal URL.  I haven't done anything with that myself so I am not sure if there would be a way to specifically set that and reverse proxy it.  Have you checked out the forums to see if there is a way to have the controller behind a reverse proxy and to set the portal URL to something that will go through that?

IndiGP, 2023-03-13:
We are currently trying to reverse proxy itself but no success one mobile device until now.. Will search for a solution further and if I have news, I will post it here. 
I tried to listen on 8088 and proxying it directly to the normal path.. Maybe I will find a solution for that. It's same thing like updateing devices. On the first times it was working, but now its not possbile anymore, because its telling, it cannot reach ssl port. I think they changed protocoll maybe and URL will not be transmitted anymore. But other problem. We will look forward and thank you for your help!

closed 2023-03-13 by IndiGP