mbentley/docker-omada-controller #289: [Feature]: SNMP Monitoring

meverickluke, 2023-03-04:
### What problem are you looking to solve?

Hello, 

i like to monitor my Omada Controller with Checkmk and SNMPv2c, but i have not found a way to bind/open UDP161.

### Describe the solution that you have in mind

Can you please add a environment variable to enable snmp 161/UDP.

mbentley, 2023-03-13:
See this thread: https://community.tp-link.com/en/business/forum/topic/246334?replyId=545282

> that setting only enables SNMP on switches, routers and AP's not the OC200. If you do an snmpwalk on an EAP or switch IP you should get a response.

mbentley, 2023-03-13:
Let me know if you're trying to do something different but the controller itself doesn't expose SNMP.

closed 2023-03-13 by mbentley