mbentley/docker-omada-controller #290: [Bug]: MANAGE_HTTPS_PORT 443 and v5.8: can't connect to WebUI

dke, 2023-03-08:
### Controller Version

v5.8.4

### Describe the Bug

I want to run the controller with `MANAGE_HTTPS_PORT` on port 443.

I understand I should be able to do the trick via changing these three lines from the default `docker run` command

```
  --sysctl net.ipv4.ip_unprivileged_port_start=0 \
  -p 8043:8043 \
  -e MANAGE_HTTPS_PORT=8043 \
```

into this:

```
  --sysctl net.ipv4.ip_unprivileged_port_start=0 \
  -p 443:443 \
  -e MANAGE_HTTPS_PORT=443 \
```

So, for starters, running stuff on the default 8043 port works.

Then, switching to using port 443 (`docker stop`, `docker rm`, `docker run` with correspondingly adjusted command, see below) however produces a setup which doesn't work, i.e. I can't connect to `172.17.0.12:443`. The browser says "connection failed", `curl` says 

```
root@vanilla:~# curl https://172.17.0.12:443
curl: (7) Failed to connect to 172.17.0.12 port 443 after 0 ms: Connection refused
```

This is what seems me to be the relevant lines of `netstat -pant` on the docker host:

```
root@vanilla:~# netstat -pant | grep docker
tcp        0      0 0.0.0.0:29814           0.0.0.0:*               LISTEN      11110/docker-proxy
tcp        0      0 0.0.0.0:29812           0.0.0.0:*               LISTEN      11152/docker-proxy
tcp        0      0 0.0.0.0:29813           0.0.0.0:*               LISTEN      11131/docker-proxy
tcp        0      0 0.0.0.0:29811           0.0.0.0:*               LISTEN      11173/docker-proxy
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      11284/docker-proxy
tcp        0      0 0.0.0.0:8843            0.0.0.0:*               LISTEN      11238/docker-proxy
tcp        0      0 0.0.0.0:8088            0.0.0.0:*               LISTEN      11261/docker-proxy
tcp6       0      0 :::29814                :::*                    LISTEN      11116/docker-proxy
tcp6       0      0 :::29812                :::*                    LISTEN      11158/docker-proxy
tcp6       0      0 :::29813                :::*                    LISTEN      11138/docker-proxy
tcp6       0      0 :::29811                :::*                    LISTEN      11180/docker-proxy
tcp6       0      0 :::443                  :::*                    LISTEN      11292/docker-proxy
tcp6       0      0 :::8843                 :::*                    LISTEN      11246/docker-proxy
tcp6       0      0 :::8088                 :::*                    LISTEN      11269/docker-proxy
```

The container logs say:

```
root@vanilla:~# docker logs -f omada-controller-v5
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'Europe/Berlin'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Setting 'manage.https.port' to 443 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
[...]
```

And looking into the container I find:

```
root@98f017b76e5c:/opt/tplink/EAPController/lib# cat ../properties/omada.properties
[...]
#web config
# After Omada Controller is initialized, the configuration of the following ports will be overwritten.
# If you want to change their configurations, please configure them in the controller software.
manage.http.port=8088
manage.https.port=443
portal.http.port=8088
portal.https.port=8843
[...]
```

But looking into `netstat -pant` in the running container:
```
root@vanilla:~# docker exec -it omada-controller-v5 bash
root@0aeb5253941d:/opt/tplink/EAPController/lib# netstat -pant
root@98f017b76e5c:/opt/tplink/EAPController/lib# netstat -pant
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:8043            0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:8088            0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:8843            0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29814           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29812           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29813           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:29811           0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.1:27217         0.0.0.0:*               LISTEN      -
tcp        0      0 172.19.0.2:29811        172.17.2.38:33855       ESTABLISHED -
tcp        0      0 172.19.0.2:29811        172.17.2.19:40231       ESTABLISHED -
tcp        0      0 127.0.0.1:55250         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 172.19.0.2:29811        172.17.2.17:38364       ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:46740         ESTABLISHED -
tcp        0      0 127.0.0.1:40588         127.0.0.1:27217         ESTABLISHED -
tcp       32      0 172.19.0.2:39902        52.31.140.226:443       CLOSE_WAIT  -
tcp        0      0 127.0.0.1:46740         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:40588         ESTABLISHED -
tcp        0      0 127.0.0.1:40602         127.0.0.1:27217         ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:54794         ESTABLISHED -
tcp        0      0 127.0.0.1:54794         127.0.0.1:27217         ESTABLISHED -
tcp       32      0 172.19.0.2:39916        52.31.140.226:443       CLOSE_WAIT  -
tcp        0      0 172.19.0.2:29811        172.17.2.18:53029       ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:55250         ESTABLISHED -
tcp        0      0 127.0.0.1:27217         127.0.0.1:40602         ESTABLISHED -
root@98f017b76e5c:/opt/tplink/EAPController/lib#
```

So my conclusion would be: the port is rendered correctly into `omada.properties`, but the software doesn't honor it and runs on 8043 anyways.

The entire `docker run` command is given below.

### Expected Behavior

I can configure the container to run with the management port on 443 and can connect to the web UI.

### Steps to Reproduce

See description.

### How You're Launching the Container

```plain
docker run -d \
  --name omada-controller-v5 \
  --restart unless-stopped \
  --sysctl net.ipv4.ip_unprivileged_port_start=0 \
  -p 8088:8088 \
  -p 443:443 \
  -p 8843:8843 \
  -p 27001:27001/udp \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -p 29814:29814 \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=443 \
  -e PGID="508" \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e PORT_ADOPT_V1=29812 \
  -e PORT_APP_DISCOVERY=27001 \
  -e PORT_DISCOVERY=29810 \
  -e PORT_MANAGER_V1=29811 \
  -e PORT_MANAGER_V2=29814 \
  -e PORT_UPGRADE_V1=29813 \
  -e PUID="508" \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Europe/Berlin \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:5.8
```

### Container Logs

```plain
Will be attached.
```

### Additional Context

See description.

dke, 2023-03-08:
[logs.txt](https://github.com/mbentley/docker-omada-controller/files/10920430/logs.txt)

mbentley, 2023-03-08:
This is expected. Setting the port via env car only works on first run otherwise the app itself reads the port from the database.  You need to change the port through the user interface after first run.

dke, 2023-03-08:
Thanks for the fast response. Seems I missed that in the docs.

closed 2023-03-08 by dke