mbentley/docker-omada-controller #291: [Bug]: Error on start up

NickKatchur, 2023-03-18:
### Controller Version

v5.9

### Describe the Bug

[image]

The container will not start due to failure after updating the controller firmware

### Expected Behavior

Controller to start

### Steps to Reproduce

Upgrade from version 5.5 to 5.9

### How You're Launching the Container

```plain
Have tried both the docker compose file below and downloading directly using QNAP container station built in fuctionality

version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.9
    restart: unless-stopped
    stop_grace_period: 60s
    ports:
      - 8843:8843
      - 8088:8088
      - 8043:8043
      - 29814:29814
      - 29813:29813
      - 29812:29812
      - 29811:29811
      - 29810:29810
      - 27002:27002
      - 27001:27001
    environment:
      - TZ=America/Kentucky/Louisville
    volumes:
      - /share/Container/Omada/data:/opt/tplink/EAPController/data
      - /share/Container/Omada/logs:/opt/tplink/EAPController/logs

volumes:
  omada-data:
  omada-logs:
```

### Container Logs

```plain
03-17-2023 00:00:07.284 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
03-17-2023 00:00:07.291 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
03-17-2023 00:00:07.298 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
03-17-2023 00:00:07.603 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
03-17-2023 00:00:07.316 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
03-17-2023 00:00:07.334 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
03-17-2023 00:00:19.713 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 121
03-17-2023 00:00:19.720 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
03-17-2023 00:00:19.721 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
03-17-2023 00:00:21.371 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
03-17-2023 00:00:21.371 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
03-17-2023 00:00:21.828 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
03-17-2023 00:00:21.829 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start
03-17-2023 00:00:24.798 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.6.3 using Java 17.0.6 on ea62f5dcd059 with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.6.3.jar started by omada in /opt/tplink/EAPController/lib)
[... 37 lines cut ...]
03-17-2023 00:02:14.108 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start
03-17-2023 00:02:16.648 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.6.3 using Java 17.0.6 on ea62f5dcd059 with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.6.3.jar started by omada in /opt/tplink/EAPController/lib)
03-17-2023 00:02:16.662 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
03-17-2023 00:02:43.677 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
03-17-2023 00:02:53.410 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
03-17-2023 00:03:13.916 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(): log mq consume task is start...
03-17-2023 00:03:14.090 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Started OmadaLinuxMain in 59.429 seconds (JVM running for 78.325)
03-17-2023 00:03:16.230 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask finished
03-17-2023 00:03:16.231 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before OmadacInitTask
03-17-2023 00:03:16.295 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before init bean
03-17-2023 00:03:20.310 INFO [main] [] c.t.s.e.s.c.c(): start schedule remove expire device... period = 10
03-17-2023 00:03:20.344 INFO [main] [] c.t.s.e.s.c.c(): update rateLimiterCache, permitsPerSecond = 5.0
03-17-2023 00:03:21.004 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): manager workGroup core thread  num is 16, max thread num is 16
03-17-2023 00:03:21.194 INFO [main] [] c.t.s.o.m.d.d.m.m.c.DeviceMsgConfig(): setMsgThreadPool thread coreSize is 4, maxSize is 4,queue size is 4500
03-17-2023 00:03:26.324 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): upgradeSendReq workGroup core thread num is 1, max thread num is 1
03-17-2023 00:03:29.945 INFO [main] [] c.t.s.o.m.c.d.m.s.w.s.c.WirelessGroupConfig(): ssidWorkerGroup thread size is 4, queue size is 1500
03-17-2023 00:03:32.116 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: after init bean
03-17-2023 00:03:32.567 ERROR [main] [] c.t.s.o.s.d.a(): Controller install failed. Failed to start.
03-17-2023 00:03:32.567 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Data abnormal due to install failed. Failed to start.
03-17-2023 00:03:32.567 INFO [main] [] c.t.s.o.s.d.a(): record: after OmadacInitTask
03-17-2023 00:03:32.567 ERROR [main] [] c.t.s.o.s.t.FailExitTask(): Failed to start omada controller, going to exit
03-17-2023 00:03:32.593 INFO [Thread-0] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 118
03-17-2023 00:03:32.903 INFO [SpringApplicationShutdownHook] [] c.t.s.e.p.b.a(): going to shutdown proxy executors
03-17-2023 00:03:32.903 INFO [SpringApplicationShutdownHook] [] c.t.s.e.p.a(): going to shutdown proxy thread pools.
```

mbentley, 2023-03-18:
According to your logs, you're running `v5.6.3`:
> 03-17-2023 00:00:24.798 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.6.3 using Java 17.0.6 on ea62f5dcd059 with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.6.3.jar started by omada in /opt/tplink/EAPController/lib)

I previously saw this error with another user [here](https://github.com/mbentley/docker-omada-controller/issues/228) but it was where they had taken an older version, tried to run a newer version but then tried to run an older version again which seems to cause a problem either in the persistent data or data stored in MongoDB.  Surprising that it would let you do that though as the [entrypoint should prevent you from trying to start an older version](https://github.com/mbentley/docker-omada-controller/pull/229).

Is it possible that you may have somehow tried to run an older version?

NickKatchur, 2023-03-19:
This must be it unfortunately. I tried updating, but then also tried reverting back to my previous container. Deleted the database folder and things are back up and running with the updated version. Just have to get devices adopted again. Thanks for the help

closed 2023-03-19 by NickKatchur

mbentley, 2023-03-19:
Just curious if you know what versions of the controller you tried. I'd like to make sure that the version check validation as a part of the startup is working as expected to prevent these scenarios in the future.