mbentley/docker-omada-controller #294: Run as a non-root user

MysticalMount, 2023-03-23:
### What problem are you looking to solve?

I know its designed to do that itself, but the initial group/user configuration seems to require root.  I suppose that's the trade off for making the uid:gid customiseable by environment variables.

Ill run it as root for now like a lot of other things but feel this is the type of container that shouldnt truly need it.

### Describe the solution that you have in mind

The initial config currently in the entrypoint.sh could be built into the container itself and then the container could start as 508, allowing us to run as non-root and have a successful start.

mbentley, 2023-03-23:
Like you said, it's not really running as root except for in the entrypoint.  It's a trade off, for sure but I think it's worth the flexibility.

MysticalMount, 2023-03-27:
Agreed - thansk for reply @mbentley - I only run a local airgapped home lab so not a biggie for me.

mbentley, 2023-03-29:
Closing unless there is a better option available.

closed 2023-03-29 by mbentley