mbentley/docker-omada-controller #296: [Bug]: 3.2.17 image stopped working due to JRE fatal error

lkoning, 2023-03-26:
### Controller Version

v3.2.17

### Describe the Bug

Hi, 
Somehow the image for 3.2.17 stopped working. It does nog boot because of a:
```
omada-controller  | # A fatal error has been detected by the Java Runtime Environment:
omada-controller  | #
omada-controller  | #  SIGSEGV (0xb) at pc=0x00007f6a918a88e0, pid=1, tid=0x00007f6a924b4700
```

And lots more that follows. I have this behaviour on 2 computers. I am not able to upgrade to a newer version due to that I am having 3 EAP225_V1's. Since I have these I am using this image and it never failed. 

I don't think my Server or Docker-compose setup is anything special. It just runs a very small Arch install and a number of Dockers.

### Expected Behavior

A booted controller. 

### Steps to Reproduce

1. Just try to run it.

### How You're Launching the Container

```plain
-compose.yaml
version: "3.9"
services:
  controller:
    image: mbentley/omada-controller:3.2-amd64
    container_name: omada-controller
    environment:
      - TZ=${TZ}
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - SMALL_FILES=false
    ports:
      - "8088:8088/tcp"
      - "8043:8043/tcp"
      - "27001:27001/udp"
      - "27002:27002/tcp"
      - "22443:22443"
      - "29810:29810/udp"
      - "29811-29813:29811-29813/tcp"
    volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-work:/opt/tplink/EAPController/work
      - omada-logs:/opt/tplink/EAPController/logs
    restart: on-failure:1
    networks:
      - wifi

volumes:
  omada-data:
  omada-work:
  omada-logs:

networks:
  wifi:
    external: true
```

### Container Logs

```plain
later
```

### Additional Context

Unfortunately I do not know when exactly this behaviour started. I mean, the server is up 24/7 and every now and then I run updates. I have no gui to immediately check if all Dockers are up.

mbentley, 2023-03-27:
Thanks for the report. I will see what I am able to - I am not at home but will try to look tonight as long as my cell signal works.

mbentley, 2023-03-27:
Huh, with a fresh container, it's working fine for me:

```
$ docker run -it --rm \
  --name omada-controller \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 8843:8843 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -p 29814:29814 \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e SMALL_FILES=false \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:3.2-amd64
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'Etc/UTC'
WARN: ownership not set correctly on '/opt/tplink/EAPController/data'; setting correct ownership (omada:omada)
WARN: ownership not set correctly on '/opt/tplink/EAPController/logs'; setting correct ownership (omada:omada)
WARN: ownership not set correctly on '/opt/tplink/EAPController/work'; setting correct ownership (omada:omada)
INFO: Starting Omada Controller as user omada
SLF4J: com.tp_link.eap.util.m.a
SLF4J: The following set of substitute loggers may have been accessed
SLF4J: during the initialization phase. Logging calls during this
SLF4J: phase were not honored. However, subsequent logging calls to these
SLF4J: loggers will work as normally expected.
SLF4J: See also http://www.slf4j.org/codes.html#substituteLogger
SLF4J: com.tp_link.eap.start.b.a.a
SLF4J: com.tp_link.eap.start.a
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.configuration.Configuration
SLF4J: com.tp_link.eap.configuration.Configuration
SLF4J: com.tp_link.eap.configuration.Configuration
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.configuration.b
SLF4J: com.tp_link.eap.start.a
SLF4J: com.tp_link.eap.i.b
SLF4J: com.tp_link.eap.i.b
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.service.impl.EapInitServiceImpl
SLF4J: com.tp_link.eap.start.a
SLF4J: com.tp_link.eap.start.c.c
SLF4J: com.tp_link.eap.start.c.c
SLF4J: com.tp_link.eap.start.c.a
SLF4J: com.tp_link.eap.start.c.c
SLF4J: com.tp_link.eap.util.m.a
SLF4J: com.tp_link.eap.util.m.a
SLF4J: com.tp_link.eap.start.c.c
SLF4J: com.tp_link.eap.start.c.c
startup...
Mar 27, 2023 12:44:52 AM org.hibernate.validator.internal.util.Version <clinit>
INFO: HV000001: Hibernate Validator 4.3.1.Final
Omada Controller started
```

mbentley, 2023-03-27:
What host operating system(s) are you using?  Kernel versions?  I will probably have to do some searching as to why it might not be working since I can't quickly reproduce it.  My network connectivity is really limited here so it might be a few days before I can research more.

mbentley, 2023-03-27:
Hmm, I think this is something that I’ll have to focus on as I would say it’s the most likely cause of the issue:

```
library initialization failed - unable to allocate file descriptor table - out of memory#
```

mbentley, 2023-03-29:
Did some searching on that error, first hit took me to https://superuser.com/questions/1413352/running-jdk-8-in-docker-suddenly-broken-on-arch-linux-with-unable-to-allocate-f/1413390#1413390.  This says it's a ulimit error causing the issue.  Check into that and see if upping the ulimit at the daemon level.

mbentley, 2023-03-30:
It sets the limit for the resources that an individual user can use on a system. In a number of cases, it will be something like running a database that opens a large number of files that can eat up the ulimit's limit.

What you may be seeing is that because processes are started as a child of Docker, it considers that parent process as a single consumer of that limit so as you add more processes running in Docker, it increases the utilization of resources.  I've set my ulimit higher at the OS level in general because I run a lot of processes on a single, very powerful server.

There is nothing that can be done at the container level because the limit is at the OS/kernel level.  I won't be able to have a version that runs a different version of Java as I am already building 45 images just for the different versions of Omada Controller software, and that is only providing versions for the supported ones.

lkoning, 2023-03-30:
Ok, understood. My server also runs some 20/30 containers which togehter consume like 4GB of ram where 16GB available, the headless Arch takes about close to a few 400MB. So there I can, in a way understand the issue. My workstation however has just as much ram...but runs about nothing in the backgroud...Afaik Flatpak does not use Dockerd as a sandbox. So there, I kind of not understand how this issue can popup.

Anyway, I can probably 'wait' this issue away now by following the suggested settings change. Think I could try to generate a Docker-image by using your repo. Maybe someday :)

Should I close this issue then?

mbentley, 2023-03-30:
If Flatpak and Docker are both running processes as the same user, it could be hitting the default ulimits which would explain why it would add up.

If you do want to try out another OpenJDK, all of the configuration for what package gets installed can be found [here](https://github.com/mbentley/docker-omada-controller/blob/master/install.sh#L58-L89) but if anything, it would just be a bandaid as I would almost guarantee you would eventually hit the same issue no matter what OpenJDK version you're using, at least as some point.