mbentley/docker-omada-controller #297: [Bug]: WARN [manage-work-group-1] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null

jonas740, 2023-03-30:
### Controller Version

V5.9- Chromium

### Describe the Bug

WARN [manage-work-group-1] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value nullaa

keeps showing in the log. 

### Expected Behavior

 no warns

### Steps to Reproduce

i just pulled the latest image and rebooted. 

### How You're Launching the Container

```plain
Docker compose
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (1000:1000)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'Europe/Stockholm'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8043 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
INFO: Starting Omada Controller as user omada
03-30-2023 23:55:03.428 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
03-30-2023 23:55:03.430 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
03-30-2023 23:55:03.433 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
03-30-2023 23:55:03.435 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
03-30-2023 23:55:03.437 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
03-30-2023 23:55:03.437 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
03-30-2023 23:55:03.450 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
03-30-2023 23:55:04.960 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 117
03-30-2023 23:55:04.961 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
03-30-2023 23:55:04.962 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
03-30-2023 23:55:05.258 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
03-30-2023 23:55:05.258 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
03-30-2023 23:55:05.300 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
03-30-2023 23:55:05.300 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start
  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)
03-30-2023 23:55:05.654 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.9.31 using Java 17.0.6 on beelink with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.9.31.jar started by omada in /opt/tplink/EAPController/lib)
03-30-2023 23:55:05.656 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
03-30-2023 23:55:09.858 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
03-30-2023 23:55:11.241 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
03-30-2023 23:55:13.456 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(): log mq consume task is start...
03-30-2023 23:55:13.458 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Started OmadaLinuxMain in 8.099 seconds (JVM running for 10.52)
[... 101 lines cut ...]
03-30-2023 23:55:57.725 INFO [server-comm-pool-11] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-A9-D0-62)
03-30-2023 23:55:57.737 INFO [manage-work-group-1] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-A9-D0-62)
03-30-2023 23:55:57.741 WARN [manage-work-group-1] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
03-30-2023 23:55:57.799 INFO [manage-work-group-2] [] c.t.s.o.m.d.d.m.d.a.T(): syncConfigurationForSameVersion to OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-A9-D0-62), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
03-30-2023 23:56:00.045 INFO [server-comm-pool-17] [] c.t.s.e.s.a.b.e(): serverType from channel: /192.168.1.59:29814 -> /192.168.1.1:37194 is MANAGE
03-30-2023 23:56:04.714 WARN [manage-work-group-4] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
03-30-2023 23:56:05.840 INFO [adopt-work-group-1] [] c.t.s.o.m.d.d.m.d.c.c(): Gateway OmadacId(3faf38727bdff9c9c6ee8e375368273d) SiteId(641f4045d699177d84b56617) DeviceMac(34-60-F9-CD-E5-CD) adopt[auto=true] ok
03-30-2023 23:56:05.957 INFO [adopt-work-group-1] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-CD-E5-CD)
03-30-2023 23:56:09.051 INFO [manage-work-group-7] [] c.t.s.o.m.d.p.t.c(): Device 34-60-F9-CD-E5-CD OmadacId(3faf38727bdff9c9c6ee8e375368273d) changed to status CONNECTED, which don't need to handle.
03-30-2023 23:56:10.960 INFO [server-comm-pool-0] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-CD-E5-CD)
03-30-2023 23:56:10.968 INFO [manage-work-group-8] [] c.t.s.o.m.d.d.m.i.e(): first inform no need send full config to OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-CD-E5-CD)
03-30-2023 23:56:10.969 WARN [manage-work-group-8] [] c.t.s.o.m.d.d.m.m.s.e(): fill set msg body but all setting null, skip send set msg to OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-CD-E5-CD)
03-30-2023 23:56:10.969 WARN [manage-work-group-8] [] c.t.s.o.m.d.d.m.m.s.e(): failed to genConfigBodyAndAddVersion with keys:[]
03-30-2023 23:56:10.970 INFO [manage-work-group-8] [] c.t.s.o.m.d.d.m.d.c.A(): syncConfigurationForSameVersion to OmadacId(3faf38727bdff9c9c6ee8e375368273d) DeviceMac(34-60-F9-CD-E5-CD), result:SendDeviceMsgResult(success=false, deviceResponse=null)
03-30-2023 23:56:18.204 INFO [monitor-topology-pool-0] [] c.t.s.o.c.u.d.a(): list local interface macs: [06-41-F6-E8-12-26, FE-F5-1B-4C-1A-CD, 66-8D-C6-33-02-19, 6A-4C-CD-86-AD-AE, 86-F0-48-60-D0-8C, 72-2D-FF-A2-E1-30, DA-F0-55-E0-FF-66, 02-42-CC-3D-DE-87, 02-42-D2-5D-05-81, B0-41-6F-09-42-84]
03-30-2023 23:56:24.850 WARN [manage-work-group-11] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
```

### Additional Context

Last entry in log there just keeps going on and on. I tried to Google it but no answers.

03-30-2023 23:56:24.850 WARN [manage-work-group-11] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.

In portainer the container is listed as healthy and not running like all the others strange enough

mbentley, 2023-04-01:
I'm not really sure what the error means.  Is the controller web interface available when it starts spitting out these messages?  If so, can you login?  It doesn't look like the controller process is dying after the message so it's hard to tell what's going on.  To get someone to find out what that error message might mean, you might need to post on the [TP-Link community forums](https://community.tp-link.com/en/business/forum/582) or contact their support as I don't have visibility into the inner workings of the app itself.

jonas740, 2023-04-01:
The controller seem to work just fine. It is just constantly spitting out
this warning message in the container logs.

/jonas

mbentley, 2023-04-01:
OK, I would definitely post on the forums then as I can't see anything that indicates what that is without being able to see any code.

closed 2023-04-01 by mbentley

garlik82, 2023-04-16:
> The controller seem to work just fine. It is just constantly spitting out this warning message in the container logs. /jonas Den lör 1 apr. 2023 15:30Matt Bentley ***@***.***> skrev:
> […](#)

Having the same issue. Managed to fix this?

jonas740, 2023-06-22:
Nope, still having the same issue. It just will not go away. And i update the docker quite often.

Something i am having issues with lately is setting a fixed IP on an adress where i have had an old device set to. I cleared the DHCP records but still, nope it wont work. this bug was noticed a long time ago and it still hasnt been resolved.

mbentley, 2023-06-23:
Have either of you created a post on the TP-Link forums yet?  The only thing I can think of would be some sort of setting being changed related to authentication somewhere and somehow, an unexpected field is empty.  I haven't read this thread but it looks like someone posted about what looks like a similar issue: https://community.tp-link.com/en/business/forum/topic/605566

jonas740, 2023-06-23:
Ye, I have searched a lot about this and also the issue I'm having with
setting static ip's and it seems like there is a "bug" in their software.
Last post I read was from March 2023.

I will contact TP-Link. I meant to do that a while ago but my dad became
really sick so this was kind of distant for me.

/Jonas

garlik82, 2023-06-23:
I made a backup of my settings, spun up a fresh docker container, restored my backup and had no issues ever since.

jonas740, 2023-06-23:
Could it be that when updates come that it interferes somehow.

Just thinking that it could be like with graphics drivers, if you don't
remove the old ones completely it can cause issues. It does not have to
cause issues but it can.

You using the chromium version?

/Jonas

mbentley, 2023-06-23:
It could be a number of things like a database migration that doesn't go well (or they had a bug in), if you have a process that auto-updates the controller, it could be failing to shut down cleanly (maybe not so likely), but I doubt anything happening at the OS level would be an issue at least.

RaboonTheBaboon, 2023-08-08:
Just to mention, I saw this on the Unraid build of your container and solved it.

The container template on Unraid has a load of unorthdox ports, 18043 rather than 8043 and so on. 
I set the container up with those 10843 ports, then reset them to the usual 8043 etc
As soon as I did that, the warnings started. 
As soon as I reset the ports back to 18043, as the container was originally created, hey stopped.

So my inking is that this issue is tied into the interesting port mapping not sticking issue.

jonas740, 2023-08-30:
```
08-30-2023 22:52:01.731 WARN [manage-work-group-5] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:52:31.911 WARN [manage-work-group-12] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:53:02.112 WARN [manage-work-group-15] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:53:32.307 WARN [manage-work-group-5] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:54:02.495 WARN [manage-work-group-9] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:54:32.685 WARN [manage-work-group-14] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:55:02.888 WARN [manage-work-group-3] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:55:33.089 WARN [manage-work-group-7] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:56:03.347 WARN [manage-work-group-13] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:56:33.542 WARN [manage-work-group-2] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:57:03.741 WARN [manage-work-group-7] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:57:33.930 WARN [manage-work-group-11] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:58:04.116 WARN [manage-work-group-0] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:58:34.311 WARN [manage-work-group-4] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
08-30-2023 22:59:04.507 WARN [manage-work-group-10] [] c.t.s.o.m.c.d.m.s.a.a.a(): AuthModeEnum resolve failed, invalid value null.
```

I am not sure if this is causing something still but theese "Warn" messages still appear and i just read this thread aswell that might explain something. I am not sure what this actually is so i can not really say that much. But as seen it is not only limited to one managed work group.

Speaking of ports, should ports 8043 and 8088 be open and exposed ?

Below is my docker-compose

```
version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:latest-chromium
     restart: unless-stopped
     stop_grace_period: 60s
     network_mode: host
     environment:
      - PUID=1000
      - PGID=1000
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
      - TZ=Europe/Stockholm
     volumes:
      - /home/dockerdata/omada/omada-data:/opt/tplink/EAPController/data
      - /home/dockerdata/omada/omada-work:/opt/tplink/EAPController/work
      - /home/dockerdata/omada/omada-logs:/opt/tplink/EAPController/logs

volumes:
  omada-data:
  omada-work:
  omada-logs:
```

But then again this might not be anything that limits anything its more about why are theese "WARN" messages comming and from what.

@RaboonTheBaboon might be on to something in his comment and i can´t say that i understand it so you might understand it better @mbentley

mbentley, 2023-08-30:
The ports don't need to be exposed because network mode is using the host's network namespace like a non-containerized process.  Will have to look at the logs later.