mbentley/docker-omada-controller #298: [Bug]: too many open files

osc86, 2023-03-31:
### Controller Version

v5.9.31

### Describe the Bug

Every time I start the container, I get the following error and the container changes status to unhealthy.
ERROR [main] [] c.t.s.o.s.t.DeviceConnectorSeverTask(): Failed to start up net server because of 
java.lang.IllegalStateException: failed to create a child event loop
Caused by: io.netty.channel.ChannelException: timerfd_create() failed: Too many open files

The controller is working, the website is working, I can adopt devices but I can't download firmware updates and this error message is filling up my logs.

### Expected Behavior

I don't know if it's an issue with the container image, the omada software or the docker configuration, but it doesn't look ok.
I'm using standard settings like with every other container, and never had this problem before.

### Steps to Reproduce

1. Download the image (in my case I'm using portainer running on Fedora CoreOS)
2. Add volumes for persistent storage ( /opt/tplink/EAPController/{data,logs} )
3. I'm not using any ENV parameters, Labels, didn't change anything in Runtime & Resource, Capabilities
4. Start the container 

### How You're Launching the Container

```plain
"Args": [
            "/usr/bin/java",
            "-server",
            "-Xms128m",
            "-Xmx1024m",
            "-XX:MaxHeapFreeRatio=60",
            "-XX:MinHeapFreeRatio=30",
            "-XX:+HeapDumpOnOutOfMemoryError",
            "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof",
            "-Djava.awt.headless=true",
            "-cp",
            "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:",
            "com.tplink.smb.omada.starter.OmadaLinuxMain"
        ],
```

### Container Logs

```plain
/\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)
03-31-2023 14:40:43.297 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.9.31 using Java 17.0.6 on omada with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.9.31.jar started by omada in /opt/tplink/EAPController/lib)
03-31-2023 14:40:43.301 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
03-31-2023 14:40:50.032 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
03-31-2023 14:40:52.444 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
03-31-2023 14:40:55.567 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(): log mq consume task is start...
03-31-2023 14:40:55.569 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Started OmadaLinuxMain in 12.801 seconds (JVM running for 15.696)
03-31-2023 14:40:55.661 INFO [main] [] c.t.s.o.c.p.s.e(): Handling event: org.springframework.boot.context.event.ApplicationStartedEvent[source=org.springframework.boot.SpringApplication@58b0ef59]
03-31-2023 14:40:55.673 INFO [main] [] c.t.s.o.m.d.p.b.a(): manager maintenance Handling event: org.springframework.boot.context.event.ApplicationStartedEvent[source=org.springframework.boot.SpringApplication@58b0ef59]
03-31-2023 14:40:55.694 INFO [main] [] c.t.s.o.l.p.c.c.b.b(): Handling event: org.springframework.boot.context.event.ApplicationStartedEvent[source=org.springframework.boot.SpringApplication@58b0ef59]
03-31-2023 14:40:55.737 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask finished
03-31-2023 14:40:55.737 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before OmadacInitTask
03-31-2023 14:40:55.755 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before init bean
03-31-2023 14:40:56.463 INFO [main] [] c.t.s.e.s.c.c(): start schedule remove expire device... period = 10
03-31-2023 14:40:56.470 INFO [main] [] c.t.s.e.s.c.c(): update rateLimiterCache, permitsPerSecond = 5.0
03-31-2023 14:40:56.594 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): manager workGroup core thread  num is 16, max thread num is 16
03-31-2023 14:40:56.646 INFO [main] [] c.t.s.o.m.d.d.m.m.c.DeviceMsgConfig(): setMsgThreadPool thread coreSize is 16, maxSize is 16,queue size is 4500
03-31-2023 14:40:57.995 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): upgradeSendReq workGroup core thread num is 1, max thread num is 1
[... 124 lines cut ...]
	at com.tplink.smb.ecsp.transport.netty.NettyTransporter.bind(NettyTransporter.java:48) ~[ecsp-transporter-netty-1.1.80.jar:1.1.80]
	at com.tplink.smb.ecsp.transport.api.Transporters.bind(Transporters.java:54) ~[ecsp-transporter-api-1.1.80.jar:1.1.80]
	at com.tplink.smb.ecsp.server.j.a(SourceFile:60) ~[ecsp-server-core-1.1.80.jar:1.1.80]
	at com.tplink.smb.ecsp.server.j.a(SourceFile:38) ~[ecsp-server-core-1.1.80.jar:1.1.80]
	at java.util.ArrayList.forEach(ArrayList.java:1511) ~[?:?]
	at com.tplink.smb.ecsp.server.j.a(SourceFile:37) ~[ecsp-server-core-1.1.80.jar:1.1.80]
	at com.tplink.smb.ecsp.server.EcspTransporterStarter.start(SourceFile:163) ~[ecsp-server-core-1.1.80.jar:1.1.80]
	at com.tplink.smb.omada.device.gateway.a.a(SourceFile:171) ~[device-gateway-port-local-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.task.DeviceConnectorSeverTask.execute(SourceFile:28) ~[local-starter-5.9.31.jar:5.9.31]
	... 4 more
03-31-2023 14:41:11.076 INFO [comm-pool-6] [] c.t.s.o.s.t.CloudStartUpTask(): Cloud service is forbidden.
```

### Additional Context

#docker inspect $container
```
            "Ulimits": [
                {
                    "Name": "nofile",
                    "Hard": 1024,
                    "Soft": 1024
                }
            ],
```

mbentley, 2023-03-31:
Yeah, so this will likely need a ulimit change.  Not sure exactly how you're running the container but updating the ulimit at runtime would work.  I actually didn't remember that I had set this on my own container that I'm running.  I'm using `--ulimit nofile=4096:8192` and that seems to work.  I'll need to figure out where all that needs updated in places like downstream container marketplaces for some of the NAS devices and whatnot that have the controller published but I will see if I can get that taken care of.

closed 2023-03-31 by mbentley

osc86, 2023-03-31:
Thank you for your quick reply and raising the ulimit. I think the docker default setting is still only 1024, which apparently isn't enough for today's applications.
I've increased the default-ulimit globally in /etc/sysconfig/docker and made the file immutable to avoid problems with other containers in the future.
Using the limits you provided, the container starts without errors. Thanks again!

benhedrington, 2023-04-01:
Updated ulimit on Unraid template that user may have been using to match @mbentley original suggestions in start up doc.

mbentley, 2023-04-01:
Thank you @benhedrington!

commit 5ea2e048 (2023-03-31): Added ulimits to run examples and compose; fixes #298
commit 7592abce (2023-03-31): Merge pull request #299 from mbentley/add-ulimits

Added ulimits to run examples and compose; fixes #298