mbentley/docker-omada-controller #300: [Bug]:

bobbywaz, 2023-04-07:
### Controller Version

v5.9.31

### Describe the Bug

Omada appears to startup with no errors, but when I access http://ip:8088 it redirects to https://10.168.3.59:8043/0f5f794c76c2c95f05b8cd5ac6a1e0e3/login, but then the page times out

### Expected Behavior

The page loads

### Steps to Reproduce

Run container
Access URL

### How You're Launching the Container

```plain
version: "3.1"
services:
  omada:
    container_name: omada
    image: mbentley/omada-controller:latest
    restart: unless-stopped
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=1000
      - PGID=1000
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - TZ=America/New_York
    volumes:
      - /mnt/docker/omada-data:/opt/tplink/EAPController/data
      - /mnt/docker/omada-logs:/opt/tplink/EAPController/logs
```

### Container Logs

```plain
2023-04-07T17:08:38.129791953Z INFO: Validating user/group (omada:omada) exists with correct UID/GID (1000:1000)
2023-04-07T17:08:38.140686149Z INFO: Group (omada) doesn't exist; creating
2023-04-07T17:08:38.364365668Z INFO: User (omada) doesn't exist; creating
2023-04-07T17:08:38.803546815Z INFO: Time zone set to 'America/New_York'
2023-04-07T17:08:38.816015684Z INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
2023-04-07T17:08:38.827277672Z INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
2023-04-07T17:08:38.839096897Z INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
2023-04-07T17:08:38.850375258Z INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
2023-04-07T17:08:38.862895479Z INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
2023-04-07T17:08:38.874916660Z INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
2023-04-07T17:08:38.886367827Z INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
2023-04-07T17:08:38.898743690Z INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
2023-04-07T17:08:38.910315534Z INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
2023-04-07T17:08:38.922149046Z INFO: Value of 'port.discovery' already set to 29810 in omada.properties
2023-04-07T17:08:38.954246437Z WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
2023-04-07T17:08:39.049472283Z INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
2023-04-07T17:08:39.050070442Z INFO: Starting Omada Controller as user omada
2023-04-07T17:08:41.324026972Z 04-07-2023 13:08:41.310 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
[... 98 lines cut ...]
2023-04-07T17:09:44.724187912Z 04-07-2023 13:09:44.723 WARN [https-jsse-nio-8043-exec-6] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
2023-04-07T17:09:45.895692258Z 04-07-2023 13:09:45.894 INFO [adopt-work-group-5] [] c.t.s.o.m.d.d.m.d.a.I(): Ap OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) SiteId(640edddf836d7765b01e9e67) DeviceMac(34-60-F9-B4-55-2E) adopt[auto=true] ok
2023-04-07T17:09:45.988493501Z 04-07-2023 13:09:45.987 INFO [adopt-work-group-5] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(34-60-F9-B4-55-2E)
2023-04-07T17:09:46.415312361Z 04-07-2023 13:09:46.414 INFO [server-comm-pool-5] [] c.t.s.e.s.a.b.e(): serverType from channel: /10.168.3.59:29814 -> /10.168.3.104:52350 is MANAGE
2023-04-07T17:09:46.626158669Z 04-07-2023 13:09:46.625 INFO [adopt-work-group-1] [] c.t.s.o.m.d.d.m.d.a.I(): Ap OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) SiteId(640edddf836d7765b01e9e67) DeviceMac(E8-48-B8-D5-EE-CA) adopt[auto=true] ok
2023-04-07T17:09:46.641346970Z 04-07-2023 13:09:46.640 INFO [adopt-work-group-1] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(E8-48-B8-D5-EE-CA)
2023-04-07T17:09:46.846460721Z 04-07-2023 13:09:46.845 INFO [manage-work-group-3] [] c.t.s.o.m.d.p.t.c(): Device E8-48-B8-D5-EE-CA OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) changed to status CONNECTED, which don't need to handle.
2023-04-07T17:09:47.038647454Z 04-07-2023 13:09:47.037 INFO [server-comm-pool-10] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(E8-48-B8-D5-EE-CA)
2023-04-07T17:09:47.083736687Z 04-07-2023 13:09:47.081 INFO [manage-work-group-4] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(E8-48-B8-D5-EE-CA)
2023-04-07T17:09:47.259970887Z 04-07-2023 13:09:47.258 INFO [manage-work-group-5] [] c.t.s.o.m.d.d.m.d.a.T(): syncConfigurationForSameVersion to OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(E8-48-B8-D5-EE-CA), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
2023-04-07T17:09:55.571465845Z 04-07-2023 13:09:55.570 INFO [manage-work-group-6] [] c.t.s.o.m.d.p.t.c(): Device 34-60-F9-B4-55-2E OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) changed to status CONNECTED, which don't need to handle.
2023-04-07T17:10:00.611056437Z 04-07-2023 13:10:00.609 INFO [server-comm-pool-14] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(34-60-F9-B4-55-2E)
2023-04-07T17:10:00.621344911Z 04-07-2023 13:10:00.620 INFO [manage-work-group-9] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(34-60-F9-B4-55-2E)
2023-04-07T17:10:00.677832337Z 04-07-2023 13:10:00.676 INFO [manage-work-group-10] [] c.t.s.o.m.d.d.m.d.a.T(): syncConfigurationForSameVersion to OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3) DeviceMac(34-60-F9-B4-55-2E), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
2023-04-07T17:10:32.043573688Z 04-07-2023 13:10:32.041 INFO [monitor-topology-pool-0] [] c.t.s.o.c.u.d.a(): list local interface macs: [A2-45-4C-B1-66-DD, 16-C2-93-05-93-50, AE-24-BF-2C-62-20, 7E-72-4F-BD-94-5F, A6-6F-76-2F-B0-A7, 36-DA-50-45-BD-65, FA-5C-91-4A-6E-49, 46-D0-4A-B6-69-0C, 8A-45-3D-86-D6-2F, 62-DA-9E-B8-43-3E, 5A-16-FB-29-C7-E4, 6E-82-4E-06-CF-F3, A2-65-A3-F7-E4-B5, 4A-C0-59-5F-FD-8C, 9A-D5-BE-7E-02-01, 96-F6-8F-5B-55-06, 7A-78-27-01-DA-A7, 92-F6-E2-99-18-7D, 76-79-34-7F-CE-5B, 46-9F-9F-FB-0B-92, 5A-21-67-43-7B-CB, 52-43-55-E7-7B-64, 6A-EB-D4-0B-2E-74, 46-34-0D-07-7C-5C, 7A-8F-D5-CF-2E-09, 0E-FD-FB-D8-1F-A6, 66-DA-33-43-C7-71, FE-77-37-69-07-53, 02-42-CC-7E-8E-88, 02-42-44-1C-C8-BC, 00-50-56-B1-A9-58]
2023-04-07T17:10:43.880060649Z 04-07-2023 13:10:43.879 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(0f5f794c76c2c95f05b8cd5ac6a1e0e3).
2023-04-07T17:10:44.002762831Z 04-07-2023 13:10:43.998 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
2023-04-07T17:10:44.002820162Z 04-07-2023 13:10:43.999 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
2023-04-07T17:10:44.468608061Z 04-07-2023 13:10:44.467 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
2023-04-07T17:14:31.892878977Z 04-07-2023 13:14:31.890 INFO [comm-pool-15] [] c.t.s.o.l.c.a.a(): Start pushing connected devices.
```

### Additional Context

Sorry if this isn't the place to come for support, I didn't see a Discord / Forum link anywhere

mbentley, 2023-04-07:
Yup, this is the right place to come for help.  I don't see anything from the logs that would specifically indicate any issues, looks like the controller is up and running and nothing error wise indicated from the logs.

Have you tried a different browser, incognito session, or a [hard refresh](https://fabricdigital.co.nz/blog/how-to-hard-refresh-your-browser-and-clear-cache) of the login page?  Another quick way to test would be to try to use a curl command, assuming you have linux terminal access somewhere: 

```
curl -kvSL "http://10.168.3.59:8088"
```

This should do a verbose curl to the http port, ignoring certificate errors, showing errors in the output and following redirects.  If it works, it should output the page's html after the verbose connection logs.

bobbywaz, 2023-04-10:
I have tried other browsers / other machines / phones, they all get to the redirect address and then fail. Here's the output of the curl command:

`b@MIMIKYU:/mnt/c/Users/b/Desktop$ curl -kvSL "http://10.168.3.59:8088"                                                                                                                                                                         
*   Trying 10.168.3.59:8088...                                                                                                                                                                                                                 
* TCP_NODELAY set                                                                                                                                                                                                                              
* Connected to 10.168.3.59 (10.168.3.59) port 8088 (#0)                                                                                                                                                                                        
> GET / HTTP/1.1                                                                                                                                                                                                                               
> Host: 10.168.3.59:8088                                                                                                                                                                                                                       
> User-Agent: curl/7.68.0                                                                                                                                                                                                                      
> Accept: */*                                                                                                                                                                                                                                  
>                                                                                                                                                                                                                                              
* Mark bundle as not supporting multiuse                                                                                                                                                                                                       
< HTTP/1.1 302                                                                                                                                                                                                                                 
< X-Frame-Options: SAMEORIGIN                                                                                                                                                                                                                  
< X-Content-Type-Options: nosniff                                                                                                                                                                                                              
< Content-Security-Policy: default-src 'self' https://*.tplinkcloud.com/;script-src 'self' 'unsafe-eval' 'sha256-7W9UiBaYGlOHpT1aQBLegqffUVHbYq6/ZAb+ErjUb40=' 'sha256-VGQ8jNTL2g0e8wPwOgyCQJDqhuRgfV7gRYexcBkBe4Y=' 'sha256-x2jgB1zBLi30IsfY+V
NgWjwBGeHPJxOSrzl+IdsT6k0=' 'sha256-0AHZXO4clnpdcxqdmASPBEp4JCIrtaxIX/mUuL1kzZw=' 'sha256-lfXlPY3+MCPOPb4mrw1Y961+745U3WlDQVcOXdchSQc=';style-src 'self' 'unsafe-inline';connect-src 'self' https://*.tplinkcloud.com/ https://*.tplinkcloud.co
m:8843/ wss://*.tplinkcloud.com/ https://*.tiles.mapbox.com https://api.mapbox.com https://events.mapbox.com ;frame-src 'self' data:;img-src 'self' https://*.tplinkcloud.com/ data: blob:;child-src blob: ;worker-src blob: ;object-src 'self'
 data: blob:                                                                                                                                                                                                                                   
< Referrer-Policy: strict-origin-when-cross-origin                                                                                                                                                                                             
< Set-Cookie: rememberMe=deleteMe; Path=/; Max-Age=0; Expires=Sun, 09-Apr-2023 15:00:27 GMT; SameSite=lax                                                                                                                                      
< Cache-Control: no-cache                                                                                                                                                                                                                      
< Location: http://10.168.3.59:8088/0f5f794c76c2c95f05b8cd5ac6a1e0e3/login                                                                                                                                                                     
< Content-Length: 0                                                                                                                                                                                                                            
[... cut ...]                   
* TCP_NODELAY set                                                                                                                                                                                                                              
                                                                                                                                  

                                                                                                            `

bobbywaz, 2023-04-10:
also: 
`* Rebuilt URL to: 'https://10.168.3.59:8043/
* Protocol "'https" not supported or disabled in libcurl
* Closing connection -1
curl: (1) Protocol "'https" not supported or disabled in libcurl`

mbentley, 2023-05-04:
What are you getting if you do a similar curl command?

I am not 100% sure why @bobbywaz is getting an error about not supporting https libcurl but it looks like it has something to do with quotes: https://stackoverflow.com/a/24232441.

In @bobbywaz 's case, try to do `curl -kvSL http://10.168.3.59:8088` (no quotes).

e3b0c442, 2023-05-07:
50 redirects to the same page.
```
➜  ~ curl -kvSL https://omada.REDACTED.com
*   Trying 10.1.227.4:443...
* Connected to omada.REDACTED.com (10.1.227.4) port 443 (#0)
* ALPN: offers h2
* ALPN: offers http/1.1
* [CONN-0-0][CF-SSL] (304) (OUT), TLS handshake, Client hello (1):
* [CONN-0-0][CF-SSL] (304) (IN), TLS handshake, Server hello (2):
* [CONN-0-0][CF-SSL] (304) (IN), TLS handshake, Unknown (8):
* [CONN-0-0][CF-SSL] (304) (IN), TLS handshake, Certificate (11):
* [CONN-0-0][CF-SSL] (304) (IN), TLS handshake, CERT verify (15):
* [CONN-0-0][CF-SSL] (304) (IN), TLS handshake, Finished (20):
* [CONN-0-0][CF-SSL] (304) (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / AEAD-AES256-GCM-SHA384
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=REDACTED.com
*  start date: Apr 27 13:56:55 2023 GMT
*  expire date: Jul 26 13:56:54 2023 GMT
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multiplexing
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* h2h3 [:method: GET]
* h2h3 [:path: /]
* h2h3 [:scheme: https]
* h2h3 [:authority: omada.REDACTED.com]
* h2h3 [user-agent: curl/7.87.0]
* h2h3 [accept: */*]
* Using Stream ID: 1 (easy handle 0x14f814800)
> GET / HTTP/2
> Host: omada.REDACTED.com
> user-agent: curl/7.87.0
> accept: */*
> 
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
< HTTP/2 302 
< date: Sun, 07 May 2023 19:03:06 GMT
< content-length: 0
< location: https://omada.REDACTED.com/9a19a503c2c4409a7b9c26965e6ec4ea/login
< x-frame-options: SAMEORIGIN
< x-content-type-options: nosniff
< content-security-policy: default-src 'self' https://*.tplinkcloud.com/;script-src 'self' 'unsafe-eval' 'sha256-7W9UiBaYGlOHpT1aQBLegqffUVHbYq6/ZAb+ErjUb40=' 'sha256-VGQ8jNTL2g0e8wPwOgyCQJDqhuRgfV7gRYexcBkBe4Y=' 'sha256-x2jgB1zBLi30IsfY+VNgWjwBGeHPJxOSrzl+IdsT6k0=' 'sha256-0AHZXO4clnpdcxqdmASPBEp4JCIrtaxIX/mUuL1kzZw=' 'sha256-lfXlPY3+MCPOPb4mrw1Y961+745U3WlDQVcOXdchSQc=';style-src 'self' 'unsafe-inline';connect-src 'self' https://*.tplinkcloud.com/ https://*.tplinkcloud.com:8843/ wss://*.tplinkcloud.com/ https://*.tiles.mapbox.com https://api.mapbox.com https://events.mapbox.com ;frame-src 'self' data:;img-src 'self' https://*.tplinkcloud.com/ data: blob:;child-src blob: ;worker-src blob: ;object-src 'self' data: blob:
< strict-transport-security: max-age=15724800; includeSubDomains
< referrer-policy: strict-origin-when-cross-origin
< set-cookie: rememberMe=deleteMe; Path=/; Max-Age=0; Expires=Sat, 06-May-2023 19:03:06 GMT; SameSite=lax
< cache-control: no-cache
< 
* Connection #0 to host omada.REDACTED.com left intact
* Issue another request to this URL: 'https://omada.REDACTED.com/9a19a503c2c4409a7b9c26965e6ec4ea/login'
* Found bundle for host: 0x600000500ff0 [can multiplex]
* Re-using existing connection #0 with host omada.REDACTED.com
* h2h3 [:method: GET]
* h2h3 [:path: /9a19a503c2c4409a7b9c26965e6ec4ea/login]
* h2h3 [:scheme: https]
* h2h3 [:authority: omada.REDACTED.com]
* h2h3 [user-agent: curl/7.87.0]
* h2h3 [accept: */*]
* Using Stream ID: 3 (easy handle 0x14f814800)
> GET /9a19a503c2c4409a7b9c26965e6ec4ea/login HTTP/2
> Host: omada.REDACTED.com
> user-agent: curl/7.87.0
> accept: */*
> 
< HTTP/2 302 
< date: Sun, 07 May 2023 19:03:06 GMT
< content-length: 0
< location: https://omada.REDACTED.com/9a19a503c2c4409a7b9c26965e6ec4ea/login
< x-frame-options: SAMEORIGIN
[... 1258 lines cut ...]
* h2h3 [:path: /9a19a503c2c4409a7b9c26965e6ec4ea/login]
* h2h3 [:scheme: https]
* h2h3 [:authority: omada.REDACTED.com]
* h2h3 [user-agent: curl/7.87.0]
* h2h3 [accept: */*]
* Using Stream ID: 101 (easy handle 0x14f814800)
> GET /9a19a503c2c4409a7b9c26965e6ec4ea/login HTTP/2
> Host: omada.REDACTED.com
> user-agent: curl/7.87.0
> accept: */*
> 
< HTTP/2 302 
< date: Sun, 07 May 2023 19:03:06 GMT
< content-length: 0
< location: https://omada.REDACTED.com/9a19a503c2c4409a7b9c26965e6ec4ea/login
< x-frame-options: SAMEORIGIN
< x-content-type-options: nosniff
< content-security-policy: default-src 'self' https://*.tplinkcloud.com/;script-src 'self' 'unsafe-eval' 'sha256-7W9UiBaYGlOHpT1aQBLegqffUVHbYq6/ZAb+ErjUb40=' 'sha256-VGQ8jNTL2g0e8wPwOgyCQJDqhuRgfV7gRYexcBkBe4Y=' 'sha256-x2jgB1zBLi30IsfY+VNgWjwBGeHPJxOSrzl+IdsT6k0=' 'sha256-0AHZXO4clnpdcxqdmASPBEp4JCIrtaxIX/mUuL1kzZw=' 'sha256-lfXlPY3+MCPOPb4mrw1Y961+745U3WlDQVcOXdchSQc=';style-src 'self' 'unsafe-inline';connect-src 'self' https://*.tplinkcloud.com/ https://*.tplinkcloud.com:8843/ wss://*.tplinkcloud.com/ https://*.tiles.mapbox.com https://api.mapbox.com https://events.mapbox.com ;frame-src 'self' data:;img-src 'self' https://*.tplinkcloud.com/ data: blob:;child-src blob: ;worker-src blob: ;object-src 'self' data: blob:
< strict-transport-security: max-age=15724800; includeSubDomains
< referrer-policy: strict-origin-when-cross-origin
< 
* Connection #0 to host omada.REDACTED.com left intact
* Maximum (50) redirects followed
curl: (47) Maximum (50) redirects followed
```

mbentley, 2023-05-09:
Hm, I'm trying to figure out in what scenario it would end up in a redirect loop that never completes.  How are you exposing the service if it is deploy to k8s?  Ingress?  Load balancer?  Odd that it would just suddenly stop working but I doubt it has anything to do with the packaging of the software in the container.

bobbywaz, 2023-05-21:
I'm just using docker, no K8's, no load balancing, nothing fance. I just deleted it and restarted it with the newest version and now it's booting fine, but when I access the URL through my reverse proxy (omada.bobbyw.name)  it's redirecting to the internal port... (omada.bobbyw.name:8043). Is there a way to use it without the redirect at all?

mbentley, 2023-06-03:
Sorry, I forgot to reply on this.  There is a similar discussion [about using Traefik to reverse proxy](https://github.com/mbentley/docker-omada-controller/discussions/317) that might be helpful.  The software always wants to use the internal port that the service is listening on and from my experience, doesn't handle that well otherwise without the same port internally and externally being the same.

closed 2023-06-03 by mbentley