mbentley/docker-omada-controller #301: [Bug]: Unable to access web page after upgrading

smcgann99, 2023-04-13:
### Controller Version

5.9.31

### Describe the Bug

I had an older version of this image working for some months, however I tried to upgrade to the latest version, without any success.

I'm new to docker and couldn't find specific instructions to upgrade in your docs, so I followed a generic guide.

The included log shows the full list of commands and output -

This stands out but I couldn't find anything to explain how to fix it.
**_ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Start init data._**

Any help would be greatly appreciated !

### Expected Behavior

Able to log into web interface

### Steps to Reproduce

pi@pi-server:~ $ sudo docker ps
CONTAINER ID   IMAGE                              COMMAND                  CREATED          STATUS                    PORTS                                                      NAMES
55e0ec3f4075   mbentley/omada-controller:latest   "/entrypoint.sh /usr…"   19 minutes ago   Up 19 minutes (healthy)   8043/tcp, 8088/tcp, 8843/tcp, 29811-29814/tcp, 29810/udp   omada-controller
pi@pi-server:~ $ sudo docker stop 55e0ec3f4075
55e0ec3f4075
pi@pi-server:~ $ sudo docker rm 55e0ec3f4075
55e0ec3f4075
pi@pi-server:~ $ sudo docker run --name omada-controller  mbentley/omada-controller:latest

### How You're Launching the Container

```plain
sudo docker run --name omada-controller  mbentley/omada-controller:latest
```

### Container Logs

```plain
pi@pi-server:~ $ sudo docker ps
CONTAINER ID   IMAGE                              COMMAND                  CREATED          STATUS                    PORTS                                                      NAMES
55e0ec3f4075   mbentley/omada-controller:latest   "/entrypoint.sh /usr…"   19 minutes ago   Up 19 minutes (healthy)   8043/tcp, 8088/tcp, 8843/tcp, 29811-29814/tcp, 29810/udp   omada-controller
pi@pi-server:~ $ sudo docker stop 55e0ec3f4075
55e0ec3f4075
pi@pi-server:~ $ sudo docker rm 55e0ec3f4075
55e0ec3f4075
pi@pi-server:~ $ sudo docker run --name omada-controller  mbentley/omada-controller:latest
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'Etc/UTC'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
WARN: Ownership not set correctly on '/opt/tplink/EAPController/data'; setting correct ownership (omada:omada)
WARN: Ownership not set correctly on '/opt/tplink/EAPController/logs'; setting correct ownership (omada:omada)
WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
INFO: Version check passed; image version (5.9.31) >= the last version ran (0.0.0); writing image version to last ran file...
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
[... 73 lines cut ...]
04-13-2023 03:41:22.403 INFO [main] [] c.t.s.o.d.g.a(): manage    port v2 is 29814.
04-13-2023 03:41:22.403 INFO [main] [] c.t.s.o.d.g.a(): adopt     port v1 is 29812.
04-13-2023 03:41:22.403 INFO [main] [] c.t.s.o.d.g.a(): upgrade   port is 29813.
04-13-2023 03:41:22.403 INFO [main] [] c.t.s.o.d.g.a(): transfer  port is 29815.
04-13-2023 03:41:22.403 INFO [main] [] c.t.s.o.d.g.a(): rtty port is 29816.
04-13-2023 03:41:22.604 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyUdpServer bind /0.0.0.0:29810, export localhost/127.0.0.1:29810
04-13-2023 03:41:22.646 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29812 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
04-13-2023 03:41:22.673 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
04-13-2023 03:41:22.680 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
04-13-2023 03:41:22.684 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
04-13-2023 03:41:22.714 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
04-13-2023 03:41:22.717 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
04-13-2023 03:41:22.727 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
04-13-2023 03:41:22.730 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
04-13-2023 03:41:22.736 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29815 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
04-13-2023 03:41:22.738 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29815, export localhost/127.0.0.1:29815
04-13-2023 03:41:22.751 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start RttyNettyTcpServer bind /0.0.0.0:29816, export localhost/127.0.0.1:29816
04-13-2023 03:41:22.752 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
04-13-2023 03:41:22.753 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started
04-13-2023 03:41:22.753 INFO [main] [] c.t.s.o.s.t.SuccessExitTask(): Omada Controller started
04-13-2023 03:41:22.753 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: finished run omada tasks
04-13-2023 03:41:52.002 INFO [auth-record-traffic-task-pool-0] [] c.t.s.o.c.c.d.m.s.l(): Omadac 25aa215cab36a3cc471737973870a206 has no site.
04-13-2023 03:42:21.992 INFO [auth-record-traffic-task-pool-0] [] c.t.s.o.c.c.d.m.s.l(): Omadac 25aa215cab36a3cc471737973870a206 has no site.
04-13-2023 03:42:22.018 WARN [monitor-schedule-topology-pool-0] [] c.t.s.o.m.t.p.s.t.TopologyTask(): findSiteList fail, omadacId=25aa215cab36a3cc471737973870a206
```

mbentley, 2023-04-13:
Startup looks normal to me.  What exactly happens when you try to access the login page?  Have you tried a hard refresh of your browser?

smcgann99, 2023-04-14:
Thanks for getting back - In the end I removed everything and started from scratch using info from other places.
It would help newbies like me if you could sign post an idiots guide to installing and none destructive upgrading of you excellent package ;-)

closed 2023-04-14 by smcgann99