mbentley/docker-omada-controller #304: [Bug]:Omada not starting on Raspi4: failed to start MongoDB

pjrobin, 2023-04-26:
### Controller Version

5.9.31

### Describe the Bug

I've been running Omada Docker for a while and today after an update of my Pi 4, the docker doesn't want to start anymore with the same error described in #97.
I tried to delete everything (data and log) but it won't solve the problem.

The server.log shows
`2023-04-26T14:26:58.681+0200 ***** SERVER RESTARTED *****
2023-04-26T14:26:58.682+0200 SEVERE: Invalid access at address: 0x45d333e
2023-04-26T14:26:58.688+0200 SEVERE: Got signal: 7 (Bus error).
Backtrace:0x7639be 0x762f6e 0x76300a 0xf7322270 0x75ece4 0x75cd7e 0x75cdf4 0x1f0300 0x1fc81a 0x1fcb7e 0x1fcdc8 0x1c4e64 0xf73138aa 
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo15printStackTraceERSo+0x25) [0x7639be]
 /opt/tplink/EAPController/bin/mongod() [0x762f6e]
 /opt/tplink/EAPController/bin/mongod() [0x76300a]
 /lib/arm-linux-gnueabihf/libc.so.6(+0x25270) [0xf7322270]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo10SystemInfo17collectSystemInfoEv+0x87f) [0x75ece4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo20initializeSystemInfoEv+0x7d) [0x75cd7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo36_mongoInitializerFunction_SystemInfoEPNS_18InitializerContextE+0x7) [0x75cdf4]
 /opt/tplink/EAPController/bin/mongod(_ZN5boost6detail8function17function_invoker1IPFN5mongo6StatusEPNS3_18InitializerContextEES4_S6_E6invokeERNS1_15function_bufferES6_+0x17) [0x1f0300]
 /opt/tplink/EAPController/bin/mongod(_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE+0x15d) [0x1fc81a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE+0x29) [0x1fcb7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersEiPKPKcS3_+0x22b) [0x1fcdc8]
 /opt/tplink/EAPController/bin/mongod(main+0x6f) [0x1c4e64]
 /lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x9d) [0xf73138aa]
`

### Expected Behavior

well, should start :)

### Steps to Reproduce

Install Omada Docker on Raspi 4
Try to start

### How You're Launching the Container

```plain
I'm using portainer... no idee where to find the arguments
```

### Container Logs

```plain
at com.mongodb.client.internal.MongoClientDelegate$DelegateOperationExecutor.getClientSession(MongoClientDelegate.java:291) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoClientDelegate$DelegateOperationExecutor.execute(MongoClientDelegate.java:207) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoCollectionImpl.executeCreateIndexes(MongoCollectionImpl.java:848) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoCollectionImpl.createIndexes(MongoCollectionImpl.java:831) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoCollectionImpl.createIndexes(MongoCollectionImpl.java:826) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoCollectionImpl.createIndex(MongoCollectionImpl.java:811) ~[mongodb-driver-sync-4.4.2.jar:?]
	at org.springframework.data.mongodb.core.DefaultIndexOperations.lambda$ensureIndex$0(DefaultIndexOperations.java:131) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.MongoTemplate.execute(MongoTemplate.java:560) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.DefaultIndexOperations.execute(DefaultIndexOperations.java:214) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.DefaultIndexOperations.ensureIndex(DefaultIndexOperations.java:121) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.index.MongoPersistentEntityIndexCreator.createIndex(MongoPersistentEntityIndexCreator.java:156) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.index.MongoPersistentEntityIndexCreator.checkForAndCreateIndexes(MongoPersistentEntityIndexCreator.java:146) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.index.MongoPersistentEntityIndexCreator.checkForIndexes(MongoPersistentEntityIndexCreator.java:130) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.index.MongoPersistentEntityIndexCreator.onApplicationEvent(MongoPersistentEntityIndexCreator.java:114) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.data.mongodb.core.index.MongoPersistentEntityIndexCreator.onApplicationEvent(MongoPersistentEntityIndexCreator.java:54) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.doInvokeListener(SimpleApplicationEventMulticaster.java:176) ~[spring-context-5.3.18.jar:5.3.18]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.invokeListener(SimpleApplicationEventMulticaster.java:169) ~[spring-context-5.3.18.jar:5.3.18]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.multicastEvent(SimpleApplicationEventMulticaster.java:143) ~[spring-context-5.3.18.jar:5.3.18]
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:421) ~[spring-context-5.3.18.jar:5.3.18]
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:378) ~[spring-context-5.3.18.jar:5.3.18]
	at org.springframework.data.mapping.context.AbstractMappingContext.addPersistentEntity(AbstractMappingContext.java:391) ~[spring-data-commons-2.6.3.jar:2.6.3]
	at org.springframework.data.mapping.context.AbstractMappingContext.getPersistentEntity(AbstractMappingContext.java:266) ~[spring-data-commons-2.6.3.jar:2.6.3]
	at org.springframework.data.mapping.context.AbstractMappingContext.getPersistentEntity(AbstractMappingContext.java:209) ~[spring-data-commons-2.6.3.jar:2.6.3]
[... 52 lines cut ...]
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
INFO: Starting Omada Controller as user omada
04-26-2023 16:55:44.484 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
04-26-2023 16:55:44.554 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
04-26-2023 16:55:44.568 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
04-26-2023 16:55:44.578 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
04-26-2023 16:55:44.587 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
04-26-2023 16:55:44.591 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
04-26-2023 16:55:44.661 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
04-26-2023 16:57:24.805 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Fail to start mongo DB server
04-26-2023 16:57:24.805 WARN [main] [] c.t.s.o.s.s.b(): Fail to start mongo DB server
04-26-2023 16:57:24.805 WARN [main] [] c.t.s.o.s.s.a(): com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
	at com.tplink.smb.omada.starter.server.b.a(SourceFile:151) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.server.a.e(SourceFile:74) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.OmadaBootstrap.d(SourceFile:249) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.OmadaBootstrap.q(SourceFile:329) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:116) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:278) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:92) ~[local-starter-5.9.31.jar:5.9.31]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:42) ~[local-starter-5.9.31.jar:5.9.31]
```

mbentley, 2023-04-26:
This also looks like the same as https://github.com/mbentley/docker-omada-controller/issues/302.  Can you try what I asked the other person to do here: https://github.com/mbentley/docker-omada-controller/issues/302#issuecomment-1515345355

mbentley, 2023-04-26:
What error(s) are returned when you attempt to specifically run the `latest-arm64` tagged image?  Is it complaining about the architecture?  I may have to setup an armv7l raspberry pi to test.  I only have a Raspberry Pi 3b+ using a 64 bit aarch64 kernel on Debian.

The only way I am able to test otherwise, is with either an amd64 or arm64 machine using emulation to armv7l and both of them start both the armv7l and arm64 images as expected.

pjrobin, 2023-04-26:
When I start the lastest-arm64 then I get no error, no log, the container is marked as running but I get an empty log.

```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

mbentley, 2023-04-26:
Hmm ok.  Must be spitting out an error about architecture and failing to start then.  I will try to reproduce if I can find another Raspberry Pi to test with.

Zauberio, 2023-04-28:
Ive got the same issue. Tried downgrading to a past version, but it didnt work. Keep getting:
`INFO: Starting Omada Controller as user omada
2023-04-28 07:38:56 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2023-04-28 07:38:56 [main] [INFO]-[SourceFile:89] - going to start local mongod.
2023-04-28 07:40:36 [main] [WARN]-[SourceFile:151] - Fail to start mongo DB server
2023-04-28 07:40:36 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server
com.tplink.omada.start.a.a: Fail to start mongo DB server
	at com.tplink.omada.start.b.d.a(SourceFile:152) ~[omada-start.jar:?]
	at com.tplink.omada.start.b.b.e(SourceFile:75) [omada-start.jar:?]
	at com.tplink.omada.start.OmadaBootstrap.d(SourceFile:223) [omada-start.jar:?]
	at com.tplink.omada.start.OmadaBootstrap.q(SourceFile:287) [omada-start.jar:?]
	at com.tplink.omada.start.OmadaBootstrap.a(SourceFile:100) [omada-start.jar:?]
	at com.tplink.omada.start.OmadaBootstrap.e(SourceFile:245) [omada-start.jar:?]
	at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:80) [omada-start.jar:?]
	at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) [omada-start.jar:?]

      `

mbentley, 2023-04-28:
And what version of Raspbian are you running?  Could you also provide the output of:
- `uname -a`
- `cat /etc/os-release`
- `docker info`

mbentley, 2023-04-30:
Also, is this happening after an upgrade from one version of the controller to another?  Any idea what version you're coming from?

I just setup clean installs on a Raspberry Pi 3b+ with both 32 (armv7l) and 64 bit (arm64) versions of Raspbian and I was able to start a new  controller on both of them as expected so something odd is going on here.

On the 32 bit, I was running a Debian 11 (bullseye) based version and 64 bit was Debian 10 (buster) based.

```
# 64 bit
$ uname -a
Linux testpi-arm64 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
# 32 bit
$ uname -a
Linux testpi-armv7l 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

pjrobin, 2023-04-30:
no it happens after apt upgrade and firmware upgrade. Didn't change Omada version.

```
### /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

### /etc/apt/sources.list.d/debian-backports.list
deb http://httpredir.debian.org/debian buster-backports main contrib non-free

### /etc/apt/sources.list.d/docker.list
deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable

### /etc/apt/sources.list.d/nodesource.list
deb https://deb.nodesource.com/node_10.x buster main
deb-src https://deb.nodesource.com/node_10.x buster main

### /etc/apt/sources.list.d/openhab2.list
deb https://openhab.jfrog.io/openhab/openhab-linuxpkg testing main

### /etc/apt/sources.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ buster main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ buster main

### /etc/apt/sources.list.d/vscode.list
### Disabled by raspberrypi-sys-mods ###
```

mbentley, 2023-04-30:
OK I will have to tear but what I suspect is happening is that it started originally with an armv7l kernel and then moved to an arm64 kernel.

mbentley, 2023-05-02:
OK, did your firmware update of your raspberry pi switch it from a 32 bit kernel to a 64 bit kernel?  If I start an armv7l container on a pi with an armv7l kernel and then update the `/boot/config.txt`, set `arm_64bit=1`, and reboot and then try to start the container, I get the same errors:

```
2023-05-02T11:07:02.639+0000 ***** SERVER RESTARTED *****
2023-05-02T11:07:02.653+0000 SEVERE: Invalid access at address: 0x340333e
2023-05-02T11:07:02.704+0000 SEVERE: Got signal: 7 (Bus error).
Backtrace:0x7639be 0x762f6e 0x76300a 0xf6fa8270 0x75ece4 0x75cd7e 0x75cdf4 0x1f0300 0x1fc81a 0x1fcb7e 0x1fcdc8 0x1c4e64 0xf6f998aa
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo15printStackTraceERSo+0x25) [0x7639be]
 /opt/tplink/EAPController/bin/mongod() [0x762f6e]
 /opt/tplink/EAPController/bin/mongod() [0x76300a]
 /lib/arm-linux-gnueabihf/libc.so.6(+0x25270) [0xf6fa8270]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo10SystemInfo17collectSystemInfoEv+0x87f) [0x75ece4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11ProcessInfo20initializeSystemInfoEv+0x7d) [0x75cd7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo36_mongoInitializerFunction_SystemInfoEPNS_18InitializerContextE+0x7) [0x75cdf4]
 /opt/tplink/EAPController/bin/mongod(_ZN5boost6detail8function17function_invoker1IPFN5mongo6StatusEPNS3_18InitializerContextEES4_S6_E6invokeERNS1_15function_bufferES6_+0x17) [0x1f0300]
 /opt/tplink/EAPController/bin/mongod(_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE+0x15d) [0x1fc81a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE+0x29) [0x1fcb7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo21runGlobalInitializersEiPKPKcS3_+0x22b) [0x1fcdc8]
 /opt/tplink/EAPController/bin/mongod(main+0x6f) [0x1c4e64]
 /lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x9d) [0xf6f998aa]
```

From the `sources.list`, you have `armhf` repos (which are 32 bit) and then you are running a 64 bit kernel (`6.1.25-v8+`).  Unless you have a specific reason to run a 64 bit kernel, you can likely switch back to a 32 bit by modifying the `config.txt` to set `arm_64bit=0`.

mbentley, 2023-05-02:
> When I start the lastest-arm64 then I get no error, no log, the container is marked as running but 1) I get an empty log and 2) the container keeps on restarting.

OK, so one thing I do see is that it is throwing an error related to the architecture from my testing:

```
$ docker run -d --name omada-controller2 -v omada-data:/opt/tplink/EAPController/data -v omada-logs:/opt/tplink/EAPController/logs mbentley/omada-controller:5.9-arm64
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v8) and no specific platform was requested
416a64ae257c61a9ac511dfd6580c2874775bec5c2145b4eff26998395d7d51a
```

mbentley, 2023-05-02:
Also @pjrobin - what process did you use to update the firmware for your raspberry pi?  I just want to understand how exactly you got from a 32 bit kernel to a 64 bit because switching kernels is something I wouldn't think they should automate because of unintended consequences like this.

travisfranck, 2023-05-03:
I have a 8GB RPi4, so I think I've been running the 64-bit OS from the Day 1.  

When I ssh into my pi, it says [snipped by me]:
```
Linux rpi 6.1.21-v8+ [SNIP] aarch64
```

here is my config.txt contents.

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
gpu_mem=256
[all]
dtoverlay=gpio-fan,gpiopin=17,temp=80000
```

mbentley, 2023-05-03:
@travisfranck -   Could you also provide the output of:
- `uname -a`
- `cat /etc/os-release`
- `docker info`
- `dpkg --print-architecture`
- `for i in /etc/apt/sources.list /etc/apt/sources.list.d/*; do echo "### $i"; cat "$i";echo;done`

I just want to verify what the base OS is installed as and also get details about your kernel + docker.  That'll tell me if you were ever on 32 bit vs 64 bit.

@pjrobin - based on your output, you have a 32 bit userland with a 64 bit kernel as the output of `dpkg --print-architecture` is how docker builds the apt config:

```
deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable
```

If it shows `armhf`, that's 32 bit userland.  So if you revert back to a 32 bit kernel by setting `arm_64bit=0` in `/boot/config.txt`, I believe you should be ok after a reboot.

Installing from a 64 bit Raspbian, I see:

```
$ dpkg --print-architecture
arm64
```

That's what I would expect if the userland is 64 bit.

travisfranck, 2023-05-03:
@mbentley I really appreciate your help and dedication.

I thought the Pi was 8GB but the `docker info` command only says 4-ish.  Is that the docker environment, not the machine hardware?

The only thing that sticks out to me is the `armhf` in the dpkg command.

Just a reminder, everything was working.  I did some commands to update the packages, thinking that is good practice, but maybe something choked. I've also been pretty good about updating the Omada image as you do your magic. I didn't change the OS version any time in recent memory.

As requested:

```
pi@rpi:~ $ uname -a
Linux rpi 6.1.21-v8+ #1642 SMP PREEMPT [SNIP] aarch64 GNU/Linux

pi@rpi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@rpi:~ $ docker info
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.10.2
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.16.0
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 2
  Running: 1
  Paused: 0
  Stopped: 1
 Images: 3
 Server Version: 23.0.1
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 2456e983eb9e37e47538f59ea18f2043c9a73640
 runc version: v1.1.4-0-g5fd4c4d
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 6.1.21-v8+
 Operating System: Raspbian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 3.534GiB
 Name: rpi
 ID: M4XO:LTNI:ABPZ:W5BL:[SNIP]
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Experimental: false
 Insecure Registries:
[SNIP]
 Live Restore Enabled: false

WARNING: No memory limit support
WARNING: No swap limit support

pi@rpi:~ $ dpkg --print-architecture
armhf

pi@rpi:~ $ for i in /etc/apt/sources.list /etc/apt/sources.list.d/*; do echo "### $i"; cat "$i";echo;done
### /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi

### /etc/apt/sources.list.d/docker.list
deb [arch=armhf signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/raspbian bullseye stable

### /etc/apt/sources.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ bullseye main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ bullseye main
```

mbentley, 2023-05-03:
That should be the total RAM on the system; I would assume `free -m` would confirm that.  Seems like your pi has 32 bit userland binaries and a 64 bit kernel as well.  Do you have the specific update commands you ran to help me understand how it may end up in this sort of state?

travisfranck, 2023-05-03:
Must be a 4GB machine, with the balance going to graphic memory.  Sorry I got that wrong.

I dug through my command history and can reconstruct the following:

I believe I used these commands in 2022 without issue, while also updating the Omada image regularly:
```
$ sudo apt update
$ sudo apt upgrade
```

Early this year I felt it was time to make sure I had all the security patches and such, so I logged into VNC and used their GUI to do an update.  I also used the following commands around the same time:
```
$ sudo apt-get update
$ sudo apt-get dist-upgrade
```

And, more recently, I went back to the first means, but I think this was after the docker image wasn't working and I was hoping some updates might fix it:
```
$ sudo apt update
$ sudo apt list --upgradeable
$ sudo apt full-upgrade
$ sudo apt clean
```

So, maybe it was the `dist-upgrade` command?

mbentley, 2023-05-03:
Thanks, that might prove to be helpful. Do you know if you were on an older version of Raspbian, like 10 (buster) and it upgraded to 11 (bullseye)?  If you don’t recall, that’s no big deal but could help me narrow it down as I try o reproduce especially since upgrades as so slow on a Raspberry Pi writing to an sd card so it’s a bit time intensive.

FrKii, 2023-05-04:
I had the same issue today, also after running some Updates (update/upgrade) and rebooting my Rapsberry Pi 4:

$cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$dpkg --print-architecture
armhf

However I did not perform an dist upgrade.
In an attempt to "fix" it tried creating a new/clean container and volumes which didn't help.
Setting the "arm_64bit=0" flag in the "/boot/config.txt" and rebooting fixed the issue for me.

I'd be happy to share some more info if that helps

mbentley, 2023-05-04:
@FrKii - thanks for the report!  Some logs would be interesting to understand what happened from the upgrade.  I believe that you should have logs of what it did in `/var/log/apt`.  The `history.log` and maybe the `term.log` would be helpful.  The `history.log` is what I am mostly interested in from when you upgraded to see what packages it installed because I am almost certain that one of them would have triggered the change.  The `term.log` would be helpful to see what post-processing happened during the install/upgrade of packages.

Just check and make sure that the two logs are from whenever you did the upgrade and rebooted.  There is also a possibility that something happened a good while before the upgrade and reboot if you hadn't rebooted it in a while since the kernel change would only happen after reboot but it would be great to take a look to investigate at least.

Meanwhile, I will be able to add some information in the logging output to warn people who are running in the same issue and how to resolve it.

mbentley, 2023-05-04:
Aha - https://github.com/raspberrypi/firmware/issues/1795#issuecomment-1474883509

This was an intentional change.  Setting the `arm_64bit=0` flag in the "/boot/config.txt" and rebooting is the appropriate way to deal with this.

mbentley, 2023-05-04:
I added a known issue with instructions on how to deal with the problem.

https://github.com/mbentley/docker-omada-controller/blob/master/KNOWN_ISSUES.md#mismatched-userland-and-kernel

There is also a check in the entrypoints that will output an error to those impacted to link to that known issue and stop further attempts at starting the controller since it will be unsuccessful.  There is a bypass by setting `SKIP_USERLAND_KERNEL_CHECK` to `true` to have it skip that check if for whatever reason someone is running a 32 bit userland and a 64 bit kernel and it works fine, like maybe non-rpi4 devices might work.

closed 2023-05-04 by mbentley

travisfranck, 2023-05-11:
A belated confirmation that `arm_64bit=0` worked for me.  Thank you @FrKii for the suggestion and @mbentley for the diligence and continued maintenance of this image.

msb1111, 2023-12-05:
Hi,
I have a Rpi Model 3B with Piaware and have been running omada through the docker for several years without a problem. After a recent update the container no longer loads and shows Unhealthy. I have tried removing and reinstalling the container several times and also put arm_64bit=0 in the config file and rebooted but it hasn't helped.

Would be very grateful if someone could assist me with this. The log from portainer is as under, which shows "Fail to start mongo DB server":

In case I am unable to get it working, is there a way to save the config files and move them to a new installation of raspbian/docker?

`at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:577) ~[spring-context-5.3.28.jar:5.3.28]
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:147) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:731) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:408) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:307) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1303) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1292) ~[spring-boot-2.7.13.jar:2.7.13]
	at com.tplink.smb.omada.starter.task.SpringBootStartUpTask.retryStartApplication(SourceFile:60) ~[local-starter-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.starter.task.SpringBootStartUpTask.execute(SourceFile:41) ~[local-starter-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.starter.task.c.a(SourceFile:20) ~[local-starter-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.starter.OmadaBootstrap.f(SourceFile:356) ~[local-starter-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:99) ~[local-starter-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:42) ~[local-starter-5.13.22.jar:5.13.22]
Caused by: java.net.BindException: Address already in use
	at sun.nio.ch.Net.bind0(Native Method) ~[?:1.8.0_292]
	at sun.nio.ch.Net.bind(Net.java:461) ~[?:1.8.0_292]
	at sun.nio.ch.Net.bind(Net.java:453) ~[?:1.8.0_292]
	at sun.nio.ch.ServerSocketChannelImpl.bind(ServerSocketChannelImpl.java:222) ~[?:1.8.0_292]
	at org.apache.tomcat.util.net.NioEndpoint.initServerSocket(NioEndpoint.java:276) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.tomcat.util.net.NioEndpoint.bind(NioEndpoint.java:231) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.tomcat.util.net.AbstractEndpoint.bindWithCleanup(AbstractEndpoint.java:1317) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.tomcat.util.net.AbstractEndpoint.init(AbstractEndpoint.java:1330) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.coyote.AbstractProtocol.init(AbstractProtocol.java:652) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.coyote.http11.AbstractHttp11Protocol.init(AbstractHttp11Protocol.java:75) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.connector.Connector.initInternal(Connector.java:1009) ~[tomcat-embed-core-9.0.76.jar:9.0.76]

	... 26 more
12-05-2023 11:45:33.128 ERROR [main] [] o.a.c.u.LifecycleBase(): Failed to initialize component [Connector[HTTP/1.1-8843]]
org.apache.catalina.LifecycleException: Protocol handler initialization failed
	at org.apache.catalina.connector.Connector.initInternal(Connector.java:1011) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.util.LifecycleBase.init(LifecycleBase.java:136) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.core.StandardService.initInternal(StandardService.java:549) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.util.LifecycleBase.init(LifecycleBase.java:136) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.core.StandardServer.initInternal(StandardServer.java:1041) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.util.LifecycleBase.init(LifecycleBase.java:136) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:173) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.apache.catalina.startup.Tomcat.start(Tomcat.java:486) ~[tomcat-embed-core-9.0.76.jar:9.0.76]
	at org.springframework.boot.web.embedded.tomcat.TomcatWebServer.initialize(TomcatWebServer.java:123) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.web.embedded.tomcat.TomcatWebServer.<init>(TomcatWebServer.java:104) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.web.embedded.tomcat.TomcatServletWebServerFactory.getTomcatWebServer(TomcatServletWebServerFactory.java:481) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.web.embedded.tomcat.TomcatServletWebServerFactory.getWebServer(TomcatServletWebServerFactory.java:211) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.createWebServer(ServletWebServerApplicationContext.java:184) ~[spring-boot-2.7.13.jar:2.7.13]
	at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.onRefresh(ServletWebServerApplicationContext.java:162) ~[spring-boot-2.7.13.jar:2.7.13]
[... cut ...]
	at com.tplink.smb.omada.starter.OmadaBootstrap.n(SourceFile:249) ~[local-starter-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.starter.OmadaBootstrap.x(SourceFile:404) ~[local-starter-5.13.22.jar:5.13.22]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_292]
12-05-2023 11:48:43.614 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
12-05-2023 11:48:43.614 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.`

mbentley, 2023-12-05:
I’m going to need to see the MongoDB logs to understand why it didn’t start. See https://github.com/mbentley/docker-omada-controller/blob/master/SUPPORT_INFO.md for how to get those.