mbentley/docker-omada-controller #305: [Bug]: Omada controller starts no webage, unable to access database info

Commiccannon, 2023-04-30:
### Controller Version

I cannot get to the webpage, whatever the latest docker compose example file gives

### Describe the Bug

So i have tried to setup the docker container of omada after the windows version kept getting stuck after mongodb started. According to the logs the server all starts, mongod starts then i start getting this in the logs, it repeats about every minuite or two `04-30-2023 23:28:04.859 WARN [monitor-schedule-topology-pool-0] [] c.t.s.o.m.t.p.s.t.TopologyTask(): findSiteList fail, omadacId=3f39ad3e1aef1c52280f12f5a2cca908
04-30-2023 23:28:34.852 INFO [auth-record-traffic-task-pool-0] [] c.t.s.o.c.c.d.m.s.l(): Omadac 3f39ad3e1aef1c52280f12f5a2cca908 has no site.
04-30-2023 23:29:04.851 INFO [auth-record-traffic-task-pool-0] [] c.t.s.o.c.c.d.m.s.l(): Omadac 3f39ad3e1aef1c52280f12f5a2cca908 has no site.
04-30-2023 23:29:04.858 WARN [monitor-schedule-topology-pool-0] [] c.t.s.o.m.t.p.s.t.TopologyTask(): findSiteList fail, omadacId=3f39ad3e1aef1c52280f12f5a2cca908`
Any ideas as to how to fix would be appreciated

### Expected Behavior

Server starts, i can access the webpage and everything works

### Steps to Reproduce

1. Install docker with the supplied docker compose only changing the ports as i have many other servers
2. Start docker
3. Watch logs

### How You're Launching the Container

```plain
version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.9
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=8000
      - MANAGE_HTTPS_PORT=8001
      - PORTAL_HTTP_PORT=8002
      - PORTAL_HTTPS_PORT=8003
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Etc/UTC
    volumes:
      - C:/Omada:/opt/tplink/EAPController/data
      - C:/Omada/logs:/opt/tplink/EAPController/logs

volumes:
  omada-data:
  omada-logs:
```

### Container Logs

```plain
Files will be in a comment as this section isnt letting me add files
```

Commiccannon, 2023-04-30:
[mongod.log](https://github.com/mbentley/docker-omada-controller/files/11362676/mongod.log)
[Server.log](https://github.com/mbentley/docker-omada-controller/files/11362677/Server.log)

mbentley, 2023-05-01:
Are you running Docker for Windows with wsl2?  I haven't done any testing on Windows myself so I am not sure if there are any complications.  With the persistent data, I am not sure if there would be any issues for the persistent data.  The "no site" message is the controller basically just saying setup has not run yet so it's not an error.

I see shutdown messages happening.  Did you stop/restart the container where that is showing in the logs or did that happen on it's own?

Are you sure it's not a browser issue with the page not loading?  Could you [hard refresh in your browser](https://filecamp.com/support/problem-solving/hard-refresh/) to make sure something janky isn't being cached?  Could you try with curl to see what loads?

```
curl -kvL https://<ip-or-hostname>:8001
```

Commiccannon, 2023-05-01:
Hi, yes docker is running on wsl2 as windows 10 home can only run wsl2. I did stop the container sorry for the added confusion. Tried hard refreshing still getting nothing. And here is the output of curl
*   Trying 10.0.0.102:8001...
* connect to 10.0.0.102 port 8001 failed: Timed out
* Failed to connect to 10.0.0.102 port 8001 after 21038 ms: Couldn't connect to server
* Closing connection 0
curl: (28) Failed to connect to 10.0.0.102 port 8001 after 21038 ms: Couldn't connect to server

Commiccannon, 2023-05-03:
Hi, I have a few other servers running from docker on the same machine. I had to get omada up urgently a few days ago so put it on another old pc I have running ubuntu. I'm happy to keep working with you to try and fix the bug if you want to squash it but it's not something I need to do

mbentley, 2023-05-03:
Up to you - I’m pretty confident the container images work though.

mbentley, 2023-05-04:
Closing, feel free to reopen if you would like to pursue any further troubleshooting.

closed 2023-05-04 by mbentley

MichaelVoelkel, 2023-06-03:
Hi,
same issue, I have an aarch64 Linux VM on a Mac M1. All seems to run fine, ports are published etc. But I cannot really connect to any of said ports.

MichaelVoelkel, 2023-06-03:
ah, --net host solved, thanks