mbentley/docker-omada-controller #310: [Bug]:

Termibg22, 2023-05-21:
### Controller Version

v5.9.31

### Describe the Bug

I don't know why but the MongoDB is not initializating, i looked at the file permissions but that looks right.

### Expected Behavior

MongoDB starting without errors

### Steps to Reproduce

1. Start the container

### How You're Launching the Container

```plain
docker run
  -d
  --name='omada'
  --net='host'
  -e TZ="Europe/Paris"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="unraid"
  -e HOST_CONTAINERNAME="omada"
  -e 'MANAGE_HTTPS_PORT'='18043'
  -e 'MANAGE_HTTP_PORT'='18088'
  -e 'PORTAL_HTTPS_PORT'='18843'
  -e 'PORTAL_HTTP_PORT'='18089'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.webui='https://[IP]:[PORT:18043]/login'
  -l net.unraid.docker.icon='https://raw.githubusercontent.com/benhedrington/hedrington-unraid-docker-templates/main/hedrington-unraid-docker-templates/omada-controller.png'
  -v '/mnt/user/appdata/omada-controller/data/':'/opt/tplink/EAPController/data':'rw,slave'
  -v '/mnt/user/appdata/omada-controller/work/':'/opt/tplink/EAPController/work':'rw,slave'
  -v '/mnt/user/appdata/omada-controller/logs/':'/opt/tplink/EAPController/logs':'rw,slave'
  -e SHOW_MONGODB_LOGS=true 'mbentley/omada-controller'
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'Europe/Paris'
INFO: Setting 'manage.http.port' to 18088 in omada.properties
INFO: Setting 'manage.https.port' to 18043 in omada.properties
INFO: Setting 'portal.http.port' to 18089 in omada.properties
INFO: Setting 'portal.https.port' to 18843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
05-22-2023 01:06:44.377 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
05-22-2023 01:06:44.387 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
05-22-2023 01:06:44.393 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
05-22-2023 01:06:44.397 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
05-22-2023 01:06:44.405 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
05-22-2023 01:06:44.407 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
05-22-2023 01:06:44.451 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
2023-05-22T01:06:44.799+0200 I CONTROL  [main] ***** SERVER RESTARTED *****
2023-05-22T01:06:44.809+0200 I CONTROL  [initandlisten] MongoDB starting : pid=149 port=27217 dbpath=../data/db 64-bit host=unraid
[... 245 lines cut ...]
        at org.springframework.data.mongodb.repository.support.MongoRepositoryFactory.getTargetRepository(MongoRepositoryFactory.java:134) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
        at org.springframework.data.repository.core.support.RepositoryFactorySupport.getRepository(RepositoryFactorySupport.java:324) ~[spring-data-commons-2.6.3.jar:2.6.3]
        at org.springframework.data.repository.core.support.RepositoryFactoryBeanSupport.lambda$afterPropertiesSet$5(RepositoryFactoryBeanSupport.java:322) ~[spring-data-commons-2.6.3.jar:2.6.3]
        at org.springframework.data.util.Lazy.getNullable(Lazy.java:230) ~[spring-data-commons-2.6.3.jar:2.6.3]
        at org.springframework.data.util.Lazy.get(Lazy.java:114) ~[spring-data-commons-2.6.3.jar:2.6.3]
        at org.springframework.data.repository.core.support.RepositoryFactoryBeanSupport.afterPropertiesSet(RepositoryFactoryBeanSupport.java:328) ~[spring-data-commons-2.6.3.jar:2.6.3]
        at org.springframework.data.mongodb.repository.support.MongoRepositoryFactoryBean.afterPropertiesSet(MongoRepositoryFactoryBean.java:119) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.invokeInitMethods(AbstractAutowireCapableBeanFactory.java:1863) ~[spring-beans-5.3.18.jar:5.3.18]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1800) ~[spring-beans-5.3.18.jar:5.3.18]
        ... 20 more
05-22-2023 01:12:49.459 ERROR [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): Cannot start up springboot. reson:Error creating bean with name 'clientMongoCrudRepository' defined in com.tplink.smb.omada.client.port.mongo.omada.client.ClientMongoCrudRepository defined in @EnableMongoRepositories declared on MaintainClientOmadaRepoConfig: Invocation of init method failed; nested exception is org.springframework.dao.DataAccessResourceFailureException: Timed out after 30000 ms while waiting to connect. Client view of cluster state is {type=UNKNOWN, servers=[{address=127.0.0.1:27217, type=UNKNOWN, state=CONNECTING, exception={com.mongodb.MongoSocketOpenException: Exception opening socket}, caused by {java.net.ConnectException: Connection refused}}]; nested exception is com.mongodb.MongoTimeoutException: Timed out after 30000 ms while waiting to connect. Client view of cluster state is {type=UNKNOWN, servers=[{address=127.0.0.1:27217, type=UNKNOWN, state=CONNECTING, exception={com.mongodb.MongoSocketOpenException: Exception opening socket}, caused by {java.net.ConnectException: Connection refused}}]
05-22-2023 01:12:49.459 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): Try to reset default certificate and restart springBoot.

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)

05-22-2023 01:12:49.525 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.9.31 using Java 17.0.7 on unraid with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.9.31.jar started by omada in /opt/tplink/EAPController/lib)
05-22-2023 01:12:49.525 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
05-22-2023 01:12:52.916 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
```

mbentley, 2023-05-22:
Did a quick search for "Fatal Assertion 34433" and I'm [seeing](https://jira.mongodb.org/browse/SERVER-32199) that it's likely indicating data corruption of the db files.  Looks like others have said that repairing the db isn't likely a good option.

Any idea if there are auto-updates happening on the container for you?  Typically, db corruption is where MongoDB is getting killed before it can be shut down so it's [best to make sure that whatever is stopping containers is allowing it to cleanly shut down the container](https://github.com/mbentley/docker-omada-controller/blob/master/KNOWN_ISSUES.md#mongodb-corruption).

I hope you have backups or snapshots that you could restore.  The controller can take backups but I think you have to configure them where it would back up a config to the persistent data store.

Termibg22, 2023-05-22:
I have reinstalled it. Didn't have any backup but now i'll start to make them xD

closed 2023-05-22 by Termibg22