mbentley/docker-omada-controller #311: [Bug]: docker image keep restarting

sogood007, 2023-05-25:
### Controller Version

v5.9.31

### Describe the Bug

docker keep restarting 

### Expected Behavior

able to start

### Steps to Reproduce

It happens after I reboot the system.  I have tried to pull the latest image issue but it doesn't help 

### How You're Launching the Container

```plain
docker run -d   --name omada-controller   --restart unless-stopped   --net host   -e MANAGE_HTTP_PORT=8088   -e MANAGE_HTTPS_PORT=8043   -e PORTAL_HTTP_PORT=8088   -e PORTAL_HTTPS_PORT=8843   -e SHOW_SERVER_LOGS=true   -e SHOW_MONGODB_LOGS=false   -e SSL_CERT_NAME="tls.crt"   -e SSL_KEY_NAME="tls.key"   -e TZ=Etc/UTC   -v omada-data:/opt/tplink/EAPController/data   -v omada-work:/opt/tplink/EAPController/work   -v omada-logs:/opt/tplink/EAPController/logs   mbentley/omada-controller:latest
```

### Container Logs

```plain
INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
05-25-2023 02:08:49.113 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
05-25-2023 02:08:49.125 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
05-25-2023 02:08:49.131 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
05-25-2023 02:08:49.137 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
05-25-2023 02:08:49.144 INFO [main] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
05-25-2023 02:08:49.146 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
05-25-2023 02:08:49.187 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
05-25-2023 02:08:53.229 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 117
05-25-2023 02:08:53.233 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
05-25-2023 02:08:53.234 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
05-25-2023 02:08:53.907 ERROR [main] [] c.t.s.o.s.a(): Port 8088 is already in use.
05-25-2023 02:08:53.913 ERROR [main] [] c.t.s.o.s.a(): Port 8088 is already in use. Release the port and try again.
05-25-2023 02:08:53.913 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Port 8088 is already in use. Release the port and try again.
05-25-2023 02:08:53.913 ERROR [main] [] c.t.s.o.s.OmadaBootstrap(): Environment is not ok for controller running
05-25-2023 02:08:53.913 ERROR [main] [] c.t.s.o.s.OmadaLinuxMain(): Bootstrap prepare failed, going to exit.
05-25-2023 02:08:53.918 WARN [Thread-0] [] c.t.s.o.c.s.a(): Spring Bean Factory has not been initialed, failed to get bean class com.tplink.smb.omada.common.concurrent.thread.CommonExecutorProperties
05-25-2023 02:08:53.921 WARN [Thread-0] [] c.t.s.o.c.s.a(): Spring Bean Factory has not been initialed, failed to get bean class io.micrometer.core.instrument.MeterRegistry
Exception in thread "Thread-0" java.lang.ExceptionInInitializerError
	at com.tplink.smb.omada.common.concurrent.thread.a.a(SourceFile:85)
	at com.tplink.smb.omada.common.concurrent.thread.a.c(SourceFile:303)
	at com.tplink.smb.omada.starter.OmadaBootstrap.b(SourceFile:191)
	at com.tplink.smb.omada.starter.OmadaLinuxMain.b(SourceFile:87)
	at java.base/java.lang.Thread.run(Thread.java:833)
Caused by: java.lang.NullPointerException: Cannot invoke "com.tplink.smb.omada.common.concurrent.thread.CommonExecutorProperties.getCommCoreThreads()" because "com.tplink.smb.omada.common.concurrent.thread.a$b.a" is null
	at com.tplink.smb.omada.common.concurrent.thread.a$b.<clinit>(SourceFile:72)
	... 5 more
```

sogood007, 2023-05-25:
I just notice there is some container in the system using 8088.  Maybe that is causing issue. Let me try remap the port

mbentley, 2023-05-25:
You have a port conflict. What’s running on 8088?  Use `netstat -ltnpu` to find out.

sogood007, 2023-05-25:
tcp        0      0 0.0.0.0:8088            0.0.0.0:*               LISTEN      5723/docker-proxy   
tcp6       0      0 :::8088                 :::*                    LISTEN      5731/docker-proxy   

great command and quick response.  I was using just `netstat` which I can't find it.  I didn't know the extra options.  I notice the port using after looking at `docker ps`   Is that safe just to change 8088 from the docker run to something else

sogood007, 2023-05-25:
I tried this replace all 8088 to 8089 but I still have the same port conflict error 

``` 
docker run -d   --name omada-controller   --restart unless-stopped   --net host   -e MANAGE_HTTP_PORT=8089   -e MANAGE_HTTPS_PORT=8043   -e PORTAL_HTTP_PORT=8089   -e PORTAL_HTTPS_PORT=8843   -e SHOW_SERVER_LOGS=true   -e SHOW_MONGODB_LOGS=false   -e SSL_CERT_NAME="tls.crt"   -e SSL_KEY_NAME="tls.key"   -e TZ=Etc/UTC   -v omada-data:/opt/tplink/EAPController/data   -v omada-work:/opt/tplink/EAPController/work   -v omada-logs:/opt/tplink/EAPController/logs   mbentley/omada-controller:latest
```

sogood007, 2023-05-25:
I currently workaround by remap the other docker image which is using the 8088 to something else.  so I go the omada run again.  Still curious to see what's wrong with my command which change 8088 to 8089.  Thanks in advance.

mbentley, 2023-05-25:
You can't just remap the port, you need to change it inside the container as well due to how the controller software was written by tp-link.  See [this part of the readme](https://github.com/mbentley/docker-omada-controller#using-non-default-ports) for how to change ports.

closed 2023-05-27 by sogood007