mbentley/docker-omada-controller #322: [Feature]: Run the controller as root in rootless podman

andye2004, 2023-06-18:
### What problem are you looking to solve?

More easily run the image as root in rootless podman.

### Describe the solution that you have in mind

Have the entrypoint.sh script modified to find the actual username / groupname from the mounted /etc/passwd and /etc/group files instead of forcing the use of the 'omada' user and group name.

### Additional Context

I have tried a number of different ways to run root in rootless but unless I name the use 'omada' then I run into issues caused by the entrypoint.sh checks around the user/group ID and user being named differently. This is mainly due to the fact I want to run all my containers in the same user namespace, which means all my containers would need to run as the omada user - something I'm really not very keen on.

mbentley, 2023-06-18:
Just curious - if I turned the username and group into variables so that you could override them like with `PUSERNAME=root` `PGROUP=root`, would that solve the issue that you're hitting?  I'd have to check through the entrypoint to see if if there is anything that would also be an issue related to using `root:root` and `0:0` but that could be a quick and simple change for the most part.

mbentley, 2023-06-18:
Did a quick test and built a test `amd64` (assuming you're running amd64) image as `mbentley/omada-controller:5.9-322` if you want to try it out.  It has the additional variables of `PUSERNAME` and `PGROUP` (can always change these to something else if there is a better idea) which you should be able to set to `root` along with the existing `PUID` and `PGID` variables.

andye2004, 2023-06-18:
Wow, what what a fantastic response! It's late in the day for me here and was about to turn in but I'll give this a quick go and come back.

andye2004, 2023-06-18:
I'll test this more fully tomorrow but the change appears to work well. I started with the following command:-
```shell
podman run -it \
  -e PUID=0 \
  -e PGID=0 \
  -e PGROUP=root \
  -e PUSERNAME=root \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 8843:8843 \
  -p 27001:27001/udp \
  -p 29810:29810/udp \
  -p 29811-29816:29811-29816 \
  -v /path/to/omada/data:/opt/tplink/EAPController/data \
  -v /path/to/omada/logs:/opt/tplink/EAPController/logs \
  docker.io/mbentley/omada-controller:5.9-322
```
Running the above, I can see files created on the host system correctly with chosen username and I can access the omada controller in the browser.

Many, many thanks for all your hardwork, and specifically on looking into this one so quickly.

andye2004, 2023-06-19:
Really happy to confirm that this is working exactly as expected with the provided changes. Many thanks once again!

mbentley, 2023-06-19:
No problem - I'm just happy to be able to give something back.  Thanks for testing and I always appreciate feedback on how people might want to use any image I have.

commit 371f5734 (2023-06-19): Merge pull request #323 from mbentley/add-puser-and-group

Add pusername and group; fixes #322
closed 2023-06-19 by mbentley