mbentley/docker-omada-controller #325: [Bug]: crash on container start

rpatel3001, 2023-08-05:
### Controller Version

v5.9.31

### Describe the Bug

I am trying to migrate my omada controller from a VM to an LXC on proxmox. I have tried this with the bind mount directories copied from the VM and with fresh/empty directories. The compose file below and docker image are the same in both places, except the PUID/GUID is 1000 in the VM (but the container `chown`s the files correctly to 508 during startup). 

The container runs to the end of the logs below, hangs for a few seconds, then exits and restarts.

### Expected Behavior

successful container start

### Steps to Reproduce

1. run `docker compose up -d omada`

### How You're Launching the Container

```plain
omada:
    container_name: omada
    image: mbentley/omada-controller
    restart: unless-stopped
    network_mode: host
    environment:
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - PGID=508
      - PUID=508
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro
      - ./omada/data:/opt/tplink/EAPController/data
      - ./omada/logs:/opt/tplink/EAPController/logs
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'Etc/UTC'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8043 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'Etc/UTC'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8043 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Version check passed; image version (5.9.31) >= the last version ran (5.9.31); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
08-05-2023 20:34:19.535 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
08-05-2023 20:34:19.546 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
08-05-2023 20:34:19.556 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
08-05-2023 20:34:19.564 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
08-05-2023 20:34:19.570 INFO [main] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
08-05-2023 20:34:19.574 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
08-05-2023 20:34:19.635 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
08-05-2023 20:34:24.171 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 119
08-05-2023 20:34:24.174 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
08-05-2023 20:34:24.175 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
08-05-2023 20:34:25.210 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
08-05-2023 20:34:25.210 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
08-05-2023 20:34:25.314 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
08-05-2023 20:34:25.315 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)

08-05-2023 20:34:26.234 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.9.31 using Java 17.0.8 on corelxc with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.9.31.jar started by omada in /opt/tplink/EAPController/lib)
08-05-2023 20:34:26.250 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
```

mbentley, 2023-08-05:
If you want to run with `1000:1000`, you can use the `PUID` & `PGID` environment variables to change the default UID/GID.  You're likely hitting MongoDB data that is for one version of MongoDB but the container is running another.  The easiest way to migrate is to start up your VM, take a backup from the controller web interface, stop your VM, start up a container with your persistent data wiped out, and then import your backup otherwise you're likely going to be needing to do database migrations between versions which isn't so fun.  You should be able to check the `mongod.log` log file in `./omada/logs` and see what's failing though if you want.

rpatel3001, 2023-08-06:
Sorry if I wasn't clear, I've tried to start the container with my data wiped, with the data and logs directories empty, and it still fails to start, so I can't try restoring from a backup.

The mongo server seems to be starting fine (and the docker image hashes match so there shouldn't be a mismatch).

The mongo and server log files don't shows anything weird, they just cut off where the docker log does.

rpatel3001, 2023-08-06:
i just tried to bring it up in a different lxc and oddly enough it worked. could a port conflict cause a crash like this?

mbentley, 2023-08-06:
Ah, sorry, missed that part.  Odd that it is not showing anything relevant in the logs anywhere.  If you set the container to not auto-restart, can you see what the exit code of the container is from `docker ps`?  If it was an issue of hitting ulimits, I'd expect to see something like what was mentioned in https://github.com/mbentley/docker-omada-controller/issues/298 so I don't think it's that.  It is possible there is a port conflict - not totally sure what the behavior looks like when that happens but checking `netstat -ltnpu` from the host to see if there are port conflicts would be a good idea.

rpatel3001, 2023-08-06:
exit code was 137. one thing i forgot is that the VM had the container on a macvlan network whereas the LXC is host mode. No ports seemed to overlap, and even changing back to macvlan doesn't help in the failing LXC.

rpatel3001, 2023-08-06:
turns out to be a memory issue. I turned off a few RAM hungry node containers and it progressed further but still failed. I increased the LXC allocation from 1G to 2G and everything worked fine.

closed 2023-08-06 by rpatel3001