mbentley/docker-omada-controller #326: [Bug]: Migration

vjwizkid6, 2023-08-09:
### Controller Version

5.9.31

### Describe the Bug

Hi trying to work out the migration off the docker controller on to a newOC200 but cant for the life of me find the Migration tool in the docker controller. 

### Expected Behavior

To be able to run the Migration tool

### Steps to Reproduce

go to settinng and find the Migration tool

### How You're Launching the Container

```plain
na
```

### Container Logs

```plain
na
```

mbentley, 2023-08-09:
Does the controller backup and restore not work?

vjwizkid6, 2023-08-09:
Cant find a backup option. Where should it be

mbentley, 2023-08-09:
I take that back - you have two options.  Backup and Restore or use the built in migration wizard.

## Migration

Settings > Migration and use the `Controller Migration` option.  I've never used this so I don't know anything after that.

## Backup and Restore

Settings > Maintenance and it's right at the top.

vjwizkid6, 2023-08-09:
Thanks dont seam to have that option. I have a hunch that i am not lodged on as the master admin. Just need to work out what that was set to.

vjwizkid6, 2023-08-10:
So i wasn't loged on as admin. All sorted now thanks for the help.

closed 2023-08-10 by vjwizkid6