mbentley/docker-omada-controller #328: [Feature]: New Version 5.12.6-Beta Available

saladm2, 2023-09-04:
### What problem are you looking to solve?

A new early access/beta version of the omada software controller is available (v5.12.6)

### Describe the solution that you have in mind

Can this be incorporated into a docker release? It's the first update since v5.9.31 about 6 months ago and brings many new features. The download link below was taken from the official community forums @ https://community.tp-link.com/en/business/forum/topic/619304

### Additional Context

https://static.tp-link.com/upload/beta/2023/202309/20230904/Omada_SDN_Controller_v5.12.6_Linux_x64.tar.gz(Beta).zip

closed 2023-09-04 by mbentley

mbentley, 2023-09-04:
Beta builds are up: https://hub.docker.com/repository/docker/mbentley/omada-controller/tags?page=1&ordering=last_updated&name=beta

I have done very little testing - only validated that the controller basically starts and that's about it.

saladm2, 2023-09-04:
Awesome, thank you! So then I'd just pull mbentley/omada-controller:beta instead of mbentley/omada-controller:latest?

mbentley, 2023-09-04:
exactly.

Shurov, 2023-09-16:
thanks, switched to beta for now (want new features :D )
am I right, understanding, that 5.11 is also not in stable due to time constraints? if so, can we send our reports of working (or non-working) features, confirming the stability to unblock propagation to the stable branch?

J4CE2, 2023-09-16:
No, tplink have jumped straight to 5.12.x as early release / beta. There is no 5.11 available from tplink.

mbentley, 2023-09-16:
Their versioning is quite the mystery as they seem to increment the version numbers regularly internally without releasing a version externally - almost like every feature they add bumps the version until they're ready for a public release.

commit 2dd13050 (2023-09-04): Added beta builds; fixes #328
commit 7b7cc5f2 (2023-09-04): Merge pull request #329 from mbentley/add-beta

Added beta builds; fixes #328