mbentley/docker-omada-controller #340: [Bug]: Restart loop on latest version

mishac, 2023-09-20:
### Controller Version

5.9

### Describe the Bug

The lastest version of the container causes a container restart loop:

    INFO: Group (omada) exists; skipping creation
    INFO: User (omada) exists; skipping creation
    INFO: Time zone set to 'America/New_York'
    INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
    INFO: Value of 'manage.https.port' already set to 8443 in omada.properties
    INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
    INFO: Value of 'portal.https.port' already set to 8443 in omada.properties
    INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
    INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
    INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
    INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
    INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
    INFO: Value of 'port.discovery' already set to 29810 in omada.properties

And then it dies and restarts ... Given the port numbers shown I assume it has to do something with https://github.com/mbentley/docker-omada-controller/pull/339 ?

Reverting to 5.9-322 fixes it. 

### Expected Behavior

Not restarting.

### Steps to Reproduce

1. Update container
2. observe restart

### How You're Launching the Container

```plain
Docker compose.

    networks:
        internal:
            internal: true
        traefik_web:
            name: traefik_web
            external: true
    services:
        omada:
            container_name: omada
            environment:
                MANAGE_HTTPS_PORT: 8443
                MANAGE_HTTP_PORT: 8088
                PGID: 3200
                PORTAL_HTTPS_PORT: 8443
                PORTAL_HTTP_PORT: 8088
                PUID: 3200
                TZ: America/New_York
            image: mbentley/omada-controller:5.9-322
            labels:
                traefik.enable: true
                traefik.http.middlewares.omada-headers.headers.customrequestheaders.Host: omada.internal.example.com:8443
                traefik.http.middlewares.omada-headers.headers.customresponseheaders.Host: omada.internal.example.com
                traefik.http.middlewares.omada-middlewares.chain.middlewares: omada-redirect,omada-headers
                traefik.http.middlewares.omada-redirect.redirectregex.regex: ^https:\/\/([^\/]+)\/?$$
                traefik.http.middlewares.omada-redirect.redirectregex.replacement: https://$$1/{{REDACTED}}/login
                traefik.http.routers.omada.entrypoints: https
                traefik.http.routers.omada.middlewares: omada-middlewares
                traefik.http.routers.omada.rule: Host(`omada.internal.example.com`)
                traefik.http.routers.omada.service: omada
                traefik.http.routers.omada.tls.certresolver: http
                traefik.http.services.omada.loadbalancer.server.port: 8443
                traefik.http.services.omada.loadbalancer.server.scheme: https
            networks:
            - internal
            - traefik_web
            ports:
            - 27001:27001/udp
            - 29810:29810/udp
            - 29811-29814:29811-29814
            - 8088:8088/tcp
            - 8443:8443/tcp
            - 8843:8843
            restart: always
            volumes:
            - /config/omada/data:/opt/tplink/EAPController/data
            - /config/omada/work:/opt/tplink/EAPController/work
            - /config/omada/logs:/opt/tplink/EAPController/logs
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (3200:3200)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8443 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8443 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Validating user/group (omada:omada) exists with correct UID/GID (3200:3200)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8443 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8443 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Validating user/group (omada:omada) exists with correct UID/GID (3200:3200)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8443 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8443 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Validating user/group (omada:omada) exists with correct UID/GID (3200:3200)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8443 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties

etc etc
```

mishac, 2023-09-20:
Changing 29811-29814 to 29811-29816 in my compose didnt fix the issue.

mishac, 2023-09-20:
updating to 5.12 did fix the issue.

There was some reason I was still on 5.9 but I can't really remember what it was.... so fixed for me I guess, but for others still on 5.9 it's broken.

mbentley, 2023-09-20:
Ah you're right - the parsing of the properties file in the entrypoint is causing the issue because the property doesn't exist.  I'll have to take that into account.

closed 2023-12-10 by mbentley