mbentley/docker-omada-controller #342: [Bug]: Controller restarting

faptaincrunch, 2023-09-21:
### Controller Version

5.12.7

### Describe the Bug

After updating, the omada phone app no longer connects and the webUI will let me poke around for 30 seconds before bringing me back to the login screen. 

### Expected Behavior

Not restarting.

### Steps to Reproduce

1.Start docker
2. Check logs for loop

### How You're Launching the Container

```plain
Template through unraid.
```

### Container Logs

```plain
09-21-2023 06:20:25.509 INFO [main] [] c.t.c.c.a.c.a.a.AccountingConfig(): Success to init radius accounting handler, num = 0, invokes = {}
09-21-2023 06:20:25.527 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(): log mq consume task is start...
09-21-2023 06:20:25.529 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Started OmadaLinuxMain in 18.245 seconds (JVM running for 21.183)
09-21-2023 06:20:25.547 INFO [main] [] c.t.s.o.m.d.p.b.a(): manager maintenance Handling event: org.springframework.boot.context.event.ApplicationStartedEvent[source=org.springframework.boot.SpringApplication@1273afe9]
09-21-2023 06:20:25.612 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask finished
09-21-2023 06:20:25.612 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before OmadacInitTask
09-21-2023 06:20:25.622 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before init bean
09-21-2023 06:20:25.629 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: after init bean
09-21-2023 06:20:25.662 INFO [main] [] c.t.s.o.s.d.a(): Db unchanged. No need to compatible or init data.
09-21-2023 06:20:25.691 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): succeed get default omadac OmadacVO(id=b96d8bd26656eb851f92744c7c536419, name=Omada Controller_MPFDigital, pattern=null)
09-21-2023 06:20:25.708 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before init for hwc
09-21-2023 06:20:25.708 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: after init for hwc
09-21-2023 06:20:25.775 INFO [main] [] c.t.s.o.c.c.p.c.o.s.ActiveSiteCacheImpl(): Scheduled ActiveSitesCache period flush buf task at fixed rate of 30000 millis.
09-21-2023 06:20:25.904 INFO [main] [] c.t.s.o.d.g.c.d.ThreadConfiguration(): device-gateway datatrack workGroup core thread  num is 24, max thread num is 24
09-21-2023 06:20:26.052 INFO [main] [] c.t.s.o.p.p.r.a.c(): init nioEventLoopGroup
09-21-2023 06:20:26.336 INFO [main] [] c.t.s.o.d.g.c.f.b.c(): file download mq consume task is start...
09-21-2023 06:20:26.357 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): adopt workGroup core thread num is 8, max thread num is8
09-21-2023 06:20:26.484 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): discovery workGroup core thread num is 2, max thread num is 10
09-21-2023 06:20:27.383 INFO [main] [] c.t.s.o.d.g.c.d.ThreadConfiguration(): device-gateway datatrack time task workGroup core thread  num is 24, max thread num is 24
09-21-2023 06:20:27.759 INFO [comm-pool-2] [] c.t.s.o.s.t.OmadacInitTask(): Starting collection maintenance.
09-21-2023 06:20:27.767 INFO [main] [] c.t.s.o.p.b.b.d(): DisconnectRequestServer start
09-21-2023 06:20:27.906 INFO [main] [] c.t.s.o.s.d.a(): record: after OmadacInitTask
09-21-2023 06:20:27.906 INFO [main] [] c.t.s.o.s.t.CloudStartUpTask(): record: CloudStartUpTask start
09-21-2023 06:20:27.906 INFO [main] [] c.t.s.o.s.t.CloudStartUpTask(): record: CloudStartUpTask finished
09-21-2023 06:20:27.906 INFO [main] [] c.t.s.o.s.t.OmadacDiscoveryStartUpTask(): record: OmadacDiscoveryStartUpTask start
09-21-2023 06:20:27.907 INFO [main] [] c.t.s.o.s.t.OmadacDiscoveryStartUpTask(): record: OmadacDiscoveryStartUpTask finished
09-21-2023 06:20:27.996 INFO [main] [] c.t.s.o.d.g.a(): startServers, DISCOVERY host:null.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): discovery Port is 29810.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): manage    port v1 is 29811.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): manage    port v2 is 29814.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): adopt     port v1 is 29812.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): upgrade   port is 29813.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): transfer  port is 29815.
09-21-2023 06:20:27.999 INFO [main] [] c.t.s.o.d.g.a(): rtty port is 29816.
09-21-2023 06:20:28.030 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyUdpServer bind /0.0.0.0:29810, export localhost/127.0.0.1:29810
09-21-2023 06:20:28.040 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29812 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
09-21-2023 06:20:28.044 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
09-21-2023 06:20:28.046 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
09-21-2023 06:20:28.046 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
09-21-2023 06:20:28.048 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
09-21-2023 06:20:28.049 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
09-21-2023 06:20:28.051 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
09-21-2023 06:20:28.052 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
09-21-2023 06:20:28.053 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29815 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
[... 22 lines cut ...]
09-21-2023 06:20:49.568 INFO [manage-work-group-7] [] c.t.s.o.m.d.d.m.i.e(): first inform no need send full config to OmadacId(b96d8bd26656eb851f92744c7c536419) DeviceMac(C0-C9-E3-CB-FF-73)
09-21-2023 06:20:49.569 WARN [manage-work-group-7] [] c.t.s.o.m.d.d.m.m.s.e(): fill set msg body but all setting null, skip send set msg to OmadacId(b96d8bd26656eb851f92744c7c536419) DeviceMac(C0-C9-E3-CB-FF-73)
09-21-2023 06:20:49.569 WARN [manage-work-group-7] [] c.t.s.o.m.d.d.m.m.s.e(): failed to genConfigBodyAndAddVersion with keys:[]
09-21-2023 06:20:49.569 INFO [manage-work-group-7] [] c.t.s.o.m.d.d.m.d.c.C(): syncConfigurationForSameVersion to OmadacId(b96d8bd26656eb851f92744c7c536419) DeviceMac(C0-C9-E3-CB-FF-73), result:SendDeviceMsgResult(success=false, deviceResponse=null)
09-21-2023 06:21:06.679 WARN [https-jsse-nio-8043-exec-5] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
09-21-2023 06:21:06.690 WARN [https-jsse-nio-8043-exec-5] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
09-21-2023 06:21:19.624 INFO [server-comm-pool-20] [] c.t.s.e.s.a.b.e(): serverType from channel: /192.168.5.10:29814 -> /192.168.5.4:37808 is MANAGE
09-21-2023 06:21:20.104 INFO [tcp-message-executor-17-2] [] c.t.s.e.s.c.c(): need not set server route expire for device C0-C9-E3-CB-FF-73, as device reconnect: /192.168.5.10:29814 -> /192.168.5.4:37808
09-21-2023 06:21:20.127 INFO [manage-work-group-12] [] c.t.s.o.m.d.p.t.c(): Device C0-C9-E3-CB-FF-73 OmadacId(b96d8bd26656eb851f92744c7c536419) changed to status CONNECTED, which don't need to handle.
09-21-2023 06:21:20.134 INFO [manage-work-group-13] [] c.t.s.o.m.d.d.m.r.d(): send v2 rebuild reply[reset=false] to omadacId OmadacId(b96d8bd26656eb851f92744c7c536419) & mac DeviceMac(C0-C9-E3-CB-FF-73)
09-21-2023 06:21:27.934 INFO [monitor-topology-pool-0] [] c.t.s.o.c.u.e.a(): list local interface macs: [C2-22-31-D3-39-34, 52-08-06-BA-01-AB, C2-1F-E8-13-7C-03, DE-50-0D-72-ED-38, 96-EA-27-FB-83-72, EE-DD-0A-B8-3F-0E, EE-36-BF-01-AF-96, E2-5C-05-2B-63-95, 02-42-3 [... cut ...]
09-21-2023 06:28:27.910 INFO [monitor-topology-pool-0] [] c.t.s.o.c.u.e.a(): list local interface macs: [C2-22-31-D3-39-34, 52-08-06-BA-01-AB, C2-1F-E8-13-7C-03, DE-50-0D-72-ED-38, 96-EA-27-FB-83-72, EE-DD-0A-B8-3F-0E, EE-36-BF-01-AF-96, E2-5C-05-2B-63-95, 02-42-3C-69-D1-7F, 02-42-18-1D-60-32, 02-42-7B-0D-68-79, 60-45-CB-62-F2-95]
09-21-2023 06:29:57.899 WARN [quartzScheduler_Worker-2] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (voucher_clear_task, null) but did not execute because corresponding handler voucher_clear_task doesn't exist
```

mbentley, 2023-09-21:
#340 was related to the entrypoint trying to read values from the properties file that didn't exist before the controller itself would start so this is a different issue.  In your logs, I don't see where the controller is restarting.  Do you have logs from where that was happening (assuming it was actually restarting)?  It almost sounds like you're just getting logged out of the web interface for some reason.

What browser/OS are you using?  Have you tried another browser?  Odd that you wouldn't be able to login through the web app.  I am not seeing problems with either the web interface or the iOS app here.  Is there a specific page you're going to when you have it kick you back to the login page?

faptaincrunch, 2023-09-21:
Hm I'd only assumed it was rebooting since the webui would bounce me back to the login. I noticed it only bounces me back when looking at the logs within the controller, but the android app will no longer connecting after updating the container yesterday. Tried the webUI on different browsers and computer with same behavior. 

Controller is also extremely slow before it bounces me. Was working fine before the update. Is it possible to downgrade the docker version to test? I noticed no specific tags within 5.12 to try.

mbentley, 2023-09-21:
No, you can't downgrade without having a backup as the controller does database migrations (and attempting to downgrade will cause it to fail).

As of late, TP-Link hasn't provided very many bugfix versions of any minor versions and just keeping the latest major.minor versions up to date is a pretty significant effort since I build images for amd64, arm64, and armv7l, even with automation in place.  Typically when they release a new version, they've moved to the next minor version.

Here is the list of all of the versions they have published today:

```
https://static.tp-link.com/upload/software/2023/202309/20230920/Omada_SDN_Controller_v5.12.7_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2023/202303/20230321/Omada_SDN_Controller_v5.9.31_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2023/202302/20230227/Omada_SDN_Controller_v5.9.9_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2023/202301/20230130/Omada_SDN_Controller_v5.8.4_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202211/20221121/Omada_SDN_Controller_v5.7.4_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202210/20221024/Omada_SDN_Controller_v5.6.3_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202208/20220822/Omada_SDN_Controller_v5.5.6_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202207/20220729/Omada_SDN_Controller_v5.4.6_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202205/20220507/Omada_SDN_Controller_v5.3.1_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202203/20220322/Omada_SDN_Controller_v5.1.7_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202201/20220120/Omada_SDN_Controller_v5.0.30_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2021/202112/20211217/Omada_SDN_Controller_v4.4.8_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2021/202110/20211011/Omada_SDN_Controller_v4.4.6_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2021/202108/20210813/Omada_SDN_Controller_v4.4.4_linux_x64.tar.gz
https://static.tp-link.com/software/2021/202107/20210701/Omada_SDN_Controller_v4.4.3_linux_x64.tar.gz
https://static.tp-link.com/2021/202105/20210507/Omada_SDN_Controller_v4.3.5_linux_x64.tar.gz
https://static.tp-link.com/2021/202102/20210209/Omada_SDN_Controller_v4.2.11_linux_x64.tar.gz
https://static.tp-link.com/2020/202012/20201211/Omada_SDN_Controller_v4.2.8_linux_x64.tar.gz
https://static.tp-link.com/2020/202011/20201110/Omada_SDN_Controller_v4.2.4_linux_x64.tar.gz
https://static.tp-link.com/2020/202008/20200805/Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202208/20220818/Omada_Controller_v3.2.17_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202201/20220125/Omada_Controller_v3.2.16_linux_x64.tar.gz
https://static.tp-link.com/2020/202012/20201225/Omada_Controller_v3.2.14_linux_x64.tar.gz
https://static.tp-link.com/2020/202004/20200420/Omada_Controller_v3.2.10_linux_x64.tar.gz
https://static.tp-link.com/2020/202004/20200409/Omada_Controller_v3.2.9_linux_x64.tar.gz
https://static.tp-link.com/2020/202001/20200116/Omada_Controller_v3.2.6_linux_x64.tar.gz
https://static.tp-link.com/2019/201911/20191108/Omada_Controller_v3.2.4_linux_x64.tar.gz
```

5.9.x was one of the few instances of them releasing multiple bugfix versions since they released 5.x.

In addition, it might be worth looking at the `mongod.log` in your logs persistent volume to see if MongoDB is spitting out any errors/warnings.

noizo, 2023-09-21:
I have something similar after upgrade from 5.9 to 5.12.
```
09-21-2023 19:17:05.522 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(4edc12bb62072520cb6301c815be2dab).
09-21-2023 19:17:06.558 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
09-21-2023 19:17:06.560 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
09-21-2023 19:17:06.773 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
09-21-2023 19:17:09.212 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.d.a(): OmadacId(4edc12bb62072520cb6301c815be2dab) DeviceMac(28-87-BA-87-2A-A3) receive discovery when adopting, times[3] reach limit, set adopt fail
09-21-2023 19:17:55.499 INFO [https-jsse-nio-8043-exec-16] [] c.t.s.o.a.a.i.h.HealthInternalAdapterConfiguration(): NoSupport internal service for anomaly
09-21-2023 19:18:08.123 WARN [https-jsse-nio-8043-exec-9] [] c.t.s.o.a.d.u.WebUtils(): sessionCSRFToken is null for request /img/svg/PaginationLeftDisabled.svg
09-21-2023 19:18:15.564 WARN [https-jsse-nio-8043-exec-1] [] c.t.s.o.a.d.u.WebUtils(): sessionCSRFToken is null for request /img/svg/PaginationLeftDisabled.svg
09-21-2023 19:18:33.565 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.d.a(): OmadacId(4edc12bb62072520cb6301c815be2dab) DeviceMac(28-87-BA-87-2A-A3) receive discovery when adopting, times[3] reach limit, set adopt fail
09-21-2023 19:18:36.049 WARN [https-jsse-nio-8043-exec-9] [] c.t.s.o.a.d.u.WebUtils(): sessionCSRFToken is null for request /img/svg/PaginationLeftDisabled.svg
09-21-2023 19:20:18.257 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.d.a(): OmadacId(4edc12bb62072520cb6301c815be2dab) DeviceMac(28-87-BA-87-2A-A3) receive discovery when adopting, times[3] reach limit, set adopt fail
```

1. My main router is ER605 and it's in constant adoption loop after this update.
2. Whenever i'm trying to access log tab, i'm being logged out from omada. And logs are crying about `09-21-2023 19:18:36.049 WARN [https-jsse-nio-8043-exec-9] [] c.t.s.o.a.d.u.WebUtils(): sessionCSRFToken is null for request /img/svg/PaginationLeftDisabled.svg`

Aevrin, 2023-09-21:
I updated to the latest version, and it keeps rejecting my password.  Not sure if this matches anyone elses experience.

mbentley, 2023-09-21:
> I updated to the latest version, and it keeps rejecting my password. Not sure if this matches anyone elses experience.

So far, I have not seen anyone with that issue yet.

mbentley, 2023-09-21:
> sessionCSRFToken is null for request /img/svg/PaginationLeftDisabled.svg

This seems like something in common and it looks like it is a bug in the software.  Something is specifically calling for `/img/svg/PaginationLeftDisabled.svg` when it should be `/theme/img/svg/PaginationLeftDisabled.svg`.

So I can login and navigate around just fine but if I directly try to access `https://omada.casa.mbentley.net/img/svg/PaginationLeftDisabled.svg`, it kicks me back to the login page if I try to load anything else.

There must be a page in specific where the error is - **could either of you tell me exactly which page it is where you're experiencing this when it kicks you back to the login page?**

faptaincrunch, 2023-09-21:
> **could either of you tell me exactly which page it is where you're experiencing this when it kicks you back to the login page?**

The logs tab within the dashboard
[image]
d

Aevrin, 2023-09-21:
Mine won't let me log in.  And the initial response from the app hangs for a while, then fails.

[image]

mbentley, 2023-09-21:
@Aevrin - could you start a [new discussion](https://github.com/mbentley/docker-omada-controller/discussions/new?category=help) on this as it is unrelated?

Aevrin, 2023-09-21:
Sure, I had thought it might be related due to the app experience.

mbentley, 2023-09-25:
I see that there is a thread up on the TP-Link forums now: https://community.tp-link.com/en/business/forum/topic/623942 (thanks if one of you created it).  I also see that [someone on Reddit](https://www.reddit.com/r/TPLink_Omada/comments/16rb07h/admin_login_doesnt_work_after_updating_to_latest/k22qci5/) might have found a possible cause or at least the cause for one person.

mbentley, 2023-09-25:
One thought I had is that maybe something is cached in browser and they changed a URL but the cache still has it pointing to the previous URL which might trigger the behavior.  I can't trigger it myself just browsing the user interface, only when specifically going to a URL that will log me out as described above.  I created a [forum post](https://community.tp-link.com/en/business/forum/topic/624014?page=1) so hopefully someone will take a look.

Robert-litts, 2023-09-26:
Good Afternoon, I am also on 5.12.7 and am having similar reboot loop issue, my controller is no longer starting at all.

Does this look like a similar issue to #342?

``

INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
INFO: Version check passed; image version (5.12.7) >= the last version ran (5.12.7); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
09-26-2023 13:47:00.856 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
09-26-2023 13:47:00.863 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
09-26-2023 13:47:00.867 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
09-26-2023 13:47:00.875 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
09-26-2023 13:47:00.879 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
09-26-2023 13:47:00.881 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
09-26-2023 13:47:00.919 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
09-26-2023 13:47:02.958 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 155
09-26-2023 13:47:02.959 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
09-26-2023 13:47:02.960 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
09-26-2023 13:47:03.352 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
09-26-2023 13:47:03.352 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
09-26-2023 13:47:03.405 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
09-26-2023 13:47:03.405 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start

'
 :: Spring Boot ::                (v2.6.6)

A fatal error has been detected by the Java Runtime Environment:

SIGSEGV (0xb) at pc=0x00007f04410c1af5 (sent by kill), pid=1, tid=187

 JRE version: OpenJDK Runtime Environment (17.0.8.1+1) (build 17.0.8.1+1-Ubuntu-0ubuntu120.04)

Java VM: OpenJDK 64-Bit Server VM (17.0.8.1+1-Ubuntu-0ubuntu120.04, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)

Problematic frame:

J 905 c2 java.util.zip.ZipFile$Source.getEntryPos(Ljava/lang/String;Z)I java.base@17.0.8.1 (173 bytes) @ 0x00007f04410c1af5 [0x00007f04410c1860+0x0000000000000295]
 Core dump will be written. Default location: /opt/tplink/EAPController/lib/core

An error report file with more information is saved as:

/tmp/hs_err_pid1.log

Compiled method (c2)    6495  929       4       jdk.internal.util.jar.JarIndex::get (49 bytes)

 total in heap  [0x00007f04410c5210,0x00007f04410c73a0] = 8592

 relocation     [0x00007f04410c5370,0x00007f04410c5410] = 160

 main code      [0x00007f04410c5420,0x00007f04410c68c0] = 5280

 stub code      [0x00007f04410c68c0,0x00007f04410c68d8] = 24

 metadata       [0x00007f04410c68d8,0x00007f04410c6980] = 168

 scopes data    [0x00007f04410c6980,0x00007f04410c7030] = 1712

 scopes pcs     [0x00007f04410c7030,0x00007f04410c7310] = 736

 dependencies   [0x00007f04410c7310,0x00007f04410c7318] = 8

 handler table  [0x00007f04410c7318,0x00007f04410c7348] = 48

 nul chk table  [0x00007f04410c7348,0x00007f04410c73a0] = 88

 If you would like to submit a bug report, please visit:

  Unknown
'

mbentley, 2023-09-26:
This looks different @Robert-litts.  I would suggest filing another issue with details about the architecture, how you're starting the container, etc as the issue in #342 isn't the controller actually restarting, just the user session being reset from the interface.

esmoyer, 2023-09-29:
Just throwing my hat in here. App seems to access just fine, but web interface goes back to login screen as soon as a try to launch the site. Also, I noticed the container is slowly eating up disk space, I am assuming from logs; but it's like 1+GB per day. Shut the container down for now.

mbentley, 2023-10-08:
@faptaincrunch or @noizo - Are you all still having the same issue with your login sessions being cleared from viewing specific pages or has the problem gone away now?  The TP-Link support team has assigned an engineer to the ticket that was created from the forums.

JaCoB1123, 2023-10-08:
I'm still having the issue.

linuxd3v, 2023-10-18:
> So far, I have not seen anyone with that issue yet.

I just ran into this issue I think. Unable to login into software controller anymore since yesterday.   
Im on the latest version of dockerized omada controller, I think "controllerVer": "5.12.7".     

What happens: I enter my username and password to the web UI, I think it tries loading the web UI but then redirects me back to the login screen.

it logs following message in docker omada logs on every login attempt:   
```
omada-controller-app  | 10-18-2023 06:22:46.845 WARN [https-jsse-nio-8043-exec-1] [] c.t.s.o.a.d.u.WebUtils(): sessionCSRFToken is null for request /img/svg/PaginationLeftDisabled.svg
```

mbentley, 2023-10-19:
The only thing that makes sense to me would be something that is cached in browser that eventually gets cleared but I don't really have any evidence to back that up.

linuxd3v, 2023-10-19:
> The only thing that makes sense to me would be something that is cached in browser that eventually gets cleared but I don't really have any evidence to back that up.

yeah it started working again for me later in the same day. I defintelay tried clearing browser cache, differnet browsers and restarting docker OC container N-times.  But it started workign again I dont know why - maybe me enabling the cloud access through app "did something" .

mbentley, 2023-10-19:
Then again, I also have no idea how the backend data store works with any relation to the frontend - for all I know there are some references in the MongoDB to a specific path for a theme that either aren't getting updated properly or are failing to update.

Crocatowa, 2023-11-02:
I'm experiencing the same issue here
I had assumed it was the duplicate tenant issue, went down that path to find it wasn't that. 
Found this issue, went into the network tab during a login attempt and also have the -1200 errorCode

I've tried 3 browsers, cleared caches, safe mode, etc
But downloading and using the mobile app worked without issue
I'm still unable to log in via the browser currently

mbentley, 2023-11-02:
I forgot to update this issue but TP-Link support did reply saying that they will fix this in a future version:

> The R&D team confirmed that the "/img/svg/PaginationLeftDisabled.svg" displayed in the log is an illegal request, which will cause the login verification to fail and the current account will be forced to log out. 
We will modify the verification rule in subsequent firmware versions to "Illegal requests will be rejected directly without causing the account to be logged out."

chriexpe, 2023-11-16:
Thanks for contacting their support,  I just updated to the latest version and still having all flavors of "sessionCSRFToken is null for request", like "sessionCSRFToken is null for request /img/exp-improve.png" and "sessionCSRFToken is null for request /img/loading_apng.png".
Fingers crossed they actually solve this issue quickly.

leuchtrakete666, 2023-12-12:
in 5.13 this issue is still there.

klicking on any kind of menu, or device kicks me back to login. i downgraded to 5.9. this version run stable. no time to experiment much. i use this docker for my bnb. the guests demand internet ...

chriexpe, 2023-12-15:
After some back and forth with TP-Link support they stated that "For the Controller local login issue, it is planned to be fixed in Controller v5.14, which is expected to be released in 2024Q1.".
They even offered the beta version of it, tho it won't have Cloud Access, but I asked when there will be an public beta of it.

iggy, 2024-01-02:
As a temp workaround, I added the following rule to ublock origin to block the bad requests that were resetting the session.

```
||omada.my.domain/img/*
```

and that seems to have fixed getting logged out

PizzaProgram, 2024-01-12:
I have the same problem.
Totally fresh install, (both Rpi OS + Docker) version =  5.13.22 
Can not click anything important, (Logs or one of the AP devices) because it drops me back to login screen,
 and I have to retype my password even if "Remember..." is checked.
Tried different OSes, different browsers, all the same behaviour.

**Details:**
Installed through portainer.io docker. 
Deleted "Bridged network" and added "Host" instead because none of the ports were forwarded by default.

IMHO this whole dockered "solution" is unusable with this version and should **not** be published at all.
Is 5.9 version really stable? How do I install that instead?
Thanks!