mbentley/docker-omada-controller #346: [Bug]: Omada Controller Fails to Start Java Runtime Environment Fatal Error

Robert-litts, 2023-09-26:
### Controller Version

5.12.7

### Describe the Bug

Omada Controller will no longer launch after upgrading to 5.12. I cannot access the web GUI any longer. 

### Expected Behavior

When the container is started, I expect to be able to access the Omada Controller admin page. 

### Steps to Reproduce

1. Launch the container

### How You're Launching the Container

```plain
Docker Compose:

version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.12
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=America/New_York
    volumes:
      - /docker/omadasdn/data:/opt/tplink/EAPController/data
      - /docker/omadasdn/logs:/opt/tplink/EAPController/logs

volumes:
  omada-data:
  omada-logs:

```
```

### Container Logs

```plain
09-26-2023 14:08:21.393 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
#
# A fatal error has been detected by the Java Runtime Environment:
#
[... 101 lines cut ...]
09-26-2023 14:10:35.314 INFO [log4j-thread] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
09-26-2023 14:10:35.360 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
09-26-2023 14:10:37.397 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
09-26-2023 14:10:37.873 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
09-26-2023 14:10:37.874 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
09-26-2023 14:10:37.942 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
09-26-2023 14:10:37.942 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start
  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)
09-26-2023 14:10:38.597 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.12.7 using Java 17.0.8.1 on OmadaSDN-Controller with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.12.7.jar started by omada in /opt/tplink/EAPController/lib)
09-26-2023 14:10:38.602 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
09-26-2023 14:10:48.269 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
09-26-2023 14:10:50.657 INFO [main] [] c.t.s.o.m.l.p.e.LicenseEventCenterProperties(): licenseManagerTopic: omada.cloud.license.prd.topics
09-26-2023 14:10:53.525 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): upgradeSendReq workGroup core thread num is 1, max thread num is 1
09-26-2023 14:10:54.671 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
09-26-2023 14:10:55.350 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): manager workGroup core thread  num is 16, max thread num is 16
09-26-2023 14:10:55.723 INFO [main] [] c.t.s.o.m.c.d.m.s.a.AbstractActiveSiteCacheImpl(): Scheduled ActiveSitesCache period flush buf task at fixed rate of 30000 millis.

```
```

Robert-litts, 2023-09-26:
I'm running this within a LXC on Proxmox, and I just tried restoring to a backup from a few days ago (running 5.7) and still receiving the same Java issues. There's a potential that this is an issue with the LXC, although I am not sure what caused it. I spun up another instance of 5.12 on another LXC running Docker and it ran fine -- was able to restore my Omada SDN config without issues.

I'm not sure what caused this Java Runtime error to happen though, as I have not made any modifications to this particular LXC in a few months.

mbentley, 2023-09-27:
Well this is certainly strange.  I'll have to do some more googling to see if it is something related to a specific kernel version possibly.  That's the first thing that comes to mind.  That being said, the `/tmp/hs_err_pid1.log` file and the core dump from inside the container at the location listed could be helpful.  In order to get the core dump out, you might have to use `docker cp` or for ease of getting it out, maybe create a volume that bind mounts `/opt/tplink/EAPController/lib/core` somewhere on your host.  You may need to set the ownership of a directory correctly to `508:508` for a bind mount to be able to write to that location, wherever it is on your filesystem.

mbentley, 2023-12-10:
Closing due to lack of info (`tmp/hs_err_pid1.log` would be needed for further debugging).

closed 2023-12-10 by mbentley