mbentley/docker-omada-controller #350: [Bug]: Unable to log in after upgrade

emiellr, 2023-09-29:
### Controller Version

v5.9.31 -> v5.12.7

### Describe the Bug

When upgrading the controller by first shutting down with compose, removing the container, then pulling the compose file for the latest images, and starting back up, you can't log in anymore. It seems as if the database gets corrupted somehow. Rolling back a backup and using the older container version (v5.9.31) and doing it again still results in the same issue.

Detailed steps:

Not that I don't use their online login service, this applies to local accounts so I don't know whether this affects those accounts as well.

### Expected Behavior

The container launches and is upgraded to version 5.12.7 and I'll be able to log in

### Steps to Reproduce

- Shut down existing containers using
```sudo docker compose down```
- Wait for shutdown to complete
- Bump omada sdn version to 5.12.7 in the compose file
- pull, start compose and wait for it to start
- login error, username password not correct anymore

### How You're Launching the Container

```plain
Unknown as I'm running it through Openmediavault, though terminal output suggests it's just doing
sudo docker compose down and then removing the containers as well.
```

### Container Logs

```plain
n/a
```

mbentley, 2023-09-29:
See https://github.com/mbentley/docker-omada-controller/discussions/344#discussioncomment-7104908. It’s a software issue and I’ve filed a report upstream to TP-Link

closed 2023-09-29 by mbentley

emiellr, 2023-10-07:
Any news on whether this has been fixed yet? Maybe a mention under "known bugs" might be useful.

mbentley, 2023-10-07:
I see no updates from TP-Link in that issue and there has not been a new version released.  The more people that pile on with confirmation that they are also seeing the same issue should certainly do so on the forum thread so that TP-Link takes action.

mbentley, 2023-10-08:
> Any news on whether this has been fixed yet? Maybe a mention under "known bugs" might be useful.

Added to [known issues](https://github.com/mbentley/docker-omada-controller/blob/master/KNOWN_ISSUES.md#512---unable-to-login-after-upgrade).

emiellr, 2024-01-19:
@mbentley What's the status of the bug? Has it been fixed by TPLink yet?

mbentley, 2024-01-19:
I've seen no updates to https://community.tp-link.com/en/business/forum/topic/623942?sortDir=ASC&page=2 unfortunately.

bigjazzsound, 2024-04-20:
Has anyone tried to rollback to a working version?

mbentley, 2024-04-21:
I know that you won't be able to directly change the tag to go back.  You can't take a backup config file from the controller software & restore that from backup.  You would need a backup from 5.12 or you would need a backup of the persistent data to roll back.