mbentley/docker-omada-controller #357: [Enhancement]: Updated Health Check

mbentley, 2023-10-22:
### Controller Version

v5.x

### Describe the Bug

The health check should be updated to something better than loading the login screen.  The login screen returns a 200 even if MongoDB is down.

The API endpoint `/api/v2/anon/info` is better as it actually tries to talk to the db.
https://github.com/mbentley/docker-omada-controller/discussions/352#discussioncomment-7350985

The updates should include no longer using a shell script because it's not really needed - the health check can be called directly like in the examples in the [Dockerfile docs](https://docs.docker.com/engine/reference/builder/#healthcheck).  Different versions of the controller from 3.x, 4.x and 5.x might not have the same API endpoints as well so they would need to differ.

### Expected Behavior

Better health check

### Steps to Reproduce

n/a

### How You're Launching the Container

```plain
n/a
```

### Container Logs

```plain
n/a
```

mbentley, 2023-11-04:
I will want to check to see if this API endpoint returns an error when the MongoDB is incorrectly configured (i.e. - `EAP_MONGOD_URI` is empty)

bdunks, 2024-02-14:
If you decide to enhance the health checks,  you may consider replacing the `127.0.0.1` with a variable.

I am running the container using the macvlan network driver, and the existing health check will fail because the container has a different IP than the host.

I replaced `healthcheck.sh` with this and it worked for me.  But, I didn't regression test other networking scenarios:

```sh
CONTAINER_IP=$(hostname -I | awk '{print $1}')
wget --quiet --tries=1 --no-check-certificate -O /dev/null --server-response --timeout=5 "https://${CONTAINER_IP}:${MANAGE_HTTPS_PORT:-8043}/login" || exit 1
```

I've also seen healthcheck scripts that allow overriding the target url through an environmental variable, but another environmental variable may be overkill.

Ultimately, I realized I could declare a new (overriding) `healthcheck:` in my docker-compose file, so I have no pressing need for an update.  Just something to consider if you end up enhancing this in the future.

Thank you for your hard work keeping this well-maintained and up-to-date!

mbentley, 2024-02-14:
That’s interesting. You don’t have a loop interface when using macvlan?  My macvlan based containers all do.

bdunks, 2024-02-14:
You're right.  It looks like I misdiagnosed the issue.  At some point during troubleshooting I also corrected the MANAGE_HTTPS_PORT.  It didn't occur to me in that moment that the incorrectly configured port had been the root cause, not the host/ip.

Thanks for your quick response and for teaching me something new!

mbentley, 2024-03-20:
Was testing this today and maybe I am misunderstanding the wget timeouts but there doesn't seem to be an overall timeout for the entire connection's transaction with the server similar to how curl does.  In this case, wget waits about 30 seconds while waiting for a response from the API despite setting each timeout setting individually:

```
root@1bda6acfb5be:/opt/tplink/EAPController/lib# time wget --verbose --tries=1 --no-check-certificate -O /dev/null --server-response --dns-timeout=.25 --connect-timeout=1 --read-timeout=1 "https://127.0.0.1:${MANAGE_HTTPS_PORT:-8043}/api/v2/anon/info"; echo $?
--2024-03-20 10:54:05--  https://127.0.0.1:8043/api/v2/anon/info
Connecting to 127.0.0.1:8043... connected.
WARNING: cannot verify 127.0.0.1's certificate, issued by 'OU=TP-Link,O=TP-Link,CN=localhost':
  Self-signed certificate encountered.
    WARNING: certificate common name 'localhost' doesn't match requested host name '127.0.0.1'.
HTTP request sent, awaiting response...
  HTTP/1.1 500
  X-Frame-Options: SAMEORIGIN
  X-Content-Type-Options: nosniff
  Content-Security-Policy: default-src 'self' https://*.tplinkcloud.com/;script-src 'self' 'unsafe-eval' 'sha256-7W9UiBaYGlOHpT1aQBLegqffUVHbYq6/ZAb+ErjUb40=' 'sha256-VGQ8jNTL2g0e8wPwOgyCQJDqhuRgfV7gRYexcBkBe4Y=' 'sha256-x2jgB1zBLi30IsfY+VNgWjwBGeHPJxOSrzl+IdsT6k0=' 'sha256-0AHZXO4clnpdcxqdmASPBEp4JCIrtaxIX/mUuL1kzZw=' 'sha256-lfXlPY3+MCPOPb4mrw1Y961+745U3WlDQVcOXdchSQc=';style-src 'self' 'unsafe-inline';connect-src 'self' https://*.tplinkcloud.com/ https://*.tplinkcloud.com:8843/ wss://*.tplinkcloud.com/ https://*.tiles.mapbox.com https://api.mapbox.com https://events.mapbox.com ;frame-src 'self' data:;img-src 'self' https://*.tplinkcloud.com/ https://*.mzstatic.com/ https://play-lh.googleusercontent.com/ data: blob:;child-src blob: ;worker-src blob: ;object-src 'self' data: blob:
  Strict-Transport-Security: max-age=31536000; includeSubDomains
  Referrer-Policy: strict-origin-when-cross-origin
  Content-Type: application/json
  Transfer-Encoding: chunked
  Date: Wed, 20 Mar 2024 10:54:35 GMT
  Connection: close
2024-03-20 10:54:35 ERROR 500: (no description).

real    0m30.040s
user    0m0.005s
sys     0m0.005s
8
```

This isn't ideal because the health check needs to complete quicker than that.