mbentley/docker-omada-controller #358: [Feature]: Publish latest Beta 5.13.10

Gruni22, 2023-10-27:
### What problem are you looking to solve?

Hi,
Thanks for those great images.
Could you please publish the latest Beta?
https://static.tp-link.com/upload/beta/2023/202310/20231027/Omada_SDN_Controller_v5.13.10_Linux_x64.deb(Beta).zip

Thanks!

Gruni22, 2023-10-27:
Link seems not to work. You may find it here

[https://community.tp-link.com/en/business/forum/topic/628384](https://community.tp-link.com/en/business/forum/topic/628384)

mbentley, 2023-10-27:
Their links tend to have a user-agent or origin checker against them.  I'll need to use the `https://static.tp-link.com/upload/beta/2023/202310/20231027/Omada_SDN_Controller_v5.13.10_Linux_x64.tar.gz(Beta).gz` instead of the deb.  And of course, they have inconsistent packaging of their binaries.  The last beta version was a .zip file and now this one is gzipped so I will have to change the install.sh again.

closed 2023-10-28 by mbentley

mbentley, 2023-10-29:
Beta 5.13.10 is available now in case you didn't see #359 and #362

commit 6be1072a (2023-10-28): Merge pull request #359 from mbentley/update-beta

Updated install script to deal with the different versions from the beta; fixes #358