mbentley/docker-omada-controller #363: [Feature]: make option to run image truly as non root for kubernetes

mr-sab, 2023-11-02:
### What problem are you looking to solve?

First of all thanks for this image and all the effort that has been made.

The problem I am try to solve is to make the image truly non root for kubernetes.
It is possible to with optional Variables to start the software as non root, but root is still needed to create the user, group file permissions etc.
this leaves the software running as a non root user but root permissions are still part of the pod which for a security view is not ideal

### Describe the solution that you have in mind

add  the user creation part* also in the dockerfile, this way the user is already created and can be used in the kubernetes context, so no root in the deployment process is needed, and all the check are already met as long the default user/uid are used.
if a other enduser still want to overwrite the user/uid it is business as usual.

*https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L36-L103

other option would to be build a "special" kubernetes image, so all the install, file permissions, user etc etc are in the build process. and the image would only need to run run the software on startup.

### Additional Context

Please share what you think, I would like to create a PR to make this possible. But first lets make a decision on if and how to fix this :)

mbentley, 2023-11-03:
So it would definitely be doable and the logic to make it function makes sense but I have a question on the reasoning behind wanting to run truly rootless in order to better understand the goal behind the ask.  Is the idea because your k8s cluster has (or you would like to use) a pod security configuration to prevent containers running as root or is it that you just prefer that the container doesn't start with root at all?  Or is there something else that I am missing?

I ask because if we were to reconfigure the image to start with the default, non-root user and only change the user/group/uid/gid when the user has specified a non-default one, it still comes back to that there needs to be a privilege escalation ability built into the image for that non-root user in order to be able to modify itself.  At that point, we would be back to using passwordless sudo (unless there is some other option I do not know of) but even if it is restricted for specific commands, it still leaves some pretty big opportunities for further root access I would think and then it becomes a scenario of chasing those loopholes down.

mr-sab, 2023-11-04:
I would like indeed to use the Pod Security Standards](https://kubernetes.io/docs/concepts/security/pod-security-standards/)  Restricted in my cluster as much as possible. and dont want to give the container more permissions then it needs.

I get your point, indeed if we would make the user/group etc part of the build process and the new default for running the image. If you want to change this you would still need root. But at least this gives the option to run the image without root in the first place.

other option would be to like for example [this](https://github.com/onedr0p/containers/tree/main/apps/radarr) to just do a 755. This container will run on any uid that you start it with. No sure if that is possible here, or that this is some other steppings happing that I maybe mist. and it is possible to running fully conform the latest restricted baseline.

So for next steps, should I try to make it just like to the radarr example above? For what I see now this needs the follow changes:
-make user creation and all chmod/chown steps part of the build process(aka dockerfile)
-change the needed file/directory to 755(so if works with any user/uid)
-remove user and chown/chmod steps from the entrypiont.sh
-test

mbentley, 2023-11-04:
Hmm, I'll have to think about the approach further.  I don't love the idea of opening up the permissions on everything to 755 though.  I also have to take into consideration breaking any sort of backwards compatibility with how anyone is using the controller image in it's current form as there are a number of areas where there are `chown` and `chmod` commands that happen that were used to deal with edge cases or situations where someone is upgrading from an older version where directories need to be modified or created.

mr-sab, 2023-11-04:
Yes I understand that 755 has downsides as well, then the other option is to change the default user to a non root and perform the the chown chmod as part of the dockerfile. But also leave this in the entypoint.sh
This way the image will also run as non root by default and also don't need root for normal/greenfield deployments,  but for people using docker still have the entrypoint.sh to cover the edge cases. 

Another option is to create a non root/kubernetes version of this this image, with all the changes needed. This will probably be more work to create/maintain but is in my option cleaner.

mbentley, 2025-03-15:
So I am taking another pass at this since #551 also came up.  Pulling this from over there:

I did a quick pass on the entrypoint to see what all would fail and it's a good chunk of it or at least a good chunk would fail given the right conditions:

- Skip the user/group checks: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L46-L98
- Copying over properties files, if missing: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L100-L111
- If the UID/GID is different, changing the ports will fail to write to the file: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L123-L185
- Re-creating the html directory will fail, if missing: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L187-L194
- Re-creating the pdf directory will fail, if missing: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L196-L203
- Fixing permissions will fail: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L205-L217
- Fixing /tmp permissions will fail: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L219-L225
- Creating a db directory may fail if it has the wrong ownership: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L227-L234
- Cert import to the keystore will fail: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L236-L274
- Re-enabling TLS versions will fail: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L276-L292
- This workaround will fail: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L375-L396
- Would need to handle the gosu instances: https://github.com/mbentley/docker-omada-controller/blob/master/entrypoint-5.x.sh#L435-L448

mbentley, 2025-03-15:
Just to be clear, I’ve yet to actually test the image on k8s.  I have only tested in a local Docker daemon.  You do also need to set the persistent data dirs ownership for data and logs appropriately but the rootless entrypoint should complain.

mr-sab, 2025-03-16:
Thank you for looking it in this, I tested it and it fails on the check of the data dir
In k8s you dont change the user of the volume only the group please see:

https://medium.com/marionete/kubernetes-securitycontext-with-practical-examples-67d890558d11
https://kubernetes.io/docs/tasks/configure-pod-container/security-context/

mbentley, 2025-03-18:
Do you have an example k8s deployment manifest that you're using just so that when I test, I can compare to what you're doing exactly?  I don't use k8s for my own home use so I haven't taken the time to create one for this.  I have plenty of clusters to test with due to work in my home lab though so that's not a problem.

cnu80, 2025-03-18:
Hi, I tried it and I got the error:

```
+ omada-controller-5b4fc88679-jgnpw › omada-controller
omada-controller-5b4fc88679-jgnpw omada-controller INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
omada-controller-5b4fc88679-jgnpw omada-controller INFO: Group (omada) doesn't exist; creating
omada-controller-5b4fc88679-jgnpw omada-controller groupadd: Permission denied.
omada-controller-5b4fc88679-jgnpw omada-controller groupadd: cannot lock /etc/group; try again later.
- omada-controller-5b4fc88679-jgnpw › omada-controller
+ omada-controller-5b4fc88679-jgnpw › omada-controller
omada-controller-5b4fc88679-jgnpw omada-controller INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
omada-controller-5b4fc88679-jgnpw omada-controller INFO: Group (omada) doesn't exist; creating
omada-controller-5b4fc88679-jgnpw omada-controller groupadd: Permission denied.
omada-controller-5b4fc88679-jgnpw omada-controller groupadd: cannot lock /etc/group; try again later.
- omada-controller-5b4fc88679-jgnpw › omada-controller
```

My K8S deployment is attached. I use 

`kubectl apply -k .`

to deploy the application.

[Archiv.zip](https://github.com/user-attachments/files/19318137/Archiv.zip)

thanks

mbentley, 2025-03-18:
You need to use the image `mbentley/omada-controller:5.15-rootless-test`, run it with the uid/gid of `508:508`, and an env var of `ROOTLESS=true`.

Those logs show it running the normal entrypoint and it looks like in your deployment, you're using an older image: `mbentley/omada-controller:5.6`. You can't just set the PUID and PGID env vars, you have to actually specify the user and group, like:

```
...
  securityContext:
    runAsUser: 508
    runAsGroup: 508
```

cnu80, 2025-03-18:
sorry, I uploaded the old deployment. securitycontext and ROOTLESS are already configured. But I am using the beta image, I will try it with the 5.15-rootless-test  thanks

mr-sab, 2025-03-20:
I retried it and it almost worked could also be some config error on my side.
After I made the changes needed to run it rootless omada did a factroy reset but the pod was running fine and webui was working(setup page because of the factory reset)
I needed to change some things in my pvc and mounts, so I think it is/was a issue on my side.

Wil look into it this weekend to see what it did wrong, change it back to my old 'working' config from git and now my data is back aswell.

mbentley, 2025-03-20:
Here is what I've deployed using @cnu80 's example manifests.
[k8s-omada.zip](https://github.com/user-attachments/files/19376220/k8s-omada.zip)

I imagine you could strip down the deployment to get it running in another namespace, just exposing some basic ports for http/https for testing to not impact your current controller.

mr-sab, 2025-03-22:
tested it in a other namespace, and looks to working great!

mbentley, 2025-03-23:
So is it acceptable that the user needs to be `508:508` or is it desirable to be able to choose any UID?  I could opening up permissions of `/opt/tplink/EAPController/properties` to allow `other` to write which would allow any uid to work.  Overall I wouldn't be opposed to that, it would just change those permissions for all of the images which I would say is a pretty low risk since it's just properties files.

mbentley, 2025-03-23:
Oh yeah, @cnu80 - not sure if I mention this but you had a port that was set to TCP instead of UDP I believe.  Here is the updated service yaml file:

```yaml
apiVersion: v1
kind: Service
metadata:
  name: omada-controller-service
spec:
  type: LoadBalancer
  selector:
    app: omada-controller
  ports:
  - name: manage-http
    port: 8088
    targetPort: 8088
    protocol: TCP
  - name: manage-https
    port: 8043
    targetPort: 8043
    protocol: TCP
  - name: portal-https
    port: 8843
    targetPort: 8843
    protocol: TCP
  - name: app-discovery
    port: 27001
    targetPort: 27001
    protocol: UDP
  - name: adopt-v1
    port: 29812
    targetPort: 29812
    protocol: TCP
  - name: upgrade-v1
    port: 29813
    targetPort: 29813
    protocol: TCP
  - name: manager-v1
    port: 29811
    targetPort: 29811
    protocol: TCP
  - name: manager-v2
    port: 29814
    targetPort: 29814
    protocol: TCP
  - name: udp-discovery
    port: 29810
    targetPort: 29810
    protocol: UDP
  - name: udp-management
    protocol: UDP
    port: 19810
    targetPort: 19810
  - name: transfer-v2
    port: 29815
    targetPort: 29815
    protocol: TCP
  - name: rtty
    port: 29816
    targetPort: 29816
    protocol: TCP
```

mbentley, 2025-03-23:
I just pushed [documentation](https://github.com/mbentley/docker-omada-controller/tree/add-rootless?tab=readme-ov-file#running-rootless) and [example k8s manifests](https://github.com/mbentley/docker-omada-controller/tree/add-rootless/k8s) based on those you shared @cnu80.  Would love some feedback on those to see if that is sufficient before I create a PR.

sofiaurora270, 2025-03-24:
I've also read the documentation, and the example manifests, and everything seems to be just like how I got it working. I'm only using one PVC and using subpath to mount logs+data to the same pvc.

sofiaurora270, 2025-03-25:
Oh btw, at work, were starting to use openshift, and at openshift, you cant select user, but openshift assings user randomly on pod start.

So for those who want to run this on openshift, ability for any userid/groupid would be valuable.

mr-sab, 2025-03-27:
I agree with the openshift statement @sofiaurora270 , would be great to run it as any random uid.

@mbentley documentation looks good, one point of feedack is to include the [fsGroupChangePolicy: "OnRootMismatch"](https://kubernetes.io/docs/tasks/configure-pod-container/security-context/#configure-volume-permission-and-ownership-change-policy-for-pods) and maybe remove the PUID GIUD because they are not used in combination with the ROOTLESS options

mbentley, 2025-03-28:
I've incorporated those and I just merged in rootless support. I appreciate the help @sofiaurora270 @cnu80 @mr-sab !

closed 2025-03-28 by mbentley

sofiaurora270, 2025-03-28:
Does this mean that we can just use the normal image now with proper env, so there is no excplicit rootless image? I saw that you updated all images just now.

mbentley, 2025-03-28:
That's right - proper env vars and the UID/GID set with the normal image.

commit 0e90a994 (2025-03-28): Merge pull request #560 from mbentley/add-rootless

Add rootless; fixes #363