mbentley/docker-omada-controller #371: [Bug]: Controller can't connect to the network after power outage

chriexpe, 2023-11-17:
### Controller Version

5.12.7

### Describe the Bug

After a power outage, Omada Controller has network issues while starting, where mostly it'll be restarting, and when it does boot and I can access the login screen, I get this error when trying to login "Unable to login, check your network connection", also it doesn't connect to the internet (ER605 w/ PPPoE).
The only way for me to get it working is either by replacing the Gateway (ER605) and using a Huawei router that I have here, or by reseting the Gateaway and configuring it as independent device (which is how I'm using now). After this I can use the controller just fine.

### Expected Behavior

The controller boots normally and is able to access the network and manage devices.

### Steps to Reproduce

Unplug all omada devices + RPI from the power
Plug everything again and wait for everything to boot

In my case now I wouldn't have network, internet or access to the controller

### How You're Launching the Container

```plain
Default docker-compose from docs on Portainer Stacks, running on RPI4B 8GB w/ OS Lite x64
```

### Container Logs

```plain
07-07-2023 11:50:29.095 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): succeed get default omadac OmadacVO(id=c6c64d10db22a9be2c564705f9859460, name=OC_Christian)
07-07-2023 11:50:29.127 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: before init for hwc
07-07-2023 11:50:29.127 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(): record: after init for hwc
07-07-2023 11:50:29.224 INFO [main] [] c.t.s.o.c.p.c.o.s.ActiveSiteCacheImpl(): Scheduled ActiveSitesCache period flush buf task at fixed rate of 30000 millis.
07-07-2023 11:50:29.497 INFO [main] [] c.t.s.o.d.g.c.d.ThreadConfiguration(): device-gateway datatrack workGroup core thread  num is 16, max thread num is 16
07-07-2023 11:50:29.826 INFO [main] [] c.t.s.o.p.p.r.a.c(): init nioEventLoopGroup
07-07-2023 11:50:30.211 INFO [main] [] c.t.s.o.d.g.c.f.b.c(): file download mq consume task is start...
07-07-2023 11:50:30.240 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): adopt workGroup core thread num is 8, max thread num is8
07-07-2023 11:50:30.426 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(): discovery workGroup core thread num is 2, max thread num is 10
07-07-2023 11:50:30.464 INFO [main] [] c.t.s.o.m.l.p.e.LicenseEventCenterProperties(): licenseManagerTopic: omada.cloud.license.prd.topics
07-07-2023 11:50:32.033 INFO [main] [] c.t.s.r.p.b(): register rtty message Topic: omada.rtty.message, groupId: omada-rtty-message, handler: com.tplink.smb.rtty.proxy.a.b@7977e801
07-07-2023 11:50:32.033 INFO [main] [] c.t.s.r.p.b(): register rtty message Topic: omada.rtty.manager, groupId: omada-rtty-manager, handler: com.tplink.smb.rtty.proxy.a.a@619539b6
07-07-2023 11:50:32.110 INFO [main] [] c.t.s.o.d.g.c.d.ThreadConfiguration(): device-gateway datatrack time task workGroup core thread  num is 16, max thread num is 16
07-07-2023 11:50:33.208 INFO [main] [] c.t.s.o.p.b.a.c(): DisconnectRequestServer start
07-07-2023 11:50:33.435 INFO [main] [] c.t.s.o.s.d.a(): record: after OmadacInitTask
07-07-2023 11:50:33.436 INFO [main] [] c.t.s.o.s.t.CloudStartUpTask(): record: CloudStartUpTask start
07-07-2023 11:50:33.436 INFO [main] [] c.t.s.o.s.t.CloudStartUpTask(): record: CloudStartUpTask finished
07-07-2023 11:50:33.436 INFO [main] [] c.t.s.o.s.t.OmadacDiscoveryStartUpTask(): record: OmadacDiscoveryStartUpTask start
07-07-2023 11:50:33.437 INFO [main] [] c.t.s.o.s.t.OmadacDiscoveryStartUpTask(): record: OmadacDiscoveryStartUpTask finished
07-07-2023 11:50:33.537 INFO [main] [] c.t.s.o.d.g.a(): startServers, DISCOVERY host:null.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): discovery Port is 29810.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): manage    port v1 is 29811.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): manage    port v2 is 29814.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): adopt     port v1 is 29812.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): upgrade   port is 29813.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): transfer  port is 29815.
07-07-2023 11:50:33.538 INFO [main] [] c.t.s.o.d.g.a(): rtty port is 29816.
07-07-2023 11:50:33.606 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyUdpServer bind /0.0.0.0:29810, export localhost/127.0.0.1:29810
07-07-2023 11:50:33.641 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29812 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
07-07-2023 11:50:33.647 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
07-07-2023 11:50:33.648 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
07-07-2023 11:50:33.649 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
07-07-2023 11:50:33.650 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
07-07-2023 11:50:33.651 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
07-07-2023 11:50:33.653 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
07-07-2023 11:50:33.654 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
07-07-2023 11:50:33.655 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29815 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
07-07-2023 11:50:33.655 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29815, export localhost/127.0.0.1:29815
07-07-2023 11:50:33.658 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start RttyNettyTcpServer bind /0.0.0.0:29816, export localhost/127.0.0.1:29816
07-07-2023 11:50:33.658 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
07-07-2023 11:50:33.659 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started
[... 119 lines cut ...]
        at com.tplink.eap.cloudsdk.httpclient.ApiHttpClient.sendPostRequest(SourceFile:302) ~[cloudsdk-1.0.13.jar:?]
        at com.tplink.eap.cloudsdk.httpclient.ApiHttpClient.sendPostRequest(SourceFile:272) ~[cloudsdk-1.0.13.jar:?]
        at com.tplink.eap.cloudsdk.httpclient.ApiHttpClient.sendPostRequest(SourceFile:211) ~[cloudsdk-1.0.13.jar:?]
        at com.tplink.smb.omada.cloud.platform.c.b(SourceFile:129) ~[omada-cloud-core-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.cloud.platform.a.b.c.a(SourceFile:28) ~[omada-cloud-core-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.cloud.platform.b.e.c(SourceFile:65) ~[omada-cloud-core-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.system.a.c.a.a(SourceFile:82) ~[omada-system-core-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.system.port.a.d.a(SourceFile:71) ~[omada-system-core-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.manager.c.a.b(SourceFile:47) ~[manager-port-local-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.manager.c.a.a(SourceFile:39) ~[manager-port-local-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.manager.c.f.c(SourceFile:144) ~[manager-port-local-5.9.31.jar:5.9.31]
        at com.tplink.smb.omada.manager.c.f.handle(SourceFile:108) ~[manager-port-local-5.9.31.jar:5.9.31]
        at com.tplink.smb.component.schedule.common.TaskExecutorService.lambda$runTask$0(TaskExecutorService.java:75) ~[solution-components-schedule-common-1.3.5.jar:1.3.5]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) ~[?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[?:?]
        at java.lang.Thread.run(Thread.java:833) [?:?]
```

### Additional Context

On tplinkcloud the controller host has this many IPs
[image]

mbentley, 2023-11-17:
I don't think this is specific to the packaging of the Docker image and sounds more like an issue with the software itself.  Have you submitted an issue on the TP-Link forums?

chriexpe, 2023-11-17:
> I don't think this is specific to the packaging of the Docker image and sounds more like an issue with the software itself. Have you submitted an issue on the TP-Link forums?

On the forum there was only one reply saying that it sould be on "host" and that's it, I did try my luck on TP-Link support but I guess I contacted the wrong one cuz this was their reply:

> For docker Controller, it is better to find docker support on the issue, and we have no much information on it.
> You can consider OC200 or Software Controller directly download on Linux OS.

mbentley, 2023-11-17:
Could you provide a link to the compose file?  I am not familiar with the location you specified.  It looks like in the logs that the controller did start successfully.  I'm not really following where the problem is introduced - it sounds like if you restart the controller, you then can't login to the controller.

Sounds like their support would just want you to try to reproduce the issue directly on a native OS to remove the container as a possible root of the issue which is understandable on their end.

chriexpe, 2023-11-17:
> Could you provide a link to the compose file? I am not familiar with the location you specified. It looks like in the logs that the controller did start successfully.

version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:latest
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - PORT_TRANSFER_V2=29815
      - PORT_RTTY=29816
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Etc/UTC
    volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-logs:/opt/tplink/EAPController/logs

volumes:
  omada-data:
  omada-logs:

> I'm not really following where the problem is introduced - it sounds like if you restart the controller, you then can't login to the controller.

Good question, before having these issues I used your docker image on my Unraid server with fixed macvlan br0 IP (same range as host) and it worked fine, but after installing on my RPI I started getting these weird errors, and all settings are default aside from the Host IP (RPI OS) being fixed. (PS. I did try to setup macvlan on the RPI but still had that error)
Before using Portainer Stacks I did try that same docker-run from your docs and after a power outage, got this same error.

J4CE2, 2023-11-17:
Please try login with a local admin (not cloud admin accountt).
Cloud admin account will not work if your internet is down / not connected

mbentley, 2023-11-17:
Good point - logging in with cloud credentials would not work if you're attempting to do so.

Looks like from the logs that you have 2 Switches, 1 AP, and 1 gateway (router), is that right?  It looks like the AP gets it's configuration synchronized correctly but your router does not.  I have no idea what this error means but you can see in the logs below that there is an error executing "syncConfigurationForSameVersion":

```
07-07-2023 11:50:51.544 INFO [adopt-work-group-1] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(c6c64d10db22a9be2c564705f9859460) DeviceMac(14-EB-B6-E5-1D-19)
07-07-2023 11:50:54.663 INFO [manage-work-group-15] [] c.t.s.o.m.d.p.t.c(): Device 14-EB-B6-E5-1D-19 OmadacId(c6c64d10db22a9be2c564705f9859460) changed to status CONNECTED, which don't need to handle.
07-07-2023 11:50:56.851 INFO [server-comm-pool-3] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(c6c64d10db22a9be2c564705f9859460) DeviceMac(14-EB-B6-E5-1D-19)
07-07-2023 11:50:56.861 INFO [manage-work-group-1] [] c.t.s.o.m.d.d.m.i.e(): first inform no need send full config to OmadacId(c6c64d10db22a9be2c564705f9859460) DeviceMac(14-EB-B6-E5-1D-19)
07-07-2023 11:50:56.863 WARN [manage-work-group-1] [] c.t.s.o.m.d.d.m.m.s.e(): fill set msg body but all setting null, skip send set msg to OmadacId(c6c64d10db22a9be2c564705f9859460) DeviceMac(14-EB-B6-E5-1D-19)
07-07-2023 11:50:56.863 WARN [manage-work-group-1] [] c.t.s.o.m.d.d.m.m.s.e(): failed to genConfigBodyAndAddVersion with keys:[]
07-07-2023 11:50:56.864 INFO [manage-work-group-1] [] c.t.s.o.m.d.d.m.d.c.A(): syncConfigurationForSameVersion to OmadacId(c6c64d10db22a9be2c564705f9859460) DeviceMac(14-EB-B6-E5-1D-19), result:SendDeviceMsgResult(success=false, deviceResponse=null)
```

Whoever you got from support was being low effort and not even looking at the logs apparently, assuming this is the error that happens every time.

Did some quick google searching and it came back to https://github.com/mbentley/docker-omada-controller/issues/248 but there was no resolution shared.  I guess the most obvious question would be is if there is a firmware upgrade available for your router?  It could be there is an incompatibility between your router and the controller version.

chriexpe, 2023-11-18:
> Please try login with a local admin (not cloud admin accountt). Cloud admin account will not work if your internet is down / not connected

Hmmm I didn't thought about that, actually I always used exactly the same username and password as Cloud Admin, so that's the reason for that error!

chriexpe, 2023-11-19:
> It looks like the AP gets it's configuration synchronized correctly but your router does not.

So that might explain why it doesn't connect to the internet after booting, or at least, part off it. Now that I can access the Controller, I checked and the Gateway (ER605) didn't even detect that the EPON ONU is connected to it, and no matter how I connect or/and reboot both of them, it still shows as nothing is connected to it... When the Gateway wast setup as standalone device it worked fine.
[image]

This is the settings on the controller:
[image]

>  I guess the most obvious question would be is if there is a firmware upgrade available for your router?

The Controller doesn't seem to detect new version of the Gateway, I had to manually update it through it's own IP in order to update to the latest version (2.2.2).

This is part of  the Log when this was happening:

11-18-2023 23:00:55.795 INFO [https-jsse-nio-8043-exec-19] [] c.t.s.o.a.a.i.h.HealthInternalAdapterConfiguration(): NoSupport internal service for anomaly

11-18-2023 23:00:57.302 INFO [https-jsse-nio-8043-exec-9] [] c.t.s.o.a.a.i.h.HealthInternalAdapterConfiguration(): NoSupport internal service for anomaly

11-18-2023 23:01:02.094 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.d.b.c(): May be invalid upTime for mac DeviceMac(14-EB-B6-E5-1D-19) with omadac OmadacId(64722017c43d8ffc9bbfe416ead2bd0f), oldUpTime 0 days 00:20:11, reportedUpTime 0 days 00:03:20, discoveryMsg OsgDiscoveryV2DTO(super=BaseOsgDiscoveryDTO(super=AbstractDiscoveryDTO(timestamp=1700348462084, ecspVersion=2.2.0, ecspFirstVersionEnum=V2, deviceType=gateway, deviceMac=14-EB-B6-E5-1D-19, omadacId=64722017c43d8ffc9bbfe416ead2bd0f), deviceInfo=com.tplink.smb.omada.device.manage.message.discovery.osg.OsgDiscoveryDeviceInfo@21ef3094[model=ER605,modelVersion=2.0,firmwareVersion=2.2.2 Build 20231017 Rel.68869,hardwareVersion=ER605 v2.0,specialModel=<null>,upTime=0 days 00:03:20,wireless=<null>,additionalProperties={},ip=192.168.0.1,ipv6List=<null>,factory=false,additionalProperties={}], deviceMisc=com.tplink.smb.omada.device.manage.message.discovery.osg.OsgDeviceMisc@af3018d[portNum=5,extraPortNum=com.tplink.smb.omada.device.manage.message.discovery.osg.OsgExtraPortNum@41edc707[usbLteWan=1,additionalProperties={extraPort=0}],support11ac=<null>,supportLag=<null>,supportMesh=<null>,supportRoaming=<null>,customizeRegion=<null>,supportChannelLimit=<null>,channelLimitMode=<null>,supportDfs=<null>,additionalProperties={},modelType=<null>,category=<null>]), controllerSetting=com.tplink.smb.omada.device.manage.message.discovery.osg.OsgDiscoveryControllerSettingV2@4da980b7[controllerId=64722017c43d8ffc9bbfe416ead2bd0f,destOmadacId=<null>,additionalProperties={}])

11-18-2023 23:01:02.144 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.d.a(): MANAGED_BY_OWN Device 14-EB-B6-E5-1D-19 on omadac 64722017c43d8ffc9bbfe416ead2bd0f is discovered.

11-18-2023 23:01:02.281 INFO [Thread-7] [] c.t.e.c.c.n(): Exception occurs when connecting SEF server:java.net.UnknownHostException: n-device-entry-omada.tplinkcloud.com: Temporary failure in name resolution

11-18-2023 23:01:06.730 INFO [server-comm-pool-2] [] c.t.s.e.s.a.b.e(): serverType from channel: /192.168.0.80:29814 -> /192.168.0.1:33723 is MANAGE

11-18-2023 23:01:07.282 INFO [Thread-7] [] c.t.e.c.c.n(): Exception occurs when connecting SEF server:java.net.UnknownHostException: n-device-entry-omada.tplinkcloud.com

11-18-2023 23:01:12.879 INFO [adopt-work-group-6] [] c.t.s.o.m.d.d.m.d.c.e(): Gateway OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) SiteId(65581e9bfea6db760676c2a4) DeviceMac(14-EB-B6-E5-1D-19) adopt[auto=true] ok

11-18-2023 23:01:13.076 INFO [adopt-work-group-6] [] c.t.s.o.m.d.d.m.a.d(): send empty setting to OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) DeviceMac(14-EB-B6-E5-1D-19)

11-18-2023 23:01:16.189 INFO [manage-work-group-3] [] c.t.s.o.m.d.p.t.c(): Device 14-EB-B6-E5-1D-19 OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) changed to status CONNECTED, which don't need to handle.

11-18-2023 23:01:17.676 INFO [server-comm-pool-4] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) DeviceMac(14-EB-B6-E5-1D-19)

11-18-2023 23:01:17.695 INFO [manage-work-group-2] [] c.t.s.o.m.d.d.m.i.e(): first inform no need send full config to OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) DeviceMac(14-EB-B6-E5-1D-19)

11-18-2023 23:01:17.696 WARN [manage-work-group-2] [] c.t.s.o.m.d.d.m.m.s.e(): fill set msg body but all setting null, skip send set msg to OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) DeviceMac(14-EB-B6-E5-1D-19)

11-18-2023 23:01:17.696 WARN [manage-work-group-2] [] c.t.s.o.m.d.d.m.m.s.e(): failed to genConfigBodyAndAddVersion with keys:[]

11-18-2023 23:01:17.697 INFO [manage-work-group-2] [] c.t.s.o.m.d.d.m.d.c.C(): syncConfigurationForSameVersion to OmadacId(64722017c43d8ffc9bbfe416ead2bd0f) DeviceMac(14-EB-B6-E5-1D-19), result:SendDeviceMsgResult(success=false, deviceResponse=null)

mbentley, 2024-06-27:
I know that I don't have an update.  I just searched the forums and don't see anything and unfortunately the logs do not explain why the router wouldn't just come up with the previous config and sync with the controller when it can.

s1r0x, 2024-06-27:
Today i tried your beta 5.14 but it wont start up (some exception on startup)

My problem is that there is an UnknownHostException when i try to search for updates, same problem as the originally in this issue
(java.net.UnknownHostException: n-device-omada-api.tplinkcloud.com)

chriexpe, 2024-06-28:
> Today i tried your beta 5.14 but it wont start up (some exception on startup)
> 
> My problem is that there is an UnknownHostException when i try to search for updates, same problem as the originally in this issue (java.net.UnknownHostException: n-device-omada-api.tplinkcloud.com)
That version is broken, try with this one:
https://github.com/mbentley/docker-omada-controller/issues/418#issuecomment-2143029097

s1r0x, 2024-06-28:
Thx, that one is working

s1r0x, 2024-07-14:
hmmm, maybe it has something to do with the certificate on tplink side.....

[image]

J4CE2, 2024-07-14:
> hmmm, maybe it has something to do with the certificate on tplink side.....

https://n-device-omada-api.tplinkcloud.com/

Can you open that in a browser on your network?

s1r0x, 2024-07-14:
no, can you?

J4CE2, 2024-07-14:
> no, can you?

What error do you get? 

The host works for me, just get a cert error and the following text

Whitelabel Error Page
This application has no explicit mapping for /error, so you are seeing this as a fallback.
--
Sun Jul 14 18:57:40 GMT 2024
There was an unexpected error (type=Not Found, status=404).
No message available

s1r0x, 2024-07-14:
If i add the certificate manually as trusted/as exception, i get the same result as you. But i don't think that this is the correct way, because the problem is the certificate

J4CE2, 2024-07-14:
I was trying to determe if something at DNS level was blocking you to connect. But seems that is not the case.

s1r0x, 2024-07-14:
and checking the firmware is working in with your controller?
06-28-2024 17:57:56.225 INFO [check-update-work-group-0] [] c.t.s.o.m.c.o(): Checking last firmwares of managed de....

J4CE2, 2024-07-14:
Works ok here no errors in logs

01-15-2024 03:41:50.652 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
01-15-2024 03:41:50.875 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.

s1r0x, 2024-07-15:
'-- net host' did it, now it's working

mbentley, 2024-07-15:
From the [SSL Shopper cert check checker:](https://www.sslshopper.com/ssl-checker.html#hostname=n-device-omada-api.tplinkcloud.com)
[image]

The chain is actually correct, it's just that it's a self signed certificate from their own certificate authority.

It would not surprise me if they have the certificate trusted within the app itself seeing as it is a self-signed certificate by TP-Link that is good for a very long time.  It's not in the java keystore through, that's just the controller's cert.

s1r0x, 2024-07-15:
Yes, i meant with "something wrong" that it is untrusted (selfsigned) but my problem was my missing network

mbentley, 2024-07-17:
So just to circle back on this, if you haven't I would reach out to TP-Link support either on the [community forums](https://community.tp-link.com/en/business/forum/582) or through their [support page](https://www.tp-link.com/support/) as the original problem certainly seems like it is software related and not image packaging related.

closed 2024-07-17 by mbentley

s1r0x, 2024-07-17:
i haven't reported the issue with the selfsigned certificate to tplink