mbentley/docker-omada-controller #377: FYI: 5.13.22 has been released

threeonparfive, 2023-12-02:
### What problem are you looking to solve?

Just wanted to let you know in case you weren't aware that TP-Link has just released the 5.13.22 Linux build.  Thanks for all of your great work!

### Describe the solution that you have in mind

.

buntspexxxxt, 2023-12-02:
The linux version is not yet available

threeonparfive, 2023-12-02:
> The linux version is not yet available

Yes, it is: https://www.tp-link.com/us/support/download/omada-software-controller/

mbentley, 2023-12-02:
They sure do make it hard to monitor for updates...

https://www.tp-link.com/us/support/download/omada-software-controller/v5/ has it 
https://www.tp-link.com/en/support/download/omada-software-controller/v5/ does not

Additionally, the French version of the site had it up a couple days ago and they have not yet updated the TP-Link forum post with the new version.  If I am not able to get to this in the next couple of hours to update and test the changes, it will probably be tomorrow evening.

closed 2023-12-04 by mbentley

mbentley, 2023-12-04:
Images are up.  Let me know if there are any issues or if there are any references to 5.12 (or older) that I missed.

rudelm, 2023-12-04:
Thanks for the update, I've just updated the container from 5.12.7 to 5.13.22 without problems. From the Omada forums I've seen this post, that might explain why it was visible under us but not on en:
https://community.tp-link.com/en/business/forum/topic/617356?replyId=1277664

```
Generally speaking, the official firmware and release note will be pushed to the Cloud first, then the TP-Link global "EN" website, and slightly later to the TP-Link local websites (usually 1-2 weeks delay or so). You may download it from the "EN" website for the early access, or wait for the "US" website for releasing.
```

Assuming that the Controller updates follow the same pattern as the device firmware does.

mbentley, 2023-12-05:
Nice find.  Can’t see I’ve seen that before but it could be something new.  Pretty typical of SaaS vs on prem offerings though.

commit 3bef02c0 (2023-12-14): Merge pull request #379 from mbentley/add-5.13.22

Added 5.13.22; fixes #378 & fixes #377