mbentley/docker-omada-controller #383: [Bug]: Unable to restore config from backup

JesseWaas, 2023-12-10:
### Controller Version

v5.13.22

### Describe the Bug

Hi, I have noticed that a "Restore failed" error seems to be thrown when attempting to import/restore config files using the controller software run from the v5.13.22 Docker image. The error occurs when attempting to restore a previously backed up config file during initial setup or later through the system settings > maintenance menu. The issue does not appear to affect a native install of v5.13.22 from TP-Link (I can import the same config file there without issue).

Interestingly the issue only seems to affect restoring config backups made with recent versions of the controller. I appear to be able to import a backup from v5.9.9 for example, but not v5.13.22.

I have attached a screenshot of the error messages and a related Omada server.log excerpt that includes some exception details:

`java.lang.reflect.InaccessibleObjectException: Unable to make field private static final long java.util.LinkedHashMap.serialVersionUID accessible: module java.base does not "opens java.util" to unnamed module @5a6d67c3`

[omada_server_log_exrpt.txt](https://github.com/mbentley/docker-omada-controller/files/13626368/omada_server_log_exrpt.txt)

[image]

### Expected Behavior

Config file should be restored without error.

### Steps to Reproduce

(1) Deploy Omada controller using Docker image for v5.13.22.
(2) Complete minimal setup for local controller
(3) Navigate to `System Settings` > `Maintenance`
(4) Under the `Backup group` leave all settings at default (settings only, do not retain user info, export to local file) and select export
(5) Under `Restore` leave all settings at default (import from local file, retain device info), browse to the backup file that was just made, and then click `Restore`
(6) Observe that the restore operation fails (error condition described here).

### How You're Launching the Container

```plain
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  --ulimit nofile=4096:8192 \
  -p 8088:8088 \
  -p 8043:8043 \
  -p 8843:8843 \
  -p 27001:27001/udp \
  -p 29810:29810/udp \
  -p 29811-29816:29811-29816 \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e PGID="508" \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e PORT_ADOPT_V1=29812 \
  -e PORT_APP_DISCOVERY=27001 \
  -e PORT_DISCOVERY=29810 \
  -e PORT_MANAGER_V1=29811 \
  -e PORT_MANAGER_V2=29814 \
  -e PORT_TRANSFER_V2=29815 \
  -e PORT_RTTY=29816 \
  -e PORT_UPGRADE_V1=29813 \
  -e PUID="508" \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:5.13
```

### Container Logs

```plain
12-09-2023 17:34:30.576 INFO [https-jsse-nio-8043-exec-2] [] c.t.s.o.b.c.r.b.a(): omadacId=ffa97d1e5d1288da6105cc589c32c8de startRestoreStage, level CONTROLLER
12-09-2023 17:34:30.581 INFO [https-jsse-nio-8043-exec-2] [] c.t.s.o.b.a.c.a.a.f(): file: false, skipConfirm: false, skipDevice: false
12-09-2023 17:34:30.582 INFO [https-jsse-nio-8043-exec-2] [] c.t.s.o.b.a.c.a.a.f(): omadacId=ffa97d1e5d1288da6105cc589c32c8de restoring beginning...
12-09-2023 17:34:30.593 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.l(): omadacId=OmadacId(ffa97d1e5d1288da6105cc589c32c8de) may sdn backup file, try excute restore.
12-09-2023 17:34:30.598 INFO [restore-work-group-0] [] c.t.s.o.m.c.d.a(): base db version  5.13.20 is same than compatible version 5.13.20
12-09-2023 17:34:30.598 INFO [restore-work-group-0] [] c.t.s.o.b.c.u.b.m.c(): omadacId=OmadacId(ffa97d1e5d1288da6105cc589c32c8de) sdn backup file, db version is=5.13.20?softVer=5.13.22
12-09-2023 17:34:30.599 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=ffa97d1e5d1288da6105cc589c32c8de restore startRestoreItem
12-09-2023 17:34:30.668 WARN [restore-work-group-0] [] c.t.s.o.b.c.r.d(): omadacId=OmadacId(ffa97d1e5d1288da6105cc589c32c8de) restore sdn failed, do rollback.
12-09-2023 17:34:31.805 WARN [restore-work-group-0] [] c.t.s.o.b.c.u.s.m.a.a.b(): no maps were found in omadac OmadacId(ffa97d1e5d1288da6105cc589c32c8de), no need to execute clear
12-09-2023 17:34:32.713 INFO [restore-work-group-0] [] c.t.s.o.b.c.u.d.a(): start deleteDeviceData for OmadacId(ffa97d1e5d1288da6105cc589c32c8de)
12-09-2023 17:34:32.727 INFO [restore-work-group-0] [] c.t.s.o.m.d.d.m.d.g(): delete all deviceImage by omadacId:ffa97d1e5d1288da6105cc589c32c8de
12-09-2023 17:34:32.728 INFO [restore-work-group-0] [] c.t.s.o.b.c.u.d.a(): finish deleteDeviceData for OmadacId(ffa97d1e5d1288da6105cc589c32c8de)
12-09-2023 17:34:32.742 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=ffa97d1e5d1288da6105cc589c32c8de restore failed
12-09-2023 17:34:32.747 ERROR [restore-work-group-0] [] c.t.s.o.b.c.r.d(): omadacId=OmadacId(ffa97d1e5d1288da6105cc589c32c8de) restore failed with Exception
java.lang.reflect.InaccessibleObjectException: Unable to make field private static final long java.util.LinkedHashMap.serialVersionUID accessible: module java.base does not "opens java.util" to unnamed module @5a6d67c3
	at java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354) ~[?:?]
	at java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297) ~[?:?]
	at java.lang.reflect.Field.checkCanSetAccessible(Field.java:178) ~[?:?]
	at java.lang.reflect.Field.setAccessible(Field.java:172) ~[?:?]
	at com.tplink.smb.omada.backup.api.f.c.a(SourceFile:27) ~[backup-api-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.api.f.b.a(SourceFile:54) ~[backup-api-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.unit.a.a.a(SourceFile:31) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.restore.d.a(SourceFile:666) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.restore.d.a(SourceFile:567) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.restore.d.a(SourceFile:193) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.restore.l.a(SourceFile:153) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.restore.l.a(SourceFile:70) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.core.a.a.a(SourceFile:106) ~[backup-core-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.a.c.a.a.f.b(SourceFile:227) ~[maintenance-port-web-5.13.22.jar:5.13.22]
	at com.tplink.smb.component.lock.api.AbstractReadWriteLockService.doWithTryLockWrite(AbstractReadWriteLockService.java:107) ~[solution-components-lock-api-1.2.0-20231116.185612-174.jar:1.2.0-SNAPSHOT]
	at com.tplink.smb.omada.lock.a.a.b(SourceFile:77) ~[omada-lock-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.lock.a.a.b(SourceFile:92) ~[omada-lock-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.a.c.a.a.f.a(SourceFile:224) ~[maintenance-port-web-5.13.22.jar:5.13.22]
	at com.tplink.smb.omada.backup.a.c.a.a.f.a(SourceFile:164) ~[maintenance-port-web-5.13.22.jar:5.13.22]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) ~[?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[?:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
12-09-2023 17:34:32.753 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=ffa97d1e5d1288da6105cc589c32c8de restore failed
```

### Additional Context

Possibly related SO article: https://stackoverflow.com/questions/70756414/java-lang-reflect-inaccessibleobjectexception-unable-to-make-field-private-fina

mbentley, 2023-12-10:
Thanks for the detailed report and the SO article!  I appreciate the research and this one is easy to fix.

mbentley, 2023-12-13:
I had to revert this commit because it was breaking 32 bit arm due to it still being on Java 8.  I don't have time to fix it right now properly but I will.

@JesseWaas - you can override the CMD from the Docker image, setting it to:
`/usr/bin/java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true --add-opens java.base/java.util=ALL-UNNAMED -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain`

Which will give you the same functionality as the PR I had to revert.

mbentley, 2023-12-14:
OK, this should actually be fixed in a way it doesn't break armv7l once the images build.

commit 89aa5e59 (2023-12-10): Added JVM arg to allow restore to function; fixes #383
commit bc86bddd (2023-12-10): Merge pull request #384 from mbentley/fix-restore-fail

Added JVM arg to allow restore to function; fixes #383
commit 19e82773 (2023-12-13): Revert "Added JVM arg to allow restore to function; fixes #383"
commit 536b0820 (2023-12-13): Merge pull request #387 from mbentley/revert-384-fix-restore-fail

Revert "Added JVM arg to allow restore to function; fixes #383"
closed 2023-12-14 by mbentley