mbentley/docker-omada-controller #385: [Bug]: Could not create the Java Virtual Machine

SelfHostedJawn, 2023-12-12:
### Controller Version

5.13.22

### Describe the Bug

I am getting the following message in my log after updating to the latest version:

Unrecognized option: --add-opens
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.

I tried going back a version to 5.12 and 5.8 but kept getting the same error.

### Expected Behavior

Run like normal. Cannot connect to the controller currently.

### Steps to Reproduce

Running the docker-compose file using docker-compose up -d. 

### How You're Launching the Container

```plain
Running the docker-compose file using docker-compose up -d.
```

### Container Logs

```plain
WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
INFO: Version check passed; image version (5.13.22) >= the last version ran (5.13.22); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
Unrecognized option: --add-opens
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
```

### MongoDB Logs

```plain
2023-12-10T20:26:16.448+0000 [signalProcessingThread] got signal 15 (Terminated), will terminate after current cmd ends
2023-12-10T20:26:16.448+0000 [signalProcessingThread] now exiting
2023-12-10T20:26:16.448+0000 [signalProcessingThread] dbexit: 
2023-12-10T20:26:16.449+0000 [signalProcessingThread] shutdown: going to close listening sockets...
2023-12-10T20:26:16.449+0000 [signalProcessingThread] closing listening socket: 8
2023-12-10T20:26:16.449+0000 [signalProcessingThread] closing listening socket: 11
2023-12-10T20:26:16.449+0000 [signalProcessingThread] removing socket file: /tmp/mongodb-27217.sock
2023-12-10T20:26:16.450+0000 [initandlisten] Port 27217 is no longer valid
2023-12-10T20:26:16.457+0000 [signalProcessingThread] shutdown: going to flush diaglog...
2023-12-10T20:26:16.457+0000 [signalProcessingThread] shutdown: going to close sockets...
2023-12-10T20:26:16.457+0000 [signalProcessingThread] shutdown: waiting for fs preallocator...
2023-12-10T20:26:16.457+0000 [signalProcessingThread] shutdown: lock for final commit...
2023-12-10T20:26:16.457+0000 [signalProcessingThread] shutdown: final commit...
2023-12-10T20:26:16.458+0000 [signalProcessingThread] shutdown: closing all files...
2023-12-10T20:26:16.458+0000 [conn15] end connection 127.0.0.1:46982 (5 connections now open)
2023-12-10T20:26:16.458+0000 [conn7] end connection 127.0.0.1:46662 (5 connections now open)
2023-12-10T20:26:16.458+0000 [conn12] end connection 127.0.0.1:46848 (4 connections now open)
2023-12-10T20:26:16.459+0000 [conn8] end connection 127.0.0.1:46664 (3 connections now open)
2023-12-10T20:26:16.459+0000 [conn14] end connection 127.0.0.1:46984 (1 connection now open)
2023-12-10T20:26:16.463+0000 [conn13] end connection 127.0.0.1:46970 (0 connections now open)
2023-12-10T20:26:16.467+0000 [signalProcessingThread] closeAllFiles() finished
2023-12-10T20:26:16.467+0000 [signalProcessingThread] journalCleanup...
2023-12-10T20:26:16.467+0000 [signalProcessingThread] removeJournalFiles
2023-12-10T20:26:16.487+0000 [signalProcessingThread] shutdown: removing fs lock...
2023-12-10T20:26:16.487+0000 [signalProcessingThread] dbexit: really exiting now
```

### Additional Context

docker inspect omada-controller --format '{{json .Image}}' ->
"sha256:0f68d38a0f8e29e80f734b879c45931af5c11f5d1e5e614e89adf64b527e557e"

Still running on 32bit raspberry pi (Linux raspberrypi 5.10.103-v7l+). Will upgrade in the future, but not right now.

mbentley, 2023-12-12:
Ah shoot.  The 32 bit version still has Java 8 which does not have the same flag.  The flag isn't required for Java 8.

Temporarily when you run the container, use this as a CMD by adding this to your compose file:

```
command: ["/usr/bin/java","-server","-Xms128m","-Xmx1024m","-XX:MaxHeapFreeRatio=60","-XX:MinHeapFreeRatio=30","-XX:+HeapDumpOnOutOfMemoryError","-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof","-Djava.awt.headless=true","-cp","/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:","com.tplink.smb.omada.starter.OmadaLinuxMain"]
```

This should be at the same indention level as the `image:` line.

mbentley, 2023-12-12:
https://github.com/mbentley/docker-omada-controller/pull/384 is where the problem was introduced; this was to fix a backup & restore issue.

ProXM1, 2023-12-13:
Still not working (same issue as above running on RPi 32bits).

Thank you!

SelfHostedJawn, 2023-12-13:
After adding the command to the docker compose file, I get the following:
`12-13-2023 18:47:03.316 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
12-13-2023 18:47:03.319 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started
`
and 
`# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb3112be0, pid=1, tid=0x6061c470
#
# JRE version: OpenJDK Runtime Environment (8.0_292-b10) (build 1.8.0_292-8u292-b10-0ubuntu1~16.04.1-b10)
# Java VM: OpenJDK Client VM (25.292-b10 mixed mode linux-aarch32 )
# Problematic frame:
# j  com.tplink.smb.omada.client.domain.a.z.b(Lcom/tplink/smb/omada/client/domain/model/clientimage/e;)Lcom/tplink/smb/omada/client/common/api/internal/dto/ClientAuthStatusResultDTO;+244
#
# Core dump written. Default location: /opt/tplink/EAPController/lib/core or core.1
#
# An error report file with more information is saved as:
# /tmp/hs_err_pid1.log`

mbentley, 2023-12-14:
OK, this should now be fixed in the PR https://github.com/mbentley/docker-omada-controller/pull/388 once the images build.

mbentley, 2023-12-14:
Let me know if it is still not solved.

closed 2023-12-14 by mbentley

ProXM1, 2023-12-14:
Just pulled the image again and got this:

`Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.`

Am I doing something wrong?

mbentley, 2023-12-14:
Which image tag did you pull (full name)?

And what's the image digest?
`docker inspect <container-name> --format '{{json .Image}}'`

What about your raspberry pi?
what OS version - `cat /etc/os-release`
details about your Docker version - `docker info`

Could I get the full logs from the container?

I currently do not have a 32 bit raspberry pi install anywhere so I have to test with qemu to emulate which works fine here from the last image update.  That image digest is `"sha256:ccc432752ae7bebf0f50e593905a12d1030f1a7201142706b29466fa5bae74ce"`

ProXM1, 2023-12-14:
## Questions:

Image tag:
`mbentley/omada-controller:latest-armv7l`
Image digest:
`sha256:ccc432752ae7bebf0f50e593905a12d1030f1a7201142706b29466fa5bae74ce`

## RPi info:

### OS Version:
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
```
### Docker info:
```
Client: Docker Engine - Community
 Version:    24.0.7
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.11.2
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.21.0
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 5
  Running: 4
  Paused: 0
  Stopped: 1
 Images: 5
 Server Version: 24.0.7
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: d8f198a4ed8892c764191ef7b3b06d8a2eeb5c7f
 runc version: v1.1.10-0-g18a0cb0
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 5.10.103-v7l+
 Operating System: Raspbian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: armv7l
 CPUs: 4
[... 92 lines cut ...]
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
INFO: Version check passed; image version (5.13.22) >= the last version ran (5.13.22); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
Unrecognized option: --add-opens
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
```

**Once again, thank you for your help!**

mbentley, 2023-12-14:
Are you using some sort of tool to launch and manage the container?  It seems like your container has retained the old command as the image does have the correct Cmd defined:

```
$ docker inspect mbentley/omada-controller:latest-armv7l | jq -r '.[]|.Config.Cmd'
[
  "/usr/bin/java",
  "-server",
  "-Xms128m",
  "-Xmx1024m",
  "-XX:MaxHeapFreeRatio=60",
  "-XX:MinHeapFreeRatio=30",
  "-XX:+HeapDumpOnOutOfMemoryError",
  "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof",
  "-Djava.awt.headless=true",
  "-cp",
  "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:",
  "com.tplink.smb.omada.starter.OmadaLinuxMain"
]
```

You might want to run the same command as I did and compare that against your container.  I imagine your container is going to show something like this where it still has the `--add-opens` argument:

```
# docker inspect omada-controller | jq '.[]|.Config.Cmd'
[
  "/usr/bin/java",
  "-server",
  "-Xms128m",
  "-Xmx1024m",
  "-XX:MaxHeapFreeRatio=60",
  "-XX:MinHeapFreeRatio=30",
  "-XX:+HeapDumpOnOutOfMemoryError",
  "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof",
  "-Djava.awt.headless=true",
  "--add-opens",
  "java.base/java.util=ALL-UNNAMED",
  "-cp",
  "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:",
  "com.tplink.smb.omada.starter.OmadaLinuxMain"
]
```

ProXM1, 2023-12-14:
I'm using Portainer to manage my Docker environment.

Also that did kind of the trick to get things started but it's still not working:
```
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
INFO: Version check passed; image version (5.13.22) >= the last version ran (5.13.22); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
error: exec: "$": executable file not found in $PATH
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
INFO: Version check passed; image version (5.13.22) >= the last version ran (5.13.22); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
error: exec: "$": executable file not found in $PATH
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
INFO: Version check passed; image version (5.13.22) >= the last version ran (5.13.22); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
error: exec: "$": executable file not found in $PATH
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
[... 10 lines cut ...]
INFO: Starting Omada Controller as user omada
error: exec: "$": executable file not found in $PATH
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'America/New_York'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
[... cut ...]
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
INFO: Version check passed; image version (5.13.22) >= the last version ran (5.13.22); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
error: exec: "$": executable file not found in $PATH
```

mbentley, 2023-12-14:
The CMD now seems to literally be set to a $ or begin with a $.

ProXM1, 2023-12-14:
LOL, literally leaving it all in default did the trick... 

Thank you so much for your help and patience!