mbentley/docker-omada-controller #397: [Feature]: Update Controller to v5.13.23

obsidiangroup, 2024-01-13:
### What problem are you looking to solve?

Would like to see a container with the 5.13.23 release available.

mbentley, 2024-01-13:
Where are you seeing a 5.13.23?  I only see 5.13.22.

sutram, 2024-01-13:
I saw it in the controller UI and was able to down the tarball.

Omada_SDN_Controller_v5.13.23_linux_x64_20231228194009_1705039261727.tar.gz

obsidiangroup, 2024-01-13:
When I logged into my controller instance, it prompted me to upgrade and like @sutram was able to download the tar.  But when going directly to the omada site, I could not locate it.

obsidiangroup, 2024-01-14:
I've built the container locally with the 5.13.23 version.  Everything seems stable but will report back in a day or so.
[image]

mbentley, 2024-01-15:
It has finally been published (see #400).  Closing in favor of that for now.  Will get the new version published this morning.

closed 2024-01-15 by mbentley

mbentley, 2024-01-15:
The new version has been released.

javito1081, 2024-01-22:
im using docker compose but i dont know how to update to 5.13.23 :-( i tried stopping the docker compose and remove the image then restart the docker compose and is still not updating, can someone please tell me the procedure im missing?