mbentley/docker-omada-controller #404: [Bug]: Unable to restore config from backup

mariolos, 2024-01-15:
### Controller Version

5.13.23

### Describe the Bug

Hi, I 
have a problem with restoring settings from backup file (file was exported  today) . 

Exception line:
"java.lang.reflect.InaccessibleObjectException: Unable to make field private static final long java.util.LinkedHashMap.serialVersionUID accessible: module java.base does not "opens java.util" to unnamed module @5a6d67c3"

### Expected Behavior

Config file should be restored without error.

### Steps to Reproduce

Clean install latest version.
Try to restore from fresh backup.

### How You're Launching the Container

```plain
standard way by docker-compose
```

### Container Logs

```plain
024/01/15 20:43:50	stdout	java.lang.reflect.InaccessibleObjectException: Unable to make field private static final long java.util.LinkedHashMap.serialVersionUID accessible: module java.base does not "opens java.util" to unnamed module @5a6d67c3
2024/01/15 20:43:50	stdout	01-15-2024 19:43:50.069 ERROR [restore-work-group-0] [] c.t.s.o.b.c.r.d(): omadacId=OmadacId(0a0b7c040042eac2c227bbf5f592e68c) restore failed with Exception
2024/01/15 20:43:50	stdout	01-15-2024 19:43:50.061 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=0a0b7c040042eac2c227bbf5f592e68c restore failed
```

DoubleZed, 2024-04-01:
Hi, I have the same issue (on controller 5.13.30) even with a config file literally created 4 minutes ago, but I see way more error messages. I also see that when using the backup recovery on an existing environment, the recovery process actually starts deleting the configuration from the environment (I am guessing it passes initial validation) and then fails after it has reached about 4% (that is the last value visible on the progress bar) and then comes back with a message in the UI that Recovery Failed.

**Here are the container logs**
`
04-01-2024 21:16:23.346 INFO [backup-work-thread] [] c.t.s.o.b.a.c.a.a.c(): omadacId=OmadacId(a4de91538e99302bdba890943c7b876c) execute backup...
04-01-2024 21:16:23.714 INFO [backup-work-thread] [] c.t.s.o.b.a.c.a.a.c(): omadacId=OmadacId(a4de91538e99302bdba890943c7b876c) backup ending...
04-01-2024 21:20:20.761 INFO [https-jsse-nio-8043-exec-3] [] c.t.s.o.b.c.r.b.a(): omadacId=a4de91538e99302bdba890943c7b876c startRestoreStage, level CONTROLLER
04-01-2024 21:20:20.761 INFO [https-jsse-nio-8043-exec-3] [] c.t.s.o.b.a.c.a.a.f(): file: false, skipConfirm: false, skipDevice: false
04-01-2024 21:20:20.761 INFO [https-jsse-nio-8043-exec-3] [] c.t.s.o.b.a.c.a.a.f(): omadacId=a4de91538e99302bdba890943c7b876c restoring beginning...
04-01-2024 21:20:20.765 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.l(): omadacId=OmadacId(a4de91538e99302bdba890943c7b876c) may sdn backup file, try excute restore.
04-01-2024 21:20:20.768 INFO [restore-work-group-0] [] c.t.s.o.m.c.d.a(): base db version  5.13.30 is same than compatible version 5.13.30
04-01-2024 21:20:20.768 INFO [restore-work-group-0] [] c.t.s.o.b.c.u.b.m.c(): omadacId=OmadacId(a4de91538e99302bdba890943c7b876c) sdn backup file, db version is=5.13.30?softVer=5.13.30.8
04-01-2024 21:20:20.768 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=a4de91538e99302bdba890943c7b876c restore startRestoreItem
04-01-2024 21:20:20.805 WARN [restore-work-group-0] [] c.t.s.o.b.c.r.d(): omadacId=OmadacId(a4de91538e99302bdba890943c7b876c) restore sdn failed, do rollback.
04-01-2024 21:20:22.182 INFO [restore-work-group-0] [] c.t.s.o.b.c.u.d.a(): start deleteDeviceData for OmadacId(a4de91538e99302bdba890943c7b876c)
04-01-2024 21:20:22.188 INFO [restore-work-group-0] [] c.t.s.o.m.d.d.m.d.g(): delete all deviceImage by omadacId:a4de91538e99302bdba890943c7b876c
04-01-2024 21:20:22.189 INFO [restore-work-group-0] [] c.t.s.o.b.c.u.d.a(): finish deleteDeviceData for OmadacId(a4de91538e99302bdba890943c7b876c)
04-01-2024 21:20:22.246 INFO [restore-work-group-0] [] c.t.s.o.m.c.d.m.s.a(): Site SiteId(651803af19e6995fcad31656) create success in omadac OmadacId(a4de91538e99302bdba890943c7b876c).
04-01-2024 21:20:22.247 INFO [comm-pool-15] [] c.t.s.o.c.c.p.e.c(): Observed SiteManageEvent on ClientCacheHandler: com.tplink.smb.omada.client.common.port.eventcenter.dto.SiteManageToClientDTO@40fb38bb.
04-01-2024 21:20:22.248 INFO [log-site-manage-work-group-1] [] c.t.s.o.l.p.c.e.i(): omadacId=a4de91538e99302bdba890943c7b876c,siteId=651803af19e6995fcad31656,siteName=SWE,operation=create, modify site info of users.
04-01-2024 21:20:22.249 INFO [common-block-pool-1] [] c.t.s.o.c.c.p.e.m(): Observed SiteManageEvent: com.tplink.smb.omada.client.common.port.eventcenter.dto.SiteManageToClientDTO@59413f00.
04-01-2024 21:20:22.249 INFO [domainEvent-saml-handle-work-group-1] [] c.t.s.o.i.d.m.s.u.a.g(): omadacId=a4de91538e99302bdba890943c7b876c,siteId=651803af19e6995fcad31656,siteName=SWE,operation=create, modify site info of users.
04-01-2024 21:20:22.252 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=a4de91538e99302bdba890943c7b876c restore failed
04-01-2024 21:20:22.253 ERROR [restore-work-group-0] [] c.t.s.o.b.c.r.d(): omadacId=OmadacId(a4de91538e99302bdba890943c7b876c) restore failed with Exception
java.lang.reflect.InaccessibleObjectException: Unable to make field private static final long java.util.LinkedHashMap.serialVersionUID accessible: module java.base does not "opens java.util" to unnamed module @365f33e3
	at java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354) ~[?:?]
	at java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297) ~[?:?]
	at java.lang.reflect.Field.checkCanSetAccessible(Field.java:178) ~[?:?]
	at java.lang.reflect.Field.setAccessible(Field.java:172) ~[?:?]
	at com.tplink.smb.omada.backup.api.f.c.a(SourceFile:27) ~[backup-api-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.api.f.b.a(SourceFile:54) ~[backup-api-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.unit.a.a.a(SourceFile:31) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.restore.d.a(SourceFile:666) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.restore.d.a(SourceFile:567) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.restore.d.a(SourceFile:193) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.restore.l.a(SourceFile:153) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.restore.l.a(SourceFile:70) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.core.a.a.a(SourceFile:106) ~[backup-core-5.13.30.8.jar:5.13.30.8]
	at com.tplink.smb.omada.backup.a.c.a.a.f.b(SourceFile:227) ~[maintenance-port-web-5.13.30.8.jar:5.13.30.8]
[... cut ...]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[?:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
04-01-2024 21:20:22.256 INFO [restore-work-group-0] [] c.t.s.o.b.c.r.b.a(): omadacId=a4de91538e99302bdba890943c7b876c restore failed`

mbentley, 2024-04-01:
Ah, I totally missed this because it looks to be a duplicate of https://github.com/mbentley/docker-omada-controller/issues/383 which was fixed in https://github.com/mbentley/docker-omada-controller/pull/389.  Could you verify what the CMD is from your docker container?  A `docker inspect <container-name-or-id> --format '{{json .Args}}'` will get those.  It should be something like this:

```json
["/usr/bin/java","-server","-Xms128m","-Xmx1024m","-XX:MaxHeapFreeRatio=60","-XX:MinHeapFreeRatio=30","-XX:+HeapDumpOnOutOfMemoryError","-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof","-Djava.awt.headless=true","--add-opens","java.base/java.util=ALL-UNNAMED","-cp","/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:","com.tplink.smb.omada.starter.OmadaLinuxMain"]
```

Or when prettied up with `jq` (like `docker inspect omada-controller --format '{{json .Args}}' | jq .`) so it is easier to read:

```json
[
  "/usr/bin/java",
  "-server",
  "-Xms128m",
  "-Xmx1024m",
  "-XX:MaxHeapFreeRatio=60",
  "-XX:MinHeapFreeRatio=30",
  "-XX:+HeapDumpOnOutOfMemoryError",
  "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof",
  "-Djava.awt.headless=true",
  "--add-opens",
  "java.base/java.util=ALL-UNNAMED",
  "-cp",
  "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:",
  "com.tplink.smb.omada.starter.OmadaLinuxMain"
]
```

The addition of `--add-opens java.base/java.util=ALL-UNNAMED` should have fixed this.

DoubleZed, 2024-04-02:
Here is the output of the command:
```
[
  "/usr/bin/java",
  "-server",
  "-Xms128m",
  "-Xmx1024m",
  "-XX:MaxHeapFreeRatio=60",
  "-XX:MinHeapFreeRatio=30",
  "-XX:+HeapDumpOnOutOfMemoryError",
  "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof",
  "-Djava.awt.headless=true",
  "-cp",
  "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:",
  "com.tplink.smb.omada.starter.OmadaLinuxMain"
]
```
It does seem to be missing the `--add-opens java.base/java.util=ALL-UNNAMED`. What should I do to fix it?

mbentley, 2024-04-02:
Good deal - glad it was a relatively quick and easy fix.  I thought this sounded familiar :)

mbentley, 2024-04-02:
@mariolos - I imagine that this is the cause & fix for you as well so I am going to close this issue out.

closed 2024-04-02 by mbentley