mbentley/docker-omada-controller #405: [Bug]: omada ignores http_proxy settings

xlemassacre, 2024-01-24:
### Controller Version

v5.13.23

### Describe the Bug

To have internet access via proxy I have added http_proxy environment to the container.
When i'm attaching to the container and use wget to request internet resources it goes to the proxy. But the traffic from omada (e.g. downloading firmware) is not send send to the proxy, but directly on the host network.

### Expected Behavior

proxy variables should be considered

### Steps to Reproduce

1. set container environments for proxy (http_proxy, https_proxy, no_proxy) to a proxy
2. in omada ui search for firmware updates or trigger a firmware upgrade

### How You're Launching the Container

```plain
containers:
      - env:
        - name: MANAGE_HTTPS_PORT
          value: "8043"
        - name: MANAGE_HTTP_PORT
          value: "8088"
        - name: PGID
          value: "508"
        - name: PGROUP
          value: omada
        - name: PORTAL_HTTPS_PORT
          value: "8843"
        - name: PORTAL_HTTP_PORT
          value: "8088"
        - name: SHOW_MONGODB_LOGS
          value: "true"
        - name: SHOW_SERVER_LOGS
          value: "true"
        - name: TZ
          value: Europe/Berlin
        - name: http_proxy
          value: http://192.168.1.17:3128
        - name: https_proxy
          value: http://192.168.1.17:3128
        - name: no_proxy
          value: svc,local,localhost,maschin,192.168.1.13,10.152.183.0/24

this is a kubernetes deployment
```

### Container Logs

```plain
24-2024 20:53:53.484 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): upgrade task upgradeId [28a1ab74778748dfa6c2a4fcfc31d647] of OmadacId(753ae46f108c36e6569cbbb1f8c5153c) is running...
01-24-2024 20:53:56.417 INFO [https-jsse-nio-8043-exec-9] [] c.t.s.c.l.a.AbstractReadWriteLockService(): [readWriteLockService]businessId:omadac.id:753ae46f108c36e6569cbbb1f8c5153c get writeLock module:manager:upgrade.online.check fail, execute onFail
```

mbentley, 2024-01-24:
My recommendation that could fix it today without needing to wait for me would be overriding the default CMD to add in your proxy settings directly to the java args:

https://docs.oracle.com/en/java/javase/17/docs/api/java.base/java/net/doc-files/net-properties.html

Here's the default:

`/usr/bin/java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true --add-opens java.base/java.util=ALL-UNNAMED -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain`

Try adding something like:

`-Dhttp.proxyHost=192.168.1.17 -Dhttp.proxyPort=3128 -Dhttp.nonProxyHosts="svc,local,localhost,maschin,192.168.1.13,10.152.183.0/24" -Dhttps.proxyHost=192.168.1.17 -Dhttps.proxyPort=3128 -Dhttps.nonProxyHosts="svc,local,localhost,maschin,192.168.1.13,10.152.183.0/24"`

xlemassacre, 2024-01-30:
Thanks for the response. As the CMD is part of the Dockerfile I guess that means that I would need to modify your Dockerfile and build my own container?

Or is there another way to pass this additional arguments?

mbentley, 2024-01-30:
You can override the command in a k8s deployment.  Been a while since I have looked up docs on this but https://yuminlee2.medium.com/kubernetes-command-and-arguments-in-pod-c3f1be61ba1a might explain how to do this.

xlemassacre, 2024-02-13:
I tried to overwrite it but it somehow didn't work and the container didn't start anymore. I guess I wait until it can be fixed directly in the image.