mbentley/docker-omada-controller #418: [Bug]: Beta / Pre-Release 5.14.0.11 does not start up

IAmKonni, 2024-03-25:
### Controller Version

5.14.0.11

### Describe the Bug

I switched from tag 5.13 to beta to test the newest pre-release. I never had any problems in updating to newer versions, this is the very first time i'm facing a problem.
The controller does not start up, it runs into a boot loop (see logs).

### Expected Behavior

Starting like always. =)

### Steps to Reproduce

Change tag in compose file from 5.13 to beta and run `docker-compose up -d`

### How You're Launching the Container

```plain
version: "3"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:beta
    ports:
      - "8043:8043/tcp"
      - "443:443/tcp"
      - "8488:8488/tcp"
      - "8443:8443/tcp"
      - "8483:8483/tcp"
      - "29810:29810/udp"
      - "29811:29811"
      - "29812:29812"
      - "29813:29813"
      - "29814:29814"

    environment:
      MANAGE_HTTP_PORT: 8488
      MANAGE_HTTPS_PORT: 8443
      PORTAL_HTTP_PORT: 8488
      PORTAL_HTTPS_PORT: 8483
      SHOW_SERVER_LOGS: 'true'
      SHOW_MONGODB_LOGS: 'false'
      SSL_CERT_NAME: 'tls.crt'
      SSL_KEY_NAME: 'tls.key'
      TZ: 'Europe/Berlin'
    volumes:
      - ./data:/opt/tplink/EAPController/data
      - ./work:/opt/tplink/EAPController/work
      - ./logs:/opt/tplink/EAPController/logs
    restart: unless-stopped
```

### Container Logs

```plain
03-25-2024 18:32:38.365 ERROR [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): Cannot retry start up springboot. reson:Error creating bean with name 'com.tplink.smb.omada.identityaccess.domain.model.openapi.c.q': Unsatisfied dependency expressed through field 'e'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.common.a.a': Unsatisfied dependency expressed through field 'c'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.common.e.a': Unsatisfied dependency expressed through field 'b'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.b.p': Unsatisfied dependency expressed through field 'm'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.domain.model.user.A': Unsatisfied dependency expressed through field 'c'; nested exception is org.springframework.beans.factory.BeanCurrentlyInCreationException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.mongo.adaptor.persistence.tenant.a': Bean with name 'com.tplink.smb.omada.identityaccess.port.mongo.adaptor.persistence.tenant.a' has been injected into other beans [com.tplink.smb.omada.identityaccess.domain.model.d.q] in its raw version as part of a circular reference, but has eventually been wrapped. This means that said other beans do not use the final version of the bean. This is often the result of over-eager type matching - consider using 'getBeanNamesForType' with the 'allowEagerInit' flag turned off, for example.
03-25-2024 18:32:38.365 ERROR [main] [] c.t.s.o.s.t.FailExitTask(): Failed to start omada controller, going to exit
03-25-2024 18:32:38.366 INFO [Thread-0] [] c.t.s.o.s.OmadaBootstrap(): Failed to shutdown customThread.
java.lang.ExceptionInInitializerError: null
	at com.tplink.smb.omada.common.concurrent.thread.a.a(SourceFile:243) ~[omada-common-5.14.0.11.jar:5.14.0.11]
	at com.tplink.smb.omada.starter.OmadaBootstrap.b(SourceFile:195) ~[local-starter-5.14.0.11.jar:5.14.0.11]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.b(SourceFile:87) ~[local-starter-5.14.0.11.jar:5.14.0.11]
	at java.lang.Thread.run(Thread.java:840) [?:?]
Caused by: java.lang.IllegalStateException: org.springframework.boot.web.servlet.context.AnnotationConfigServletWebServerApplicationContext@398f2bd3 has not been refreshed yet
	at org.springframework.context.support.AbstractApplicationContext.assertBeanFactoryActive(AbstractApplicationContext.java:1147) ~[spring-context-5.3.30.jar:5.3.30]
	at org.springframework.context.support.AbstractApplicationContext.getBean(AbstractApplicationContext.java:1159) ~[spring-context-5.3.30.jar:5.3.30]
	at com.tplink.smb.omada.common.spring.a.a(SourceFile:28) ~[omada-common-5.14.0.11.jar:5.14.0.11]
[... 137 lines cut ...]
2024-03-25T18:29:48.178+0100 I NETWORK  [conn22] received client metadata from 127.0.0.1:32780 conn22: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.0-101-generic" }, platform: "Java/Private Build/17.0.10+7-Ubuntu-120.04.1" }
2024-03-25T18:29:48.178+0100 I NETWORK  [conn21] received client metadata from 127.0.0.1:32776 conn21: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.0-101-generic" }, platform: "Java/Private Build/17.0.10+7-Ubuntu-120.04.1" }
2024-03-25T18:29:48.179+0100 I NETWORK  [listener] connection accepted from 127.0.0.1:32792 #23 (5 connections now open)
2024-03-25T18:29:48.179+0100 I NETWORK  [conn23] received client metadata from 127.0.0.1:32792 conn23: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.0-101-generic" }, platform: "Java/Private Build/17.0.10+7-Ubuntu-120.04.1" }
2024-03-25T18:29:48.183+0100 I NETWORK  [conn23] end connection 127.0.0.1:32792 (4 connections now open)
2024-03-25T18:29:48.183+0100 I NETWORK  [conn21] end connection 127.0.0.1:32776 (3 connections now open)
2024-03-25T18:29:48.183+0100 I NETWORK  [conn22] end connection 127.0.0.1:32780 (2 connections now open)
2024-03-25T18:29:48.691+0100 I NETWORK  [listener] connection accepted from 127.0.0.1:32800 #24 (3 connections now open)
2024-03-25T18:29:48.691+0100 I NETWORK  [conn24] received client metadata from 127.0.0.1:32800 conn24: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.0-101-generic" }, platform: "Java/Private Build/17.0.10+7-Ubuntu-120.04.1" }
2024-03-25T18:29:48.802+0100 I NETWORK  [conn24] end connection 127.0.0.1:32800 (2 connections now open)
2024-03-25T18:29:48.803+0100 I NETWORK  [conn16] end connection 127.0.0.1:60954 (1 connection now open)
2024-03-25T18:29:48.803+0100 I NETWORK  [conn17] end connection 127.0.0.1:60952 (0 connections now open)
2024-03-25T18:29:48.817+0100 I CONTROL  [signalProcessingThread] got signal 15 (Terminated), will terminate after current cmd ends
2024-03-25T18:29:48.817+0100 I NETWORK  [signalProcessingThread] shutdown: going to close listening sockets...
2024-03-25T18:29:48.817+0100 I NETWORK  [signalProcessingThread] removing socket file: /tmp/mongodb-27217.sock
2024-03-25T18:29:48.817+0100 I FTDC     [signalProcessingThread] Shutting down full-time diagnostic data capture
2024-03-25T18:29:48.818+0100 I STORAGE  [signalProcessingThread] WiredTigerKVEngine shutting down
2024-03-25T18:29:48.837+0100 I STORAGE  [signalProcessingThread] shutdown: removing fs lock...
2024-03-25T18:29:48.837+0100 I CONTROL  [signalProcessingThread] now exiting
2024-03-25T18:29:48.837+0100 I CONTROL  [signalProcessingThread] shutting down with code:0
```

### Additional Context

The mongo db log is the first iteration after switching to 5.14 beta.

CoronaExtra, 2024-03-26:
I can report the same problem

mbentley, 2024-03-30:
I'll have to look into this when I am able to as I'm traveling now.  Not immediately sure if there is just some java arg that needs to be set or what.

IAmKonni, 2024-03-30:
No hurry, I'm totally fine with 5.13. :)

mbentley, 2024-03-31:
OK, so at least when starting up a new controller directly on the beta 5.14.0.11, I don't have startup issues.  I also started up a new 5.13, did a minimal setup of the controller and then switched it to the latest beta and I don't see a startup problem with that basic upgrade scenario.  I don't run the betas on my home environment just because I need my home network to be stable.

From whoever it running into the issue:

- What architecture are you running - amd64, arm64, or armv7l?  I've only tested amd64 and arm64 just now.
- Any idea what version of the controller you originally started with?  I might need to check to see if this is something that comes up only due to upgrade issues
- Could you also provide the `docker inspect <container-name-or-id> --format '{{json .Args}}'` output for me?  I just want to verify the container's command used.

IAmKonni, 2024-04-03:
I'm running an amd64 architecture (Intel NUC 11).
As far as I can remember I probably started with an 4.x version (maybe 4.2 or 4.3?) and updated them all iteratively (mostly i used "latest"-tag). Since you are offering beta versions I'm using distinct tags (e.g. "5.13").

`docker inspect` says:
```
["/usr/bin/java","-server","-Xms128m","-Xmx1024m","-XX:MaxHeapFreeRatio=60","-XX:MinHeapFreeRatio=30","-XX:+HeapDumpOnOutOfMemoryError","-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof","-Djava.awt.headless=true","--add-opens","java.base/java.util=ALL-UNNAMED","-cp","/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:","com.tplink.smb.omada.starter.OmadaLinuxMain"]
```

mbentley, 2024-04-03:
> Should I run again with 5.14?

No, that's ok.  I just wanted to make sure that the CMD hadn't been modified from the default but since yours matches for 5.13, it would be fine for 5.14 regardless.  Some tools used to manage containers will take the CMD from the image and hard code it to the container definition so if you update the image, it actually will not update the CMD.  At least it helps eliminate that as a possible reason why it isn't starting.

mbentley, 2024-04-03:
On a quick test, I did a basic deployment of a new controller and upgraded, skipping minor versions to hopefully reproduce it quickly:

```
mbentley/omada-controller:4.2
mbentley/omada-controller:4.4
mbentley/omada-controller:5.13
mbentley/omada-controller:beta
```

No such luck in this case.  I will have to see if I can try every version it seems.

mbentley, 2024-04-04:
Well I was hoping that going through all of the versions would help reproduce it but I went through all of these versions and the beta started correctly:

```
4.2, 4.3, 4.4, 5.0, 5.1, 5.3, 5.4, 5.5, 5.6, 5.7, 5.8, 5.9, 5.12, 5.13, beta
```

IAmKonni, 2024-04-04:
I don't care too much about this beta, but I hope this won't be the same with 5.14 final, when it arrives. :)
Maybe it has to do with some of my settings. In the worst case I have to do a new setup and start with a clean sheet.
Thank you for all your efforts so far. That's not a matter of course.

ruudkobes, 2024-04-11:
I can confirm the same issue: upgraded from the previous beta to this latest beta.
After updating  it fails to start with logs looking identical to @IAmKonni's above, same bean creation failure.

Starts and runs fine back on 5.13.

Running on AMD64

eblieb, 2024-05-09:
I stopped my 5.13 container and built a new container with
docker run -d \
  --name omada-controller-beta \
  --stop-timeout 60 \
  --restart unless-stopped \
  --net host \
  -e MANAGE_HTTP_PORT=8088 \
 -v omada-data:/opt/tplink/EAPController/data \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:beta

and it is running just fine.

eblieb, 2024-05-29:
FYI a new beta was released. Waiting for the docker container to be updated 

https://community.tp-link.com/en/business/forum/topic/663452

mbentley, 2024-05-30:
I created https://github.com/mbentley/docker-omada-controller/pull/429 to handle the beta install.  I will get that merged and hopefully it will build successfully.  I just hate changing the install.sh because it causes all of the images to be re-built due to the changes.

mbentley, 2024-05-31:
As reported in #430, it also has a similar issue that I can't reproduce.

mbentley, 2024-05-31:
@eblieb - could you provide the full logs from the beginning of the startup of the container so I can try to submit an issue on the TP-Link forums to see if they can provide an idea of why this is happening?  I can't imagine it is a problem with the packaging as a container because it should be consistent.

mbentley, 2024-05-31:
Well, strange thing just happened - I was able to reproduce the issue while I was testing running the app with OpenJDK 8.  I was curious if it was something related to the OpenJDK version but I've been able to reproduce it 3x in a row on fresh installs just doing a basic `docker run`.  I can take the image that I built (`mbentley/omada-controller:beta-5.14.20.9-openjdk8-amd64`), push it to Docker Hub, run it on another machine that is using the same OS, same kernel, same Docker version just fine though.

mbentley, 2024-05-31:
Posted to the forums: https://community.tp-link.com/en/business/forum/topic/670026

PappyChappy1, 2024-05-31:
Is there anyway to reinstall the image for 5.14.0.11? That one worked for me, so my backups are for that version...I can't downgrade due to the version #s.  And I can't upgrade to 5.14.20.9 because I'm having the startup issues.  I guess this is the cost of using beta software, but I really don't want to lose my backup configs!

mbentley, 2024-05-31:
I need to update the build process to also tag the beta images with version numbers besides just with the beta tag but for now, here are the last image digests that you can specify:

amd64:
`mbentley/omada-controller:beta-amd64@sha256:b2cbc1e119581c86f7a31c62f2b854e808b576e54f565d364bfa8439767691d7`

arm64:
`mbentley/omada-controller:beta-arm64@sha256:508258ef9a7616b94afce7aeec04bb66a28f7c201802f0ac43694adcd9753762`

armv7l:
`mbentley/omada-controller:beta-armv7l@sha256:83023cb66332231ae315d327b1426ba2078d6198b2ee2d4898d7cea7b789ffdb`

mbentley, 2024-05-31:
For beta-5.14.20.9 and further releases, there will be additional tags with the version number.  For example, https://hub.docker.com/r/mbentley/omada-controller/tags?page=&page_size=&ordering=&name=beta-5.14.20.9

PappyChappy1, 2024-06-01:
Thank you! This allowed me to reinstall 5.14.0.11, at least I'm in good shape until the next version gets fixed

eblieb, 2024-06-03:
Any update? Tried installing again and still got the same error

IAmKonni, 2024-06-03:
I retried with the latest beta and now the container starts without any issues ...

eblieb, 2024-06-03:
> I retried with the latest beta and now the container starts without any issues ...

What installer script did you use?

IAmKonni, 2024-06-03:
I just used this image with my older docker-compose.yml: `image: mbentley/omada-controller:beta-5.14.20.9`
Did not change anything else except 5.13 --> beta-5.14.20.9

eblieb, 2024-06-03:
Just tried docker-compose with the same image and getting the same error... strange how hit and miss this is.

eblieb, 2024-06-07:
Had anyone tried a native install vs the docker and seen if the beta has the same issues on a native Debian install? May test it later

mbentley, 2024-06-07:
The problem is that it seems pretty random.  There is a strong possibility that you may get the same odds of it working or not working if you install it directly on a host.  It would be interesting to see if it fails to work directly installed on the same hosts it fails on in a container though I suppose.

ruudkobes, 2024-06-07:
Thanks for your time and effort @mbentley.

I just tried beta-5.14.20.9 and it started without problem.
Not sure what that means, but I'll let it run for a while and see if it is stable for me.

eblieb, 2024-06-09:
I probably will build a VM to test it.

eblieb, 2024-06-11:
> The problem is that it seems pretty random. There is a strong possibility that you may get the same odds of it working or not working if you install it directly on a host. It would be interesting to [...]

Yup installed outside docker and runs fine.

chriexpe, 2024-06-26:
Had the 5.14.0.11 running just fine for a couple of months, decided to give a try on 5.14.20.19 and got that "Unsatisfied dependency" error too, but thanks to that image digest from mbentley I was able to go back and get it working.

J0hnMatrix, 2024-07-10:
Same issue here, not possible to start 5.14 latest release.
Had to downgrade by removing LAST_RAN_OMADA_VER.txt and specify 5.13 as tag.

Everything is back and waiting more investigation.

sant1ago, 2024-07-10:
Same issue here - I too had to roll back to 5.13 version

mbentley, 2024-07-10:
If everyone who is having this problem could also post on the [TP-Link Forums in this thread](https://community.tp-link.com/en/business/forum/topic/670026), that would be greatly appreciated.  The more noise this gets, the better.

mbentley, 2024-07-10:
I've also submitted a support request through their support site hoping that it gets more visibility.  Will update if I hear anything.

alphamike77, 2024-07-10:
> Same issue here, not possible to start 5.14 latest release. Had to downgrade by removing LAST_RAN_OMADA_VER.txt and specify 5.13 as tag.
> 
> Everything is back and waiting more investigation.

Trying to rollback now. Where is the LAST_RAN_OMADA_VER.txt located?

Thanks!

alphamike77, 2024-07-10:
Got it, thanks!

rmcfadzean, 2024-07-10:
I was previously on 15.3.22/`latest-chromium`/`fbc3d14b25453727c17be1adaa1bcf691dc0d152204e4498120102052d10f42a` and tried to upgrade to the current `latest-chromium` but encountered the same issue as above.
After changing the image to `beta-5.14.20.9-chromium` and removing the `LAST_RAN_OMADA_VER.txt`, it works without issue. I haven't tried `5.14.26.1` though.

hbsagen, 2024-07-10:
I had the same problem with 5.14 when I updated today. Had to remove the LAST_RAN_OMADA_VER.txt and run 5.13.

mschingel, 2024-07-10:
> I had the same problem with 5.14 when I updated today. Had to remove the LAST_RAN_OMADA_VER.txt and run 5.13.

How do I do that, I think im having this same issue.

mschingel, 2024-07-10:
Thanks, I figured it out too. I use Portainor so I stopped the container there.  Then "sudo su -" and "rm /opt/tplink/EAPController/data/LAST_RAN_OMADA_VER.txt" Then updated the tag in the stack to image: mbentley/omada-controller:5.13.30.8    I replaced the latest with 5.13.30.8

helio4k, 2024-07-11:
I can confirm as the rest of people above downgrade to 5.13 is necessary after updating 5.14.

evgaqa, 2024-07-11:
Same for me
To revert with portainer just recreate container with 5.13 and remove LAST_RAN_OMADA_VER.txt from /var/lib/docker/volumes/
Omada starts automatically

KingOperator, 2024-07-11:
I've the same issue after upgrading to 5.14

javito1081, 2024-07-11:
well my post [https://github.com/mbentley/docker-omada-controller/issues/442](url) has been closed cause aparently its the same as this one, has there been any development on fixing this issue?

CalvinSchwartz, 2024-07-11:
Using Watchtower it auto updated for me. Could you remove the 5.14 tag from docker hub, set it to testing or something equivalent as cleary 5.14 isn't ready to use?

javito1081, 2024-07-11:
> Using Watchtower it auto updated for me. Could you remove the 5.14 tag from docker hub, set it to testing or something equivalent as cleary 5.14 isn't ready to use?

well i have another instance on a diferent server on which i updated and is working without issues, but this one is not :-(

mbentley, 2024-07-11:
> well my post [https://github.com/mbentley/docker-omada-controller/issues/442](url) has been closed cause aparently its the same as this one, has there been any development on fixing this issue?

No, unfortunately TP-Link has not responded.  Assuming 5.14 has never successfully started on your system, you can clear the version check file (`LAST_RAN_OMADA_VER.txt` in the persistent data directory from `/opt/tplink/EAPController/data` inside the container) and roll back to 5.13.

mbentley, 2024-07-11:
> Using Watchtower it auto updated for me. Could you remove the 5.14 tag from docker hub, set it to testing or something equivalent as cleary 5.14 isn't ready to use?

Trying to provide TP-Link [as much info as possible on the forums](https://community.tp-link.com/en/business/forum/topic/670026?replyId=1383134) - I have an example where I have 6 systems, all unique, and 5 of the 6 work fine running the image but it's like it's always this one that doesn't work.  If I removed the image, or rolled back to 5.13 being `latest` it would be problematic for those who have successfully been able to run 5.14 as it would then break their controllers.

And honestly, this is a lesson in never running the `latest` tag.  I have seriously considered actually deleting it from the repo because blindly upgrading isn't a great idea.  I don't even run `latest` for my own personal controller 😆

javito1081, 2024-07-11:
> > well my post [https://github.com/mbentley/docker-omada-controller/issues/442](url) has been closed cause aparently its the same as this one, has there been any development on fixing this issue?
> 
> No, unfortunately TP-Link has not responded. Assuming 5.14 has never successfully started on your system, you can clear the version check file (`LAST_RAN_OMADA_VER.txt` in the persistent data directory from `/opt/tplink/EAPController/data` inside the container) and roll back to 5.13 by specifying 5.13 as the tag.
> 
> For example, if your container is named `omada-controller`, you can get the location of where the data is mounted, no matter if it is a bind mount or a volume:
> 
> ```
> docker inspect -f '{{ range .Mounts }}{{ if eq .Destination "/opt/tplink/EAPController/data" }}{{ .Source }}{{ end }}{{ end }}' omada-controller
> ```
> 
> Volume example:
> 
> ```shell
> $ docker inspect -f '{{ range .Mounts }}{{ if eq .Destination "/opt/tplink/EAPController/data" }}{{ .Source }}{{ end }}{{ end }}' omada-test
> /var/lib/docker/volumes/dbd53e89af74c1340784249f2961a6d549dfbe3af4e47b31d182b68d521132ff/_data
> 
> # combine that to show the directory contents
> $ sudo ls -l $(docker inspect -f '{{ range .Mounts }}{{ if eq .Destination "/opt/tplink/EAPController/data" }}{{ .Source }}{{ end }}{{ end }}' omada-test)
> [sudo] password for mbentley:
> total 84
> drwxr-xr-x 4  508  508 69632 Jul 11 16:06 db
> drwxr-xr-x 2  508  508  4096 Jul 11 15:43 html
> drwxr-xr-x 2  508  508  4096 Jul 11 15:43 keystore
> -rw-r--r-- 1 root root    10 Jul 11 15:43 LAST_RAN_OMADA_VER.txt
> drwxr-xr-x 2  508  508  4096 Jul 10 06:51 pdf
> ```
> 
> Bind mount example:
> 
> ```shell
> $ docker inspect -f '{{ range .Mounts }}{{ if eq .Destination "/opt/tplink/EAPController/data" }}{{ .Source }}{{ end }}{{ end }}' omada-controller
> /data/omada-controller/data
> 
> # combine that to show the directory contents
> $ sudo ls -l $(docker inspect -f '{{ range .Mounts }}{{ if eq .Destination "/opt/tplink/EAPController/data" }}{{ .Source }}{{ end }}{{ end }}' omada-controller)
> [sudo] password for mbentley:
> total 176
> drwxr-xr-x 2 omada omada   4096 Jul 11 01:15 autobackup/
> drwxr-xr-x 4 omada omada 151552 Jul 11 16:07 db/
> drwxr-xr-x 3 omada omada   4096 Jun 30  2023 device-firmware/
> drwxr-xr-x 2 omada omada   4096 Jul 11  2022 html/
> drwxr-xr-x 2 omada omada   4096 Jul 10 07:25 keystore/
> -rw-r--r-- 1 root  root      10 Jul 10 07:25 LAST_RAN_OMADA_VER.txt
> drwxr-xr-x 2 omada omada   4096 Jul 11  2022 pdf/
> ```

Ya i did that before posting, my controller is back to runing on 5.13, but i cant run it on 5.14, no mattar if its a fresh install or not :-(

surfek, 2024-07-12:
The same issue here... Had to revert to 5.13. Will post on TP-Link forum

lucianor, 2024-07-12:
@mbentley  did you try different versions of the base image or combinations of openjdk / mongo versions ? 
I know the latter can be challenging because it will need a DB upgrade, but I'm rebuilding this image myself and trying different combinations to see if I can make it work.. wondering if you are doing the same and if we can divide and conquer

mbentley, 2024-07-12:
> @mbentley did you try different versions of the base image or combinations of openjdk / mongo versions ? I know the latter can be challenging because it will need a DB upgrade, but I'm rebuilding [...]

No, my testing has been focused on running the exact same image on multiple machines.  In my case, it successfully runs on 5 of the 6 I tried it on.  I am not a Java/Spring Boot expert but it just seems to be a problem with how they define the dependencies and they've created a circular dependency but I have no idea why that would show up randomly for different users, especially considering I am testing this on multiple machines, a few that are clones of each other.  The one where it fails to run is a clone of another where I have just about the exact same packages installed and they're the same OS, kernel, etc.

One thing that would be simple for running different MongoDB versions would be to use an [externally hosted MongoDB](https://github.com/mbentley/docker-omada-controller/tree/master/external_mongodb).  You can just run the existing `5.14` image but just use two required env vars: of `MONGO_EXTERNAL` and `EAP_MONGOD_URI`.  I can test that on my one VM that is not working.

mbentley, 2024-07-12:
I just tested MongoDB 3, 4, and 7 on my VM where it always happens - same behavior every time.

antbarney, 2024-07-12:
Also running into this issue, as a Note I can successfully run "beta-5.14.20.9-chromium" but any of the other 5.14 images fail to start for me. I have reverted to 5.13 and seem to be back up tho.

mbentley, 2024-07-12:
> Also running into this issue, as a Note I can successfully run "beta-5.14.20.9-chromium" but any of the other 5.14 images fail to start for me. I have reverted to 5.13 and seem to be back up tho.

Interesting data point - `beta-5.14.20.9-chromium` runs fine on my VM that has issues.  The `beta-5.14.20.9` tag also runs fine although that makes sense seeing as `beta-5.14.20.9-chromium` just extends the `beta-5.14.20.9` image by adding the chromium package.

lucianor, 2024-07-12:
thanks for confirming. My tests have confirmed the same - OpenJDK and MongoDB does not make a difference  - under an arm64 (Orange Pi 5).

dougle03, 2024-07-12:
What's the solution to get the controller back up?
FYI Running in Docker on a Pi4. Thanks

surfek, 2024-07-12:
> What's the solution to get the controller back up? FYI Running in Docker on a Pi4. Thanks

get back to 5.13 version and remove LAST_RAN_OMADA_VER.txt file. After that it shall start with 5.13....

[96 more replies]

closed 2024-10-15 by mbentley