mbentley/docker-omada-controller #426: [Feature]: MongoDB repair mode

SBCV-apegram, 2024-05-15:
### What problem are you looking to solve?

I recently had a problem with a MongoDB file becoming corrupted. I was able to repair the damage, but I had to create another Docker image similar to yours that had the same version of MongoDB but didn't try to start the controller software.

### Describe the solution that you have in mind

It would be nice to have an environment variable that would start the container without running the controller software. That way, the database could be repaired in the exact same environment that the controller runs in.

For those interested, my problem was that one of the collection files wasn't recognized as a WiredTiger file. I was able to copy the file from a new instance of Omada Controller, run `mongod --repair` on the database, then `chown` the files back to the correct user.

### Additional Context

I know it's a niche case, but it would be nice to have in case this happens again or to someone else.

mbentley, 2024-05-15:
Any reason why overriding the entrypoint and CMD of the standard container image accomplish this?

SBCV-apegram, 2024-05-15:
You know, in all of my Googling I did not find this solution? 🤦‍♂️🤦‍♂️ I was searching for ways to stop the main process in the container, and no one else on the internet suggested overriding the entrypoint or command. Thanks!

mbentley, 2024-05-15:
When trying to [solve another problem last year](https://github.com/mbentley/docker-omada-controller/discussions/344#discussioncomment-7254461), I didn't think of doing just that when I wanted to run MongoDB by itself and I was suggesting that people use the `mongo` image from Docker Hub 😬

mbentley, 2024-05-15:
It still might be interesting to see if I could add an env var to trigger running a repair though.

SBCV-apegram, 2024-05-15:
I feel better about my google-fu fail now. I tried using the `mongo` image, but it complained that the database version was too old. Ubuntu really needs to catch up the version in their repos... Or does Omada controller require the ancient version?

An auto repair mode would be cool, though.

mbentley, 2024-05-15:
Yeah, the Omada Controller requires a pretty old version of MongoDB at least according to their documented requirements and MongoDB upgrades are quite a pain as well.  You technically can use the `mongo` image but you have to make sure that you match the version that's in the container.

I'm no MongoDB expert so I basically just take what documentation & "experts" say online but one interesting thing is that [this doc](https://www.mongodb.com/docs/manual/tutorial/recover-data-following-unexpected-shutdown/) mentions that when you're running with journaling, repairs shouldn't be needed in most cases unless there is actual underlying corruption but I am also not certain if something like the container being killed could cause such data corruption.

This one time command should be able to run a repair (only to be run when the controller isn't running):

```bash
docker run -it --rm \
  -u 508:508 \
  -v omada-data:/opt/tplink/EAPController/data \
  --entrypoint /opt/tplink/EAPController/bin/mongod \
  mbentley/omada-controller:5.13 \
    --repair \
    --dbpath /opt/tplink/EAPController/data/db \
    --bind_ip 127.0.0.1 \
    --port 27217
```

I could pull that command out and include it in the entrypoint to run when a user has provided a specific env var but at the same time, I would probably be just as easy to document this command in the README.

mbentley, 2024-07-17:
After further consideration, I am going to go ahead and close this as I am not going to implement a flag to run a repair due to the fact that journaling is used so needing to do a repair should be a very rare thing ([per the MongoDB docs](https://www.mongodb.com/docs/v5.0/tutorial/recover-data-following-unexpected-shutdown/)) and there is the above documented command to run a repair.  Here's the excerpt from the docs about journaling:

> If you are running with [journaling](https://www.mongodb.com/docs/v5.0/reference/glossary/#std-term-journal) enabled, there is almost never any need to run repair since the server can use the journal files to restore the data files to a clean state automatically. However, you may need to run repair in cases where you need to recover from a disk-level data corruption.

closed 2024-07-17 by mbentley