mbentley/docker-omada-controller #434: [Bug]: Multiple images with the same tag confuse k8s multi-node cluster

jackwellsxyz, 2024-07-02:
### Controller Version

v5.13

### Describe the Bug

When I restart the `omada-controller` pod and it moves to another node that previously had `omada-controller` running, k8s will attempt to run the image, even if it's an older image with the 5.13 tag. This causes the pod to continually restart because the `LAST_RAN_OMADA_VER.txt` specifies a version that is newer (as run on the previous node).

### Expected Behavior

Fix is pretty simple: Can you please add a patch version or date or some signifier to differentiate versions of `omada-controller` within a minor release that are incompatible? For example, if upgrading from one version of 5.13 to a newer one causes it to not work with the older one, then they should have different tags.

As a workaround, I am specifying the omada controller via its sha256, for example:
`image: mbentley/omada-controller@sha256:7681b0cd83c9f3bff6fff82015e3a69af32aa06635b4a88f98c8205398a5660a`. If you plan to keep the same tag versioning scheme for `omada-controller`, I'd suggest adding the sha256 approach to the docs.

### Steps to Reproduce

1. Run omada-controller pod (specified as e.g. `mbentley/omada-controller:5.13` on node A.
2. Wait for a new version of the 5.13 image to be released.
3. Kill the pod and wait for it to run on node B.
4. Kill the pod and wait for it to run again on node A and note that it will continually restart as the `LAST_RAN_OMADA_VER.txt` specifies a newer version than what is being run.

### How You're Launching the Container

```plain
Via deployment.
```

### Container Logs

```plain
.
```

mbentley, 2024-07-03:
I regularly patch the images to deal with security vulnerabilities so I would actually suggest using the major.minor tag and use `imagePullPolicy: always` for your deployment.

Additionally on my end there is https://github.com/mbentley/docker-omada-controller/issues/423 which calls out the lack of having full version tags.  I could implement that but it will just take time to do the manual work to update the full version strings in all 26 of the image build commands and add the additional manifest commands to write the manifests.  The only time when you should have hit the major.minor version issue would have been in the 5.9 and 4.9 versions as those were the only two there the major.minor repeated as far as I can tell from the published versions - I could be wrong about that though.

mbentley, 2024-07-17:
I am going to close this as there are a few ways to properly address this between the `imagePullPolicy` (what I would suggest), using [an explicit tagged version](https://github.com/mbentley/docker-omada-controller?tab=readme-ov-file#explicit-version-tags) (which was implemented from [this request](https://github.com/mbentley/docker-omada-controller/issues/423)) which will ensure you have the exact version, such as `5.14.26.1`, but will not guarantee that you're running a patched container image which is why this is not the suggested approach.  Let me know if there is anything else I can do to help.

closed 2024-07-17 by mbentley