mbentley/docker-omada-controller #443: [Feature]: Update Ubuntu base image and MongoDB version

lucianor, 2024-07-12:
### What problem are you looking to solve?

[TP-Link FAQ](https://www.tp-link.com/br/support/faq/3272/) for Omanda SDN Controller install on Linux suggests Ubuntu 22.04 as the base system, with MongoDB 7 as suggested for the database.

The image is using Ubuntu 20.04 with the built-in [MongoDB version 3.68](https://packages.ubuntu.com/focal-updates/database/mongodb).  I found this while investigating a fix for 

It would be interesting to provide a more current tech stack with either jammy or noble as the default, and with a more modern and currently supported MongoDB version. Reasons for this change:

1) focal supports [ends in less than a year ](https://ubuntu.com/about/release-cycle)- Apr 2025 and is no longer maintained
2) jammy gets access to newly updated OpenJDK tech stack and also updates to any potential security fixes in Java
3) MongoDB 3.68 is entirely deprecated
4) TP-Link may at any time upgrade the requirements for running Omada SDN Controller, increasing the risk of a making a bigger jump in technology, therefore making it hard to update the controller.
5) Having the latest thing

### Describe the solution that you have in mind

1) Use jammy Ubuntu base image from Canonical (ARG BASE=ubuntu:22.04)
2) Use MongoDB repo for installation - Add instructions from [here](https://www.mongodb.com/pt-br/docs/manual/tutorial/install-mongodb-on-ubuntu/) to install.sh
3) Write a script to migrate MongoDB 3.68 to MongoDB 7 - this is mentioned on #356  and part of https://github.com/mbentley/docker-omada-controller/commit/a64518cb137f5f5b131b01418d93e043b37c4dc0

mbentley, 2024-07-12:
So there are certainly a number of reasons why I have held off upgrading to at least 22.04 - the first and the biggest elephant in the room being that MongoDB 3.x is no longer supported by MongoDB.  There are at least still some level of security updates from Canonical with it being packaged in the official repos for 20.04 (focal) so it's basically at a dead end support wise.

As for making the jump to noble/24.04 - that's not supported according to the FAQ so 22.04 would be where the base OS upgrade would need to end up.

The good news about JDK is that [focal](https://packages.ubuntu.com/search?suite=focal&section=all&arch=any&keywords=-jre-headless&searchon=names), [jammy](https://packages.ubuntu.com/search?suite=jammy&section=all&arch=any&keywords=-jre-headless&searchon=names), and [noble](https://packages.ubuntu.com/search?suite=noble&section=all&arch=any&keywords=-jre-headless&searchon=names) all have OpenJDK 8, 11, 17, and 21 available.  Comparing the versions from focal to noble, they're both running `17.0.11+9-1` so they're the same.  I'm currently already using OpenJDK 17 for amd64 and arm64 which TP-Link said will work fine for the controller - I have no yet seen confirmation about OpenJDK 21 though.  I am guessing one of the main reason they have yet to move to Spring Boot 3.x is that they would probably need to move to OpenJDK 21 which they may have issues with (not sure, haven't tested that) but it certainly eliminates older operating systems that don't provide a version of OpenJDK 21 so that could be why.

As for MongoDB 7, I don't see any reference to MongoDB 7 being the suggested version, at least not on the US version of the FAQ (https://www.tp-link.com/us/support/faq/3272/).  It mentions: "Omada SDN Controller supports MongoDB v3 and v4. Here we will show how to install v4.4."  The bad news is that even from MongoDB directly, MongoDB 4.4, which is the oldest version that TP-Link supports, [went EOL on February 2024.](https://www.mongodb.com/legal/support-policy/lifecycles).  I will have to see if I can get someone on the TP-Link forums to provide some guidance around future MongoDB versions and see if they have any info on future support of newer versions.

lucianor, 2024-07-12:
100% with you on all the above. jammy looks like the go-to base image for now.. Similarly, OpenJDK 17 seems fine as a choice, and the upgrade gap is not big...
But MongoDB is definitely the elephant in the room. I'm fine tuning / re-using your code from here https://github.com/mbentley/docker-omada-controller/tree/mongodb-migrate-to-4.4 to make it all the way to 7.0, but putting the migrate script on top of the current stack, with the base image upgrade and a check for the mongoDB version as a stretch goal.
Why 7.0? It looks like the FAQ for my region and Canada are different than the US and Global version - https://www.tp-link.com/ca/support/faq/3272/ and says that is supported. If we are making users run an upgrade script, it may be wise to use the latest version supported to reduce the likelihood of doing it again in the future.

mbentley, 2024-07-12:
Ah I see - so it's starting with `5.14.20` and above that they mention that MongoDB 7 is supported.  Must just not have made their way through the other FAQs. 100% with you there on only wanting to do one migration of the database if I can help it.

The thing that really sucks is that I do not want to continually ship all of the MongoDB binaries forever until people upgrade so I would be leaning toward doing a specific migration image that the user would have to run manually.  The MongoDB binaries and dependencies are about 1 GB which is just a deal breaker for me.

I am working on updating the `migrate.sh` script to add the additional migrations.  It doesn't look too bad at face value so we'll see.

mbentley, 2024-07-12:
OK, so `arm64` was a pain in the ass because they didn't have ideal versions but this seems to work:

https://github.com/mbentley/docker-omada-controller/tree/mongodb-migrate-to-7.0/mongodb_migrate

At least the migrations run for both amd64 and arm64.  I haven't yet tested to make sure that I can install MongoDB 7.x and have the controller start up but this is all I can probably work on for today.

mbentley, 2024-07-12:
Another thing would be that moving to a newer version of MongoDB would completely drop support for armv7l (which I would NOT be mad about at all).

mbentley, 2024-07-13:
Here is a WIP branch: https://github.com/mbentley/docker-omada-controller/tree/update-base-and-mongo

`amd64` and `arm64` build commands:

```bash
docker build \
  --pull \
  --build-arg INSTALL_VER="5.14" \
  --build-arg ARCH="amd64" \
  --build-arg BASE=mbentley/ubuntu:22.04 \
  --progress plain \
  -f Dockerfile.v5.x \
  -t mbentley/omada-controller:5.14-22.04-mongodb7-amd64 \
  .

docker build \
  --pull \
  --build-arg INSTALL_VER="5.14" \
  --build-arg ARCH="arm64" \
  --build-arg BASE=mbentley/ubuntu:22.04 \
  --progress plain \
  -f Dockerfile.v5.x \
  -t mbentley/omada-controller:5.14-22.04-mongodb7-arm64 \
  .
```

And two images published to Docker Hub:
- mbentley/omada-controller:5.14-22.04-mongodb7-amd64
- mbentley/omada-controller:5.14-22.04-mongodb7-arm64

At least I will be able to use these to test if the migration scripts worked.

lucianor, 2024-07-13:
https://github.com/mbentley/docker-omada-controller/tree/mongodb-migrate-to-7.0/mongodb_migrate

Tested this with my current Omada 5.13 / MongoDB 3.68 on both amd64 and arm64. Both upgrades went fine:

> INFO: Migration to 7.0.12 complete!

I did try `mbentley/omada-controller:5.14-22.04-mongodb7-arm64` but got the dreaded #418 bug.. I tried 5.14.26.1 and beta-5.14.20.9.

I tried rebuilding the image with 5.13.30.8 but that does not work with MongoDB 7.0

mbentley, 2024-07-14:
🤦 bummer.  I will see if I can test it myself soon.

mbentley, 2024-07-15:
OK, I just made a pretty big update to have cleaner output - all of the MongoDB logs are being put to a file.  Repair should only be needed if the db was shut down uncleanly.  Maybe that should be a first step before doing the first upgrade, just to make sure but this output is much more easy to understand if the upgrade was successful:

```
INFO: executing upgrade process from MongoDB 3.6 to 7.0...

INFO: starting upgrade to 4.0...
INFO: starting mongod 4.0.28....done
INFO: setting feature compatibility version to 4.0...done
INFO: verifying feature compatibility version has been updated to 4.0...done
INFO: stopping mongod...done
INFO: upgrade to 4.0 complete!

INFO: upgrading from libcurl3 to libcurl4
Selecting previously unselected package libcurl4:arm64.
dpkg: considering removing libcurl3:arm64 in favour of libcurl4:arm64 ...
dpkg: yes, will remove libcurl3:arm64 in favour of libcurl4:arm64
(Reading database ... 4570 files and directories currently installed.)
Preparing to unpack .../libcurl4_7.58.0-2ubuntu3.24_arm64.deb ...
Unpacking libcurl4:arm64 (7.58.0-2ubuntu3.24) ...
Setting up libcurl4:arm64 (7.58.0-2ubuntu3.24) ...
Processing triggers for libc-bin (2.27-3ubuntu1.6) ...

INFO: starting upgrade to 4.2...
INFO: starting mongod 4.2.23....done
INFO: setting feature compatibility version to 4.2...done
INFO: verifying feature compatibility version has been updated to 4.2...done
INFO: stopping mongod...done
INFO: upgrade to 4.2 complete!

INFO: starting upgrade to 4.4...
INFO: starting mongod 4.4.18.....done
INFO: setting feature compatibility version to 4.4...done
INFO: verifying feature compatibility version has been updated to 4.4...done
INFO: stopping mongod...done
INFO: upgrade to 4.4 complete!

INFO: starting upgrade to 5.0...
INFO: starting mongod 5.0.27.....done
INFO: setting feature compatibility version to 5.0...done
INFO: verifying feature compatibility version has been updated to 5.0...done
INFO: stopping mongod...done
INFO: upgrade to 5.0 complete!

INFO: starting upgrade to 6.0...
INFO: starting mongod 6.0.16.....done
INFO: setting feature compatibility version to 6.0...done
INFO: verifying feature compatibility version has been updated to 6.0...done
INFO: stopping mongod...done
INFO: upgrade to 6.0 complete!

INFO: starting upgrade to 7.0...
INFO: starting mongod 7.0.12...{"t":{"$date":"2024-07-15T12:26:28.129Z"},"s":"I",  "c":"NETWORK",  "id":4915701, "ctx":"main","msg":"Initialized wire specification","attr":{"spec":{"incomingExternalClient":{"minWireVersion":0,"maxWireVersion":21},"incomingInternalClient":{"minWireVersion":0,"maxWireVersion":21},"outgoing":{"minWireVersion":6,"maxWireVersion":21},"isInternalClient":true}}}
..done
INFO: setting feature compatibility version to 7.0...done
INFO: verifying feature compatibility version has been updated to 7.0...done
INFO: stopping mongod...done
INFO: upgrade to 7.0 complete!

INFO: Fixing ownership of database files...done

INFO: upgrade process from MongoDB 3.6 to 7.0 was successful!
```

mbentley, 2024-07-15:
So just thinking about the migration/upgrade script further, there are some additional things I would like to do:

- Create a tarball of the database before starting the upgrade
- Add in error catching to revert to the database backup tarball if any upgrade step fails
- Check to make sure that another process isn't already using the MongoDB (in case someone forgets to stop the controller container)

And then in the actual controller image once it's updated:

- Entrypoint check to see if the MongoDB has been migrated to prevent trying to start when it can't
- Think about how I can provide some forced transition for users
  - I was thinking about having the entrypoint pull down a tarball of the required binaries for the upgrade but then I remembered that each Docker image has some unique packages needed and specific versions of the OS & if anyone is using a healthcheck, it could kill the migration in the middle.
  - I was also thinking that for a period time, I should publish two images - one with 20.04, the other with 22.04 for the same version and for the one with 20.04, I would have the entrypoint use a simple env var check that makes the user acknowledge that that they need to upgrade and until they add the env var set to `true` or something like that, the container wouldn't actually start.  It would output a message that links to a README on adding the env var as well as running the upgrade container.  It's heavy handed but could work.

I am sure I am forgetting about all sorts of scenarios that need to be addressed but it is good to think through earlier than later.

mikepell007, 2024-07-16:
Just as an FYI in case some one else runs into this.  Trying out the new image with MongoDB7 with no luck.  mongod process would crash.  Turns out my CPU does not support the avx instruction set.  This seems to be a requirement for mongodb 7.

https://www.mongodb.com/community/forums/t/fresh-install-result-in-crash-at-startup/264498

mbentley, 2024-07-16:
Oh yikes.  Good catch on that.  I will have to see if I can add a check pre-upgrade to make sure that AVX is present although I imagine that the rollback process I'm working on should work if it fails to start.

lucianor, 2024-07-16:
Should the AVX requirement be treated as a requirement for running the controller? As described in the README vs scripted requirement? 
Not sure if docker containers has access to /proc so you could grep -o 'avx[^ ]*' /proc/cpuinfo to tell if AVX is available... 
Or stick with MongoDB 6...

mbentley, 2024-07-16:
According to https://www.mongodb.com/docs/manual/administration/production-notes/#x86_64, the AVX requirement was added in MongoDB 5.x so anyone who can't run AVX basically can't run a supported MongoDB.  Regardless, MongoDB 6.x goes EOL July 2025 so that's not really a long runway.

mikepell007, 2024-07-16:
> According to https://www.mongodb.com/docs/manual/administration/production-notes/#x86_64, the AVX requirement was added in MongoDB 5.x so anyone who can't run AVX basically can't run a supported MongoDB. AVX became available in 2011 and I am not sure how long it really took before it really became broadly adopted but that's almost 13 years ago at this point where both Intel and AMD had it first available in a released CPU. Regardless, MongoDB 6.x goes EOL July 2025 so that's not really a long runway.

In my case, it happens to be an Atom C3758 (Denverton family) so not too old.  I'm guessing for the Atom format they removed some lesser used features to make it smaller/cooler/more power efficient/etc.

lucianor, 2024-07-16:
> OK, I just made a pretty big update to the migration script [in the branch](https://github.com/mbentley/docker-omada-controller/tree/mongodb-migrate-to-7.0/mongodb_migrate) to have cleaner output - all of the MongoDB logs are being put to a file. Repair should only be needed if the db was shut down uncleanly. Maybe that should be a first step before doing the first upgrade, just to make sure but this output is much more easy to understand if the upgrade was successful:

I tried building a Docker image locally from this folder and running the migration script. It didn't work, output below. I was unable to locate the logs as you mentioned. The previous version of the script worked.

```
INFO: running pre-flight checks on MongoDB...done
INFO: creating a backup (mongodb-preupgrade.tar) of MongoDB pre-upgrade...done

INFO: executing upgrade process from MongoDB 3.6 to 7.0...

INFO: starting upgrade to 4.0...
INFO: starting mongod 4.0.28...about to fork child process, waiting until server is ready for connections.
forked process: 14
ERROR: child process failed, exited with error number 14
To see additional information in this output, start without the "--fork" option.

ERROR: unexpected failure; aborting MongoDB upgrade and rolling back!
INFO: cleaning up mongo.pid...done
INFO: rolling back to the backup of MongoDB prior to the upgrade...done
INFO: the MongoDB backup file (mongodb-preupgrade.tar) is still in your persistent data directory in case you need it
INFO: successfully rolled back MongoDB using the pre-backup archive

```

mbentley, 2024-07-16:
Ah so this is part of the tricky part at the moment.  The MongoDB logs are now being sent to `/tmp/upgrade_log.txt` inside the container.  I should figure out how to handle those.  Easiest might be to actually change the location to somewhere under the persistent data directory just so that the migration logs are available because most people shouldn't be running an interactive shell to do the migration.

I will update it to go to `/opt/tplink/EAPController/data/mongodb_upgrade.log`.  Curious if I should output like the last 20 lines of the logs or something when an error occurs like this.  Thoughts?

mbentley, 2024-07-16:
> In my case, it happens to be an Atom C3758 (Denverton family) so not too old. I'm guessing for the Atom format they removed some lesser used features to make it smaller/cooler/more power [...]

Ah yeah, that's unfortunate.  I have an Intel Atom C3558 which is my pfsense firewall and it also does not have AVX support but obviously it being my pfsense, I don't use it to run an Omada Controller.

lucianor, 2024-07-16:
> I will update it to go to `/opt/tplink/EAPController/data/mongodb_upgrade.log`. Curious if I should output like the last 20 lines of the logs or something when an error occurs like this. Thoughts?

If this is a container with the sole purpose of upgrading the database, I would bring the your treated output to the docker logs, and have the full logs into the file. Also, I figured the mapping of the data folder by looking into the script - I assume you are using the same data bindings as if it was a controller container.

mbentley, 2024-07-16:
Sorry, I will add a README with better, consistent instructions as yes, the data is mapped the same as in the controller container.

One thing I could do to remove some of the noise from the logs upon error is to have an MongoDB log for each version and just dump the entirety of the logs.  A lot of the logs were previously spam from the repair when I had that enabled so not doing a repair helps with that.

I also just realized that another thing I didn't ever do was verify that there was actually data in the `./data/db` directory before starting the upgrade.

mbentley, 2024-07-16:
On another note, I was looking at [the requirements for `arm64`](https://www.mongodb.com/docs/manual/administration/production-notes/#arm64) as well.  It says "MongoDB on arm64 requires the ARMv8.2-A or later microarchitecture."  The [Cortex-A72](https://developer.arm.com/Processors/Cortex-A72) is what is used in the Raspberry Pi 4 so even if it is running `arm64`, it might not work.  Also see this [rpi forum thread](https://forums.raspberrypi.com/viewtopic.php?t=348113) about the topic.

mbentley, 2024-07-17:
There is now a [README](https://github.com/mbentley/docker-omada-controller/blob/mongodb-upgrade-to-7.0/mongodb_upgrade/README.md) with instructions.  The TODO list is at the top of the `upgrade.sh` script.

mbentley, 2024-07-17:
> Awesome, will give it another try on my Orange Pi 5 arm64 and on Intel Pentium Gold G7400 amd64 architecture. On ARM requirements, I Orange Pi 5 (RK3588s) is running MongoDB 7.0 with no issues on Ubuntu noble. I can test your script into a spare Raspberry Pi 4 2gb I have laying around - just let me know.

Ah yeah, it looks like the RK3588S is a [Cortex-A76](https://developer.arm.com/Processors/Cortex-A76) which has arm 8.1 & 8.2 extensions so that could explain why it works fine.  I also have a few Raspberry Pi 3 B+ boards that I can test on as well since the Cortex A53 (rpi 3b+) and Cortex A72 (rpi 4) have the same instruction set (`ARMv8-A`)  The [wikipedia spec table](https://en.wikipedia.org/wiki/Raspberry_Pi#Specifications) for the Raspberry Pi devices has some good details there.

lucianor, 2024-07-29:
Followed [README](https://github.com/mbentley/docker-omada-controller/blob/mongodb-upgrade-to-7.0/mongodb_upgrade/README.md). I got the repair error, may be because that my omada-controller was not gracefully shutdown:

2024-07-29T13:51:51.891+0000 F STORAGE  [initandlisten] An incomplete repair has been detected! This is likely because a repair operation unexpectedly failed before completing. MongoDB will not start up again without --repair.

Maybe you could add back the repair to prevent this issue, at least from the first step of migration? This would prevent this from plain out stopping.

mbentley, 2024-07-29:
I just [pushed a change](https://github.com/mbentley/docker-omada-controller/commit/b8f089a8abc6e34377a1de888d7d88f8761064ee) to the branch so that it will perform a repair before the first upgrade to make sure it won't run into this issue.

mbentley, 2025-02-10:
I see in the [latest beta community forum thread](https://community.tp-link.com/en/business/forum/topic/734414) that there is a note about them adding support for MongoDB 8.0.  I'm going to have to watch out in the change logs to make sure that they don't deprecate the current version of MongoDB that's used in the image since that would force a lot of people to perform a migration.

mbentley, 2025-02-12:
I've created a new branch to go to 8.0: https://github.com/mbentley/docker-omada-controller/tree/mongodb-upgrade-to-8.0 and done some basic testing:

```bash
# build migration image
docker build \
  --pull \
  --progress plain \
  -f Dockerfile.upgrade-amd64 \
  -t mbentley/omada-controller:mongodb-upgrade-amd64 \
  .

# run controller
docker run -d \
  --name omada-controller \
  --stop-timeout 60 \
  --restart unless-stopped \
  --ulimit nofile=4096:8192 \
  --net=host \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:5.15

# configure controller in the interface and then stop
docker stop omada-controller

# create my own backup to be able to roll back to test multiple times
docker run -it --rm -v omada-data:/data -w /data alpine \
  sh -c 'tar cvf db.tar db && ls -l /data/db.tar'

# run the upgrade
docker run -it --rm \
  -v omada-data:/opt/tplink/EAPController/data \
  mbentley/omada-controller:mongodb-upgrade-amd64

# roll back upgrade to test multiple times
docker run -it --rm -v omada-data:/data -w /data alpine \
  sh -c 'rm -rfv /data/db/ && tar xvf db.tar && ls -l /data/db/'
```

sofiaurora270, 2025-04-04:
Would be willing to test this thing, since we might need to at some point, due to cluster mode and all. So i think it would be nice to get this thing done to the mainline containers. Who knows, maybe we get some speed improvements also. :)

mbentley, 2025-04-04:
Putting so thoughts down...with May 2025 being the end of standard support for 20.04/focal, this is something that I probably realistically need to finalize the plan of action on this month.  There are a few tricky bits:

- This will require running the migration as a standalone container
- Many people auto-upgrade the controller when new versions roll out using tools like watchtower - this means that if I roll out an update and people are forced to run the migration, I will essentially have to purposefully stop their controller from coming up - I'll have to output a message with a link to instructions and fail to start
- Those instructions will be to run the migration image and then to start the new image with the latest OS and MongoDB again.

I was really hoping that TP-Link would have a new major version release of 6.x at some point before the EOS for 20.04/focal which would be the trigger for the breaking change seeing as this migration needs a separate image due to the OS dependencies to run so many versions of MongoDB.  That means that I will most likely have to push this change somewhat randomly.

To reduce the support burden and user pain, I can probably just create a set of image tags for the last version of 20.04, like with `-legacy` or something appended as I have to anticipate that some MongoDB migrations will not happen smoothly and instead of have people in what feels like a broken state, they can change the image tag they're using to get the old image and then they can deal with the migration later (or they can hopefully take a backup and just do a full restore with a clean install of the controller, completely bypassing the migration).

I just updated the [update-base-and-mongo](https://github.com/mbentley/docker-omada-controller/tree/update-base-and-mongo) branch to have MongoDB 8 instead of 7 and rebased [mongodb-upgrade-to-8.0](https://github.com/mbentley/docker-omada-controller/tree/mongodb-upgrade-to-8.0).

There are a few more things left in the [To Do list](https://github.com/mbentley/docker-omada-controller/blob/mongodb-upgrade-to-8.0/mongodb_upgrade/upgrade.sh#L3-L11) that would be good to have as there will likely be failures on arm64 systems that don't have the right chip level support

I'll also need to do the logic in the entrypoint to block trying to start for the new OS versions but that should be pretty easy as I'll just be able to check the OS version [like I am doing here](https://github.com/mbentley/docker-omada-controller/blob/update-base-and-mongo/install.sh#L82) but checking for not a match and know from that.

mbentley, 2025-04-04:
OK, sorry if anyone has been trying to do anything with any of the branches testing wise or whatever but I merged mongodb-upgrade-to-8.0 into [update-base-and-mongo](https://github.com/mbentley/docker-omada-controller/tree/update-base-and-mongo) just to simplify any ongoing work.

fpoussin, 2025-04-04:
Hi

If you'd like to test on a Pi 4, this might help: 
https://github.com/themattman/mongodb-raspberrypi-binaries
https://github.com/themattman/mongodb-raspberrypi-docker

I'm using it on another device running a Cortex-A72 without any issues.

mbentley, 2025-04-04:
Unfortunately those would add a good bit of technical debt, introducing significant inconsistency between amd64 and arm64  as well as having amd64 special builds for just raspberry pi devices which isn't something I would like to do.  I did that for 32 bit armv7l for a while and it was less than ideal. - plus as far as I am aware, there is no way to detect naturally through Docker a difference between `armv8.2-a` and `armv8-a` as there is just `armv8`.  I also see that the last updated binaries were for 7.x which means that I would probably need to take up building them myself or only have MongoDB running on 7.x.

At this point, I would rather just continue to build images that are `-legacy` tagged for a while longer even though the Ubuntu 20.04 will go out of support, similar to what I ended up doing for armv7l, while encouraging people to migrate to newer hardware.

As for detecting `armv8.2-a`, I can just use a simple example app in C compiled with the `armv8.2-a` arch flag:

```bash
gcc -g atomic.c -o t4g -march=armv8.2-a -lpthread -static
```

```c
#include <stdio.h>
#include <threads.h>
#include <stdatomic.h>

atomic_int acnt;
int cnt;

int f(void* thr_data)
{
    for(int n = 0; n < 1000; ++n) {
        ++cnt;
        ++acnt;
    }
    return 0;
}

int main(void)
{
    thrd_t thr[10];
    for(int n = 0; n < 10; ++n)
        thrd_create(&thr[n], f, NULL);
    for(int n = 0; n < 10; ++n)
        thrd_join(thr[n], NULL);

    printf("The atomic counter is %u\n", acnt);
    printf("The non-atomic counter is %u\n", cnt);
}
```

It will error out on systems that don't have support:

```
$ ./atomic
Illegal instruction (core dumped)
```

And execute properly on those that do:

```
$ ./atomic
The atomic counter is 10000
The non-atomic counter is 10000
```

It's only 708k statically compiled so it's not a huge burden.  It would also be helpful for the upgraded images to do a pre-flight check before trying to start the controller.

mbentley, 2025-04-05:
I do suppose that if someone wants to still use their raspberry pi and use their own MongoDB using the external MongoDB capability, that would work since they could run a version 4.4 that would work on their Raspberry Pi.

That also means I need to think about more complex checks for blocking starting because I hadn’t taken external MongoDB support into account.

lucianor, 2025-04-25:
I've been using a version of your image but setting manually a 22.04 as base image and MongoDB 7.0, on ARM64 (OrangePi 5).  Because of that I had to comment out the version_step_upgrade from older versions, just keeping 7.0 to 8.0.

However, I'm getting this error:

```
INFO: starting upgrade to 8.0...
INFO: starting mongod 8.0.4.../tmp/mongod-8.0.4: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/mongod-8.0.4)
/tmp/mongod-8.0.4: /lib/aarch64-linux-gnu/libpthread.so.0: version `GLIBC_2.30' not found (required by /tmp/mongod-8.0.4)
```

mbentley, 2025-08-06:
> I've been using a version of your image but setting manually a 22.04 as base image and MongoDB 7.0, on ARM64 (OrangePi 5).
> 
> I'm testing your upgrade script, but because of my setup I had to comment out the version_step_upgrade from older versions, just keeping 7.0 to 8.0. However, I'm getting this error:
> 
> ```
> INFO: starting upgrade to 8.0...
> INFO: starting mongod 8.0.4.../tmp/mongod-8.0.4: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/mongod-8.0.4)
> /tmp/mongod-8.0.4: /lib/aarch64-linux-gnu/libpthread.so.0: version `GLIBC_2.30' not found (required by /tmp/mongod-8.0.4)
> ```

So I realize you posted this forever ago but the reason why you're seeing this is because of the dependencies for MongoDB 8.x.  Each version of the MongoDB package is OS version dependent.  You would need to modify the version of the tarball for MongoDB from their [MongoDB download center](https://www.mongodb.com/try/download/community).  For example, the latest tarball for Ubuntu 22.04 arm64 is https://fastdl.mongodb.org/linux/mongodb-linux-aarch64-ubuntu2204-8.0.12.tgz where as the Dockerfile uses a base of 18.04 due to it being the only one that is compatible with the binaries all the way from 4.x to 8.x so the tarballs I need are from what they list as the 20.04 version.

mbentley, 2025-09-06:
`update-base-and-mongo` has been merged into `master` which sets the groundwork to get this going for the v6 beta.

mbentley, 2025-09-06:
Beta images of v6 are now available along with the DB migration image.

mbentley, 2025-09-07:
I added another task to change how I am detecting support for AVX and ARMv8.2-A as I was poking around the MongoDB 8.0 Docker image and found this in their entrypoint, which is a lot easier than using the `atomic` program I found.

```
# avx (5.0+)
# from https://github.com/docker-library/mongo/blob/6f2dda34ccd9d11a1269f3307b02d232f09a3d3d/8.0/docker-entrypoint.sh#L28
if ! grep -qE '^flags.* avx( .*|$)' /proc/cpuinfo; then
...

# ARMv8.2-A (all versions ≥5.0, any of 4.4 ≥4.4.19)
# from https://github.com/docker-library/mongo/blob/6f2dda34ccd9d11a1269f3307b02d232f09a3d3d/8.0/docker-entrypoint.sh#L43
if ! grep -qE '^Features.* (fphp|dcpop|sha3|sm3|sm4|asimddp|sha512|sve)( .*|$)' /proc/cpuinfo; then
...
```

mbentley, 2025-09-30:
I added compose examples for external MongoDB in https://github.com/mbentley/docker-omada-controller/pull/633

sofiaurora270, 2025-11-06:
I tested the upgrade with k8s and it worked just fine. I had to do some minor moficiations to the script:

* My PVC was named "omada-controller" instead of the provided "omada-data-pvc"
* I use a subpath in my PVC so that data and logs are in the same PVC so i had to add that.

But after those, everything went without a hitch. Thank you so much for providing a really nice upgrade script that just worked.

mbentley, 2025-11-09:
OK, https://github.com/mbentley/docker-omada-controller/pull/666 takes care of the very last of this so I can finally close this out.