mbentley/docker-omada-controller #445: [Bug]: Container fails to start after upgrading to 5.14

ksoviero, 2024-07-17:
### Controller Version

approx. 5.14.X

### Describe the Bug

After upgrading to latest-chromium, which at the time of this writing is 5.14.26.1-chromium, the container failed to start with the following error (full log available separately below):

```
Caused by: org.springframework.beans.factory.NoSuchBeanDefinitionException: No bean named 'commThreadPool' available
```

Upon downgrading back to 5.13 and removing the LAST_RAN_OMADA_VER.txt file, all is well again, so it appears not to be data corruption or anything like that, but rather an actual issue with the image or underlying application. 

For what it's worth, I'm fine running 5.13, but I thought I'd let you know in case you weren't aware of the issue yet. 

### Expected Behavior

I expected the service to start successfully, not throw an error and die. 

### Steps to Reproduce

1. Have an existing 5.13 instance. (This may also happen without it being an upgrade, but I have no means to test that.)
2. Change the image tag on the statefulset (or whatever you are using to orchestrate this) to 5.14.
3. Watch as the container fails to start and throws errors. 

### How You're Launching the Container

```plain
K8s, as a stateful set.
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'America/Chicago'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
INFO: Version check passed; image version (5.14.26.1) >= the last version ran (5.13.30.8); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: Starting Omada Controller as user omada
07-17-2024 02:24:43.217 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
07-17-2024 02:24:43.240 INFO [main] [] c.t.s.o.s.e.c(): Configuration omadacType is linux
07-17-2024 02:24:43.185 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
07-17-2024 02:24:43.189 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
07-17-2024 02:24:43.214 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
07-17-2024 02:24:43.239 INFO [main] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
07-17-2024 02:24:43.240 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
07-17-2024 02:24:43.313 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
07-17-2024 02:25:03.023 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start
07-17-2024 02:25:02.393 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 209
07-17-2024 02:25:02.975 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
07-17-2024 02:25:03.023 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
07-17-2024 02:25:02.395 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
07-17-2024 02:25:02.395 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
07-17-2024 02:25:02.976 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
 
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
 =========|_|==============|___/=/_/_/_/
  .   ____          _            __ _ _
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 :: Spring Boot ::               (v2.7.18)
 
07-17-2024 02:25:03.587 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.14.26.1 using Java 17.0.11 on omada-0 with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.14.26.1.jar started by omada in /opt/tplink/EAPController/lib)
07-17-2024 02:25:03.589 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
 at com.tplink.smb.omada.starter.OmadaBootstrap.b(SourceFile:201) ~[local-starter-5.14.26.1.jar:5.14.26.1]
 at com.tplink.smb.omada.starter.OmadaLinuxMain.b(SourceFile:87) ~[local-starter-5.14.26.1.jar:5.14.26.1]
Caused by: org.springframework.beans.factory.NoSuchBeanDefinitionException: No bean named 'commThreadPool' available
 at org.springframework.beans.factory.support.DefaultListableBeanFactory.getBeanDefinition(DefaultListableBeanFactory.java:874) ~[spring-beans-5.3.31.jar:5.3.31]
 at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:208) ~[spring-beans-5.3.31.jar:5.3.31]
07-17-2024 02:25:09.550 INFO [Thread-0] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 209
 at com.tplink.smb.omada.common.concurrent.thread.a.a(SourceFile:243) ~[omada-common-5.14.26.1.jar:5.14.26.1]
 at java.lang.Thread.run(Thread.java:840) [?:?]
 at org.springframework.beans.factory.support.AbstractBeanFactory.getMergedLocalBeanDefinition(AbstractBeanFactory.java:1358) ~[spring-beans-5.3.31.jar:5.3.31]
 ... 4 more
java.lang.ExceptionInInitializerError: null
07-17-2024 02:25:09.547 INFO [Thread-0] [] c.t.s.o.s.OmadaBootstrap(): Failed to shutdown customThread.
 at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:309) ~[spring-beans-5.3.31.jar:5.3.31]
 at org.springframework.context.support.AbstractApplicationContext.getBean(AbstractApplicationContext.java:1168) ~[spring-context-5.3.31.jar:5.3.31]
 at com.tplink.smb.omada.common.spring.a.a(SourceFile:28) ~[omada-common-5.14.26.1.jar:5.14.26.1]
 at com.tplink.smb.omada.common.concurrent.thread.a$a.<clinit>(SourceFile:55) ~[omada-common-5.14.26.1.jar:5.14.26.1]
07-17-2024 02:25:12.585 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type; You might want to check your annotation setup at the converter implementation
07-17-2024 02:25:16.601 INFO [Thread-0] [] c.t.s.o.s.OmadaBootstrap(): success to shutdown mongodb database
07-17-2024 02:25:16.601 INFO [Thread-0] [] c.t.s.o.s.OmadaBootstrap(): Omada Controller exited
07-17-2024 02:25:16.601 INFO [Thread-0] [] c.t.s.o.s.OmadaLinuxMain(): ShutdownHook: service stopped.
```

mbentley, 2024-07-17:
Could you share the yaml or however you're actually creating the stateful set just to verify there isn't anything odd?  I've verified that the image in question does run as expected when I do run it directly on Docker.

There are certainly a number of problems with the 5.14 release from TP-Link (like #418) but this is a new one to me.  Doesn't look to be anything related to the containerization of the application though at first glance.

ksoviero, 2024-07-17:
```yaml
# Stateful Set for Omada management controller.
apiVersion: apps/v1
kind: StatefulSet
metadata:
  name: omada
spec:
  serviceName: omada
  strategy:
    type: Recreate
  selector:
    matchLabels:
      name: omada
  template:
    metadata:
      name: omada
      labels:
        name: omada
    spec:
      containers:
        - name: omada
          image: mbentley/omada-controller:latest-chromium
          imagePullPolicy: Always
          ports:
          {{- range tuple "8088" "8043" "8843" "29812" "27001" "29810" "29811" "29814" "29815" "29816" "29813" }}
            - containerPort: {{ . }}
              name: port-{{ . }}-tcp
              protocol: TCP
            - containerPort: {{ . }}
              name: port-{{ . }}-udp
              protocol: UDP
          {{- end }}
          env:
            - name: TZ
              value: America/Chicago
          volumeMounts:
            - mountPath: /opt/tplink/EAPController/data
              name: omada-data
            - mountPath: /opt/tplink/EAPController/logs
              name: omada-logs
          livenessProbe:
            tcpSocket:
              port: 8043
          readinessProbe:
            tcpSocket:
              port: 8043
          resources:
            requests:
              cpu: 100m
              memory: 2G
      volumes:
        - name: omada-data
          persistentVolumeClaim:
            claimName: omada-data
        - name: omada-logs
          persistentVolumeClaim:
            claimName: omada-logs
---
apiVersion: v1
kind: Service
metadata:
  name: omada
  annotations:
    hostnames: omada.srv.engineereverything.io
spec:
  externalTrafficPolicy: Local
  type: LoadBalancer
  selector:
    name: omada
  ports:
  {{- range tuple "8088" "8043" "8843" "29812" "27001" "29810" "29811" "29814" "29815" "29816" "29813" }}
    - port: {{ . }}
      name: port-{{ . }}-tcp
      protocol: TCP
    - port: {{ . }}
      name: port-{{ . }}-udp
      protocol: UDP
  {{- end }}
```

ksoviero, 2024-07-17:
> Could you share the yaml or however you're actually creating the stateful set just to verify there isn't anything odd? I've verified that the image in question does run as expected when I do run it [...]

Regarding the ticket you linked to, I think they may actually be the same issue. So, the error I posted was the first time it attempted to start after the upgrade (which took some doing to find by the way, yay for log management) since I figured the first error would be the most useful for debugging, but subsequent startup attempts actually look different. 

Here is an example of the output from the crash loop: https://gist.github.com/ksoviero/efbd966ce5b11bdfb9d02ab2b9d51514

Hope that helps!

mbentley, 2024-07-17:
Ah yeah, that’s definitely the same issue.  The logs you shared might help TP-Link with additional information though!

mbentley, 2024-07-17:
Copied the logs over to that thread.  Going to go ahead and close this out since it turned out to be the same error in the end.

closed 2024-07-17 by mbentley