mbentley/docker-omada-controller #449: [Bug]: Docker Build 32cdad2e3888 is broken

ARLBR10, 2024-07-19:
### Controller Version

Docker 5.13 (Stable) -> Docker 5.14 (Stable)

### Describe the Bug

The build looks like its broken because running on a preloaded environment and new environment gives errors while startup that make it unable to work.

### Expected Behavior

Wait to sucessfully logs and panel to show

### Steps to Reproduce

(PRE-MADE INVIROMENT)
1. edit the docker-compose.yml
2. go to the line `image: mbentley/omada-controller:5.13` and edit '5.13' to '5.14'
3. delete the previous container (optional)
4. use 'docker compose up -d' 

(NEW ENVIROMENT)
1. download the docker-compose.yml
2. use 'docker compose up -d' 

### How You're Launching the Container

```plain
I am using 'docker compose' with a macvlan to create an own container lan ip

Worked fine in 5.13 

docker-compose.yml (Version 5.13)

version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.13
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    networks:
      homemacvlan:
        ipv4_address: 192.168.250.2
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=80
      - MANAGE_HTTPS_PORT=443
      - PORTAL_HTTP_PORT=80
      - PORTAL_HTTPS_PORT=443
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - PORT_TRANSFER_V2=29815
      - PORT_RTTY=29816
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Etc/UTC
    volumes:
      - omada-data:/opt/tplink/EAPController/data
[... 125 lines cut ...]
2024-07-19T03:35:11.174+0000 I CONTROL  [initandlisten] 
2024-07-19T03:35:11.174+0000 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
2024-07-19T03:35:11.174+0000 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
2024-07-19T03:35:11.174+0000 I CONTROL  [initandlisten] 
2024-07-19T03:35:11.189+0000 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '../data/db/diagnostic.data'
2024-07-19T03:35:11.189+0000 I NETWORK  [initandlisten] waiting for connections on port 27217
2024-07-19T03:35:12.158+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:36512 #1 (1 connection now open)
2024-07-19T03:35:12.158+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:36510 #2 (2 connections now open)
2024-07-19T03:35:12.170+0000 I NETWORK  [conn2] received client metadata from 127.0.0.1:36510 conn2: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "6.5.0-44-generic" }, platform: "Java/Ubuntu/17.0.11+9-Ubuntu-120.04.2" }
2024-07-19T03:35:12.171+0000 I NETWORK  [conn1] received client metadata from 127.0.0.1:36512 conn1: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "6.5.0-44-generic" }, platform: "Java/Ubuntu/17.0.11+9-Ubuntu-120.04.2" }
2024-07-19T03:35:12.242+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:36524 #3 (3 connections now open)
2024-07-19T03:35:12.243+0000 I NETWORK  [conn3] received client metadata from 127.0.0.1:36524 conn3: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "6.5.0-44-generic" }, platform: "Java/Ubuntu/17.0.11+9-Ubuntu-120.04.2" }
2024-07-19T03:35:12.270+0000 I NETWORK  [conn3] end connection 127.0.0.1:36524 (2 connections now open)
2024-07-19T03:35:12.271+0000 I NETWORK  [conn1] end connection 127.0.0.1:36512 (1 connection now open)
2024-07-19T03:35:12.271+0000 I NETWORK  [conn2] end connection 127.0.0.1:36510 (0 connections now open)
2024-07-19T03:35:12.318+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:36534 #4 (1 connection now open)
2024-07-19T03:35:12.318+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:36532 #5 (2 connections now open)
2024-07-19T03:35:12.319+0000 I NETWORK  [conn4] received client metadata from 127.0.0.1:36534 conn4: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "6.5.0-44-generic" }, platform: "Java/Ubuntu/17.0.11+9-Ubuntu-120.04.2" }
2024-07-19T03:35:12.319+0000 I NETWORK  [conn5] received client metadata from 127.0.0.1:36532 conn5: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "6.5.0-44-generic" }, platform: "Java/Ubuntu/17.0.11+9-Ubuntu-120.04.2" }
2024-07-19T03:35:12.323+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:36544 #6 (3 connections now open)
2024-07-19T03:35:12.324+0000 I NETWORK  [conn6] received client metadata from 127.0.0.1:36544 conn6: { driver: { name: "mongo-java-driver|sync", version: "4.6.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "6.5.0-44-generic" }, platform: "Java/Ubuntu/17.0.11+9-Ubuntu-120.04.2" }
2024-07-19T03:35:12.330+0000 I NETWORK  [conn6] end connection 127.0.0.1:36544 (2 connections now open)
2024-07-19T03:35:12.330+0000 I NETWORK  [conn4] end connection 127.0.0.1:36534 (1 connection now open)
2024-07-19T03:35:12.330+0000 I NETWORK  [conn5] end connection 127.0.0.1:36532 (0 connections now open)
```

ARLBR10, 2024-07-19:
I forgot to mention that this is the latest version at the moment and i was just trying to update and found an "error" if its applicable

mbentley, 2024-07-19:
Duplicate of #418; [see this post](https://github.com/mbentley/docker-omada-controller/issues/418#issuecomment-2223732459) for instructions on rolling back to 5.13

closed 2024-07-19 by mbentley