mbentley/docker-omada-controller #451: [Bug]: org.springframework.beans.factory.UnsatisfiedDependencyException after update Omada controller on Truenas SCALE

Naldjer, 2024-07-21:
### Controller Version

v5.14

### Describe the Bug

After the newest update from v5.14, the app is stuck in deploying. When im checking the logs, i see allot of bean errors.

[omada-controller_omada-controller-58b5b5d785-mmf95_omada-controller (2).log](https://github.com/user-attachments/files/16322695/omada-controller_omada-controller-58b5b5d785-mmf95_omada-controller.2.log)

### Expected Behavior

Starting from the app normally.

### Steps to Reproduce

1. update the app via Truenas scale
2. wait until the update has been finished
3. wait until getting this errors

What ive done to solve the error:
- install a new instance of the controller failed. Same error.
- stop the application and run it over
- Reboot TrueNas scale bare metal server

### How You're Launching the Container

```plain
Via the "apps" tab in TrueNas
```

### Container Logs

```plain
2024-07-21 02:15:03.038849+02:00org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.domain.model.openapi.c.q': Unsatisfied dependency expressed through field 'e'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.common.a.a': Unsatisfied dependency expressed through field 'c'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.common.e.a': Unsatisfied dependency expressed through field 'b'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.b.p': Unsatisfied dependency expressed through field 'm'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.domain.model.user.A': Unsatisfied dependency expressed through field 'c'; nested exception is org.springframework.beans.factory.BeanCurrentlyInCreationException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.mongo.adaptor.persistence.tenant.a': Bean with name 'com.tplink.smb.omada.identityaccess.port.mongo.adaptor.persistence.tenant.a' has been injected into other beans [com.tplink.smb.omada.identityaccess.domain.model.d.q] in its raw version as part of a circular reference, but has eventually been wrapped. This means that said other beans do not use the final version of the bean. This is often the result of over-eager type matching - consider using 'getBeanNamesForType' with the 'allowEagerInit' flag turned off, for example.
2024-07-21 02:15:03.038883+02:00at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.resolveFieldValue(AutowiredAnnotationBeanPostProcessor.java:713) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038894+02:00at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:693) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038911+02:00at org.springframework.beans.factory.annotation.InjectionMetadata.inject(InjectionMetadata.java:119) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038922+02:00at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor.postProcessProperties(AutowiredAnnotationBeanPostProcessor.java:408) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038939+02:00at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1431) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038949+02:00at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:619) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038959+02:00at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:542) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038974+02:00at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:335) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.038985+02:00at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.039001+02:00at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:333) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.039013+02:00at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:208) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.039022+02:00at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:955) ~[spring-beans-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.039038+02:00at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:929) ~[spring-context-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.039053+02:00at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:591) ~[spring-context-5.3.31.jar:5.3.31]
2024-07-21 02:15:03.039070+02:00at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:147) ~[spring-boot-2.7.18.jar:2.7.18]
2024-07-21 02:15:03.039081+02:00at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:732) ~[spring-boot-2.7.18.jar:2.7.18]
2024-07-21 02:15:03.039091+02:00at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:409) ~[spring-boot-2.7.18.jar:2.7.18]
2024-07-21 02:15:03.039106+02:00at org.springframework.boot.SpringApplication.run(SpringApplication.java:308) ~[spring-boot-2.7.18.jar:2.7.18]
2024-07-21 02:15:03.039117+02:00at org.springframework.boot.SpringApplication.run(SpringApplication.java:1300) ~[spring-boot-2.7.18.jar:2.7.18]
2024-07-21 02:15:03.039127+02:00at org.springframework.boot.SpringApplication.run(SpringApplication.java:1289) ~[spring-boot-2.7.18.jar:2.7.18]
2024-07-21 02:15:03.039144+02:00at com.tplink.smb.omada.starter.task.SpringBootStartUpTask.execute(SourceFile:33) ~[local-starter-5.14.26.1.jar:5.14.26.1]
2024-07-21 02:15:03.039154+02:00at com.tplink.smb.omada.starter.task.c.a(SourceFile:20) ~[local-starter-5.14.26.1.jar:5.14.26.1]
2024-07-21 02:15:03.039164+02:00at com.tplink.smb.omada.starter.OmadaBootstrap.f(SourceFile:363) ~[local-starter-5.14.26.1.jar:5.14.26.1]
2024-07-21 02:15:03.039181+02:00at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:99) ~[local-starter-5.14.26.1.jar:5.14.26.1]
2024-07-21 02:15:03.039191+02:00at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:42) ~[local-starter-5.14.26.1.jar:5.14.26.1]
2024-07-21 02:15:03.039203+02:00Caused by: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.common.a.a': Unsatisfied dependency expressed through field 'c'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.common.e.a': Unsatisfied dependency expressed through field 'b'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.b.p': Unsatisfied dependency expressed through field 'm'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.domain.model.user.A': Unsatisfied dependency expressed through field 'c'; nested exception is org.springframework.beans.factory.BeanCurrentlyInCreationException: Error creating bean with name 'com.tplink.smb.omada.identityaccess.port.mongo.adaptor.persistence.tenant.a': Bean with name 'com.tplink.smb.omada.identityaccess.port.mongo.adaptor.persistence.tenant.a' has been injected into other beans [com.tplink.smb.omada.identityaccess.domain.model.d.q] in its raw version as part of a circular reference, but has eventually been wrapped. This means that said other beans do not use the final ver [... cut ...]
2024-07-21 02:15:03.039398+02:00at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.resolveFieldValue(AutowiredAnnotationBeanPostProcessor.java:710) ~[spring-beans-5.3.31.jar:5.3.31]
```

### Additional Context

The controller worked just fine before the update. it was in the "started" state, and i couldve reach the portal.

Apps running on the Truenas server:

Homepage > off
Metube > off
PiHole > on , but still wont start while this app is off
Unify controller > on, but still wont start while this app is off.

Furthermore there has nothing been changed to the truenas server except for that update to the Omada controller app.

mbentley, 2024-07-21:
Duplicate of #418; see the first post for a link to instructions to roll back to 5.13

Naldjer, 2024-07-21:
The strange thing is, it did work with 5.14.

mbentley, 2024-07-21:
Yikes, I was afraid of that happening to someone. It’s an upstream software problem that o lot TP-Link can fix.  It’s random so it is hard to figure out when it will happen.  Your rollback will probably involve rolling back using a backup so I hope you have one.

Naldjer, 2024-07-21:
Unfortunatly i dont havent made a backup yet. Was planning to make it tomorrow when i finished the config

mbentley, 2024-07-21:
Just FYI, the app itself has the ability to automatically create backups to a format that can be re imported easily but they don’t enable it by default I don’t think.

Naldjer, 2024-07-21:
I did enabled that feature, yet i dont know if the backup already has been made.

mbentley, 2024-07-21:
There should be an autobackup folder inside the data directory, if so.

Naldjer, 2024-07-21:
I have one data backup from 7 days ago in that case, which has been made on 5.14.26

Naldjer, 2024-07-21:
Trying to rollback, but do you have any clue how i can do this in Truenas since theres no way to give any taginformation into the "standard" catalog apps? I can only choose for the latest update which should be 5.14

mbentley, 2024-07-21:
I’m not sure how TrueNAS/ interface works for Docker. Or doesn’t allow you to specify a version tag at all when selecting something from the catalog?

Naldjer, 2024-07-21:
[image]

The only way i could try to add a tag is via the "Additional Environment Variables" attribute, yet i dont know if that's gonna work.

mbentley, 2024-07-22:
I took a quick look and while it looks like there should be a way to allow for multiple versions to be defined in [the catalog definition that they have for the Omada Controller](https://github.com/truenas/charts/tree/80e98e2f6e579182ff06d78229695e828f82c24a/community/omada-controller), it only has the very latest and no other versions listed where it seems that version 1.2.4 of the app in their catalog maps only to 5.14.  I imagine that there must be some way to just deploy a container with your own equivalent of a run command but it is hard to say how it's obfuscated how your previous container was ran.  They really ought to provide a way to run different versions.

closed 2024-07-26 by mbentley