mbentley/docker-omada-controller #452: [Bug]: No image i run works anymore.

awptechnologies, 2024-07-24:
### Controller Version

v5.14 5.13 

### Describe the Bug

I have deleted all volumes and nothing seems to work it will not start

### Expected Behavior

Run like normall

### Steps to Reproduce

start container

### How You're Launching the Container

```plain
docker compose up -d
```

### Container Logs

```plain
Failed less than a second ago    "task: non-zero exit (1)"
```

mbentley, 2024-07-24:
You didn't post any logs but I have to imagine this is a duplicate of #418; [see this post](https://github.com/mbentley/docker-omada-controller/issues/418#issuecomment-2223732459) for instructions on rolling back to 5.13

closed 2024-07-24 by mbentley

awptechnologies, 2024-07-25:
I think it had something to do with docker update. I run this in docker swarm and it is able to move between 3 hosts. I rebooted deleted all the volumes and the latest image started with no issue. I run a restore in omada from my 5.13 backup and everything is good. I was running into issues but i finally came to conclusion my ap the EAP660 has failed.