mbentley/docker-omada-controller #467: [Bug]: EAP_MONGOD_URI gets truncated and additional settings lost when addition options specified

do-i-need-a-username, 2024-09-25:
### Controller Version

5.14.20.9-amd64

### Describe the Bug

This happens in most of the versions I've tested, `mbentley/omada-controller:beta-5.14.20.9-amd64`, `bentley/omada-controller:5.13-external-mongo-test-amd64`, `mbentley/omada-controller:5.13` etc.

EAP_MONGOD_URI gets truncated if it contains an `&` character. Most likely due to the use of `eval`. 

A EAP_MONGOD_URI setting of 
`mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=true&retryWrites=false&connectTimeoutMS=600000` 
becomes `mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=true` in `/opt/tplink/EAPController/properties/omada.properties`

### Expected Behavior

The entire URI to be present in the `omada.properties` file

### Steps to Reproduce

Set some additional [connection options](https://www.mongodb.com/docs/manual/reference/connection-string-options/) in the EAP_MONGOD_URI string and run the container, for example `.../omada?tls=true&retryWrites=false&connectTimeoutMS=600000`

### How You're Launching the Container

```plain
I've run it locally and in a Kubernetes cluster.
```

### Container Logs

```plain
Probably no useful logs 

INFO: Setting 'eap.mongod.uri' to mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=true in omada.properties
```

Despite EAP_MONGOD_URI containing `.../omada?tls=true&retryWrites=false&connectTimeoutMS=600000`.
```

### Additional Context

I've been trying to run the container in a Kubernetes cluster on aws with documentdb which isn't going to work due to implicit creation of collections not being possible in documentdb but the issue still exists for anyone who may want to use an external mongo instance with additional options in the connection string.

I understand an external mongo instance is still experimental and I'm on my own in terms of getting help but I thought it's still worth reporting.

mbentley, 2024-09-25:
Hmm, I'll have to think about how to handle this as you're right, it's due to `eval` which was used to allow injection of credentials without needing them to be directly passed in the environment variable.  One way to resolve this would be requiring escaping the ampersands with a `\` but I'm curious if I can urlencode the string and then decode it after the eval.

do-i-need-a-username, 2024-09-26:
My bad, I did try escaping the `&` but  it didn't work. I was escaping it incorrectly in yaml. Using `\\&` is the correct way to escape it in yaml to have it escaped as `\&` in the env var.

```yaml
env:
 - name: EAP_MONGOD_URI
   value: "mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=true\\&retryWrites=false\\&connectTimeoutMS=600000"
```

sets the env var correctly as 

```bash
EAP_MONGOD_URI=mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=true\&retryWrites=false\&connectTimeoutMS=600000
```

When the entrypoint script runs it outputs the correct value
```
INFO: Setting 'mongo.external' to true in omada.properties
INFO: Setting 'eap.mongod.uri' to mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=true&retryWrites=false&connectTimeoutMS=600000 in omada.properties
```

But then when the app boots it throws an error
```
09-25-2024 23:49:11.347 WARN [main] [] o.m.d.uri(): Ignoring unrecognized boolean value 'trueeap.mongod.uri=mongodb://127.0.0.1:27217/omadaretrywrites=falseeap.mongod.uri=mongodb://127.0.0.1:27217/omadaconnecttimeoutms=600000' in the connection string for 'tls'. Replace with either 'true' or 'false'
```

Perhaps sed is banging up the substitution.
```bash
tail -5 /opt/tplink/EAPController/properties/omada.properties
allow_select_default_vlan=false

# 2024-09-25T23:49:11.018508681
eap.mongod.uri=mongodb://omada:password@my_aws_account.ap-southeast-2.docdb-elastic.amazonaws.com:27017/omada?tls=trueeap.mongod.uri=mongodb://127.0.0.1:27217/omadaretryWrites=falseeap.mongod.uri=mongodb://127.0.0.1:27217/omadaconnectTimeoutMS=600000
mongo.external=true
```

I'm not going to proceed with AWS documentdb due to implicit creation of collections not being possible so hopefully I won't need those additional connection parameters when I eventually get around to either an in cluster mongo instance or mongo cloud.

mbentley, 2024-09-26:
Yeah, `sed` is probably also hitting the problem of needing to escape ampersand characters.  It would probably be better to have the value not need the user to escape them in the env variable but for the entrypoint to escape them for the user.  I didn't do any testing of sed with escaped characters but based on the repeating pattern for each line where the ampersand is, I would be shocked if that isn't it.

do-i-need-a-username, 2024-09-26:
I no longer need to deploy the controller since the [cloud controller](https://www.tp-link.com/au/business-networking/omada-controller-cloud-based/omada-cloud-based-controller/#overview) is apparently free to use. I can close the ticket if you wish. I assume the entry-point script works for 99.9% of people as is.

mbentley, 2024-09-26:
I appreciate the feedback and the help identifying the issues.  I think it's still worth solving so I will continue to work on that in the future.

do-i-need-a-username, 2024-09-27:
No worries. Just to add. I have confirmed that the omada controller works with a mongo cloud instance.

```properties
# 2024-09-27T00:25:24.311836614
eap.mongod.uri=mongodb+srv://omada-controller-svc:<redacted>@my-org-cluster.xyz.mongodb.net/omada?retryWrites=true&w=majority&appName=xyz
mongo.external=true
```

commit 241a6709 (2024-09-30): Escape & of mongodb connection string for eval and sed; fixes #467
commit e6ff187d (2024-09-30): Merge pull request #474 from mbentley/fix-eval-mongodb

Escape & of mongodb connection string for eval and sed; fixes #467
closed 2024-09-30 by mbentley