mbentley/docker-omada-controller #487: [Bug]: Unable to start container due to radius cert generation error

mbentley, 2024-10-16:
```
# docker exec -it omada-controller -- ps auxww
OCI runtime exec failed: exec failed: unable to start container process: exec: "--": executable file not found in $PATH: unknown
root@docker:/opt/DockerCoreServices/portainer# docker logs -n 15 omada-controller
10-16-2024 16:45:05.711 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
10-16-2024 16:45:05.712 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
10-16-2024 16:45:05.712 INFO [main] [] c.t.s.o.s.e.c(): Configuration omadacType is linux
10-16-2024 16:45:05.924 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
10-16-2024 16:45:10.984 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 210
10-16-2024 16:45:10.992 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
10-16-2024 16:45:10.993 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
10-16-2024 16:45:12.322 WARN [main] [] c.t.s.o.s.a.u.X509CertificateUtils(): Valid radius server keystore is missing. Generating one ...
Exception in thread "main" java.lang.IllegalAccessError: class com.tplink.smb.omada.system.api.util.X509CertificateUtils (in unnamed module @0x313ac989) cannot access class sun.security.x509.X500Name (in module java.base) because module java.base does not export sun.security.x509 to unnamed module @0x313ac989
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:508)
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:481)
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:591)
        at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:327)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:92)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:42)
root@docker:/opt/DockerCoreServices/portainer# docker exec -it omada-controller  ps auxww
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
omada          1  0.4  0.6 4519552 216528 ?      Ssl  16:44   0:16 java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true --add-opens java.base/java.util=ALL-UNNAMED -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain
omada        167  0.0  0.0   2552   896 ?        S    16:45   0:00 tail -F -n 0 /opt/tplink/EAPController/logs/server.log
omada        209  0.0  0.0   2616   896 ?        S    16:45   0:00 sh -c "/opt/tplink/EAPController/bin/mongod" --port 27217 --dbpath "../data/db" -pidfilepath "../data/mongo.pid" --logappend --logpath "../logs/mongod.log" --bind_ip 127.0.0.1
omada        210  0.3  0.2 1020036 95104 ?       Sl   16:45   0:13 /opt/tplink/EAPController/bin/mongod --port 27217 --dbpath ../data/db -pidfilepath ../data/mongo.pid --logappend --logpath ../logs/mongod.log --bind_ip 127.0.0.1
root        1961  0.0  0.0   5900  1536 pts/0    Rs+  17:45   0:00 ps auxww
```

_Originally posted by @kraduk in https://github.com/mbentley/docker-omada-controller/issues/485#issuecomment-2417401818_

mbentley, 2024-10-16:
Could you try to override the CMD by running a new container but specifying the CMD manually?  Try to use this which just adds `--add-opens=java.base/sun.security.x509=ALL-UNNAME`.

`java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true --add-opens java.base/java.util=ALL-UNNAMED  --add-opens=java.base/sun.security.x509=ALL-UNNAME -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain`

domushnik, 2024-10-17:
> I believe I can now reproduce this when trying to enable the radius server. Could you try to override the CMD by running a new container but specifying the CMD manually? Try to use this which just adds these two parameters: `--add-opens java.base/sun.security.x509=ALL-UNNAMED --add-opens java.base/sun.security.util=ALL-UNNAMED`.
> 
> `java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true --add-opens java.base/java.util=ALL-UNNAMED --add-opens java.base/sun.security.x509=ALL-UNNAMED --add-opens java.base/sun.security.util=ALL-UNNAMED -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain`

Hello @mbentley ,
I have just tried to update the controller with the latest image 5.14(32.2) and it worked as expected. So I didn't need to add any specific CMDs to the container. Thank you very much for you work.

mbentley, 2024-10-17:
Thanks for that.  They must have moved the cert generation to only when RADIUS is enabled.

Regardless, I created an issue with them to see if they can update the cert generation code to not use these methods: https://community.tp-link.com/en/business/forum/topic/710680?page=1 as it adds additional parameters I have to deal with.

commit 6e364007 (2024-10-18): Merge pull request #488 from mbentley/add-more-opens

Added additional parameters to fix #487
commit 704d3007 (2024-10-18): Added additional parameters to fix #487
closed 2024-10-18 by mbentley