mbentley/docker-omada-controller #490: [Bug]: Omada Router fails to provision or adopt recently

seangreen2, 2024-10-18:
### Controller Version

5.14.32.2

### Describe the Bug

Omada Router TL-R605 v1.0 (1.3.1 Build 20231207 Rel.61384) fails to provision or adopt.

### Expected Behavior

Never had an issue except the past several days where upon a container restart, the router would provision and adopt without issue. Now it always fails and never completes.

### Steps to Reproduce

Issue happens on container start, no steps to reproduce

### How You're Launching the Container

```plain
services:

  omada-controller:

    command:
      - "/usr/bin/java"
      - "-server"
      - "-Xms128m"
      - "-Xmx1024m"
      - "-XX:MaxHeapFreeRatio=60"
      - "-XX:MinHeapFreeRatio=30"
      - "-XX:+HeapDumpOnOutOfMemoryError"
      - "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof"
      - "-Djava.awt.headless=true"
      - "-cp"
      - "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:"
      - "com.tplink.smb.omada.starter.OmadaLinuxMain"

    container_name: "omada-controller"

    entrypoint:
      - "/entrypoint.sh"

    environment:
      - "MANAGE_HTTP_PORT=8088"
      - "MANAGE_HTTPS_PORT=8043"
      - "PORTAL_HTTP_PORT=8088"
      - "PORTAL_HTTPS_PORT=8043"
      - "SHOW_MONGODB_LOGS=false"
      - "SHOW_SERVER_LOGS=true"
      - "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

    hostname: "omada-controller"

    image: "docker.io/mbentley/omada-controller:latest"

    ipc: "private"

    logging:
      driver: "journald"
      options: {}

    network_mode: "bridge"

    ports:
      - "27001:27001/udp"
      - "29810:29810/udp"
      - "29811:29811/tcp"
      - "29812:29812/tcp"
      - "29813:29813/tcp"
      - "29814:29814/tcp"
[... 67 lines cut ...]

10-18-2024 12:29:26.597 INFO [manager-topology-pool-3] [] c.t.s.o.m.d.d.m.m.s.e(): send setMsg to OmadacId(1e949824271f495269cbaa64643afd0e) DeviceMac(90-9A-4A-CD-49-09), status is PROVISIONING, need sync full config, ignore

10-18-2024 12:30:16.696 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.d.a(): MANAGED_BY_OWN Device 90-9A-4A-CD-49-09 on omadac 1e949824271f495269cbaa64643afd0e is discovered.

10-18-2024 12:30:22.382 INFO [app-push-scheduler-0] [] c.t.s.o.l.c.a.a(): Start pushing connected devices.

10-18-2024 12:30:22.519 INFO [app-push-scheduler-0] [] c.t.s.o.l.c.a.a(): Start pushing disconnected devices.

10-18-2024 12:30:24.417 INFO [adopt-work-group-4] [] c.t.s.o.m.d.d.m.a.a(): start watching full adopt result of omadacId:OmadacId(1e949824271f495269cbaa64643afd0e) mac:DeviceMac(90-9A-4A-CD-49-09), businessId:BusinessId(topic=smb_omada_business_f9d75875-1dc5-46ae-9907-1c550cf3031c, id=09b1f3ed-47f7-4a2a-a99d-fbc16d497fb4)

10-18-2024 12:30:24.432 INFO [adopt-work-group-5] [] c.t.s.o.m.d.d.m.d.o.h(): Gateway OmadacId(1e949824271f495269cbaa64643afd0e) SiteId(Default) DeviceMac(90-9A-4A-CD-49-09) adopt[auto=true] ok

10-18-2024 12:30:24.501 INFO [manage-work-group-0] [] c.t.s.o.m.d.a.u.f.b.c(): receive adopt success event for TargetFirmwareOnDeviceAdoptSuccessSubscriber: TargetFirmwareEvent(super=AbstractDomainEvent(id=f864995db53d40cf90071b38bd44a0fe, createTime=Fri Oct 18 12:30:24 UTC 2024, type=null), omadacId=OmadacId(1e949824271f495269cbaa64643afd0e), siteId=SiteId(Default), deviceMac=DeviceMac(90-9A-4A-CD-49-09), compoundModel=TL-R605 v1.0, version=1.3.1, timeStamp=1729254624501)

10-18-2024 12:30:24.501 INFO [adopt-work-group-5] [] c.t.s.o.m.d.d.m.a.e(): send mini setting to OmadacId(1e949824271f495269cbaa64643afd0e) DeviceMac(90-9A-4A-CD-49-09)

10-18-2024 12:30:26.587 WARN [quartzScheduler_Worker-2] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_device_status_push, null) but did not execute because corresponding handler log_device_status_push doesn't exist

10-18-2024 12:30:43.275 WARN [async-business-timeout-pool-0] [] c.t.s.o.m.d.d.m.a.a(): timeout wait adopt watch result of OmadacId(1e949824271f495269cbaa64643afd0e) DeviceMac(90-9A-4A-CD-49-09) BusinessId(topic=smb_omada_business_f9d75875-1dc5-46ae-9907-1c550cf3031c, id=cca51452-9757-4c83-abbc-5a2b1c6841c8), response:4

10-18-2024 12:30:56.600 INFO [comm-pool-14] [] c.t.s.o.m.d.a.u.f.c(): Target firmware check task run

10-18-2024 12:30:56.601 INFO [comm-pool-14] [] c.t.s.o.m.d.a.u.f.c(): no omadac need handle target firmware upgrade
```

mbentley, 2024-10-18:
I only have one suggestion - set `stop_grace_period: 60s` in your compose file.  I don't think that a forceful shutdown of the controller would cause an issue as this is mostly to prevent database corruption but it is worth trying.  See [this part of the README](https://github.com/mbentley/docker-omada-controller?tab=readme-ov-file#preventing-database-corruption) for more info.

If that doesn't help, I would suggest [creating a post on the community forums](https://community.tp-link.com/en/business/forum/582) or filing a support case as this is most likely a problem with the software and not related to the packaging in a container.

mbentley, 2025-02-26:
Closing due to inactivity.

closed 2025-02-26 by mbentley