mbentley/docker-omada-controller #496: [Bug]: Login loop when accessing via NPM

morkyy, 2024-11-08:
### Controller Version

v5.14.32

### Describe the Bug

This started happening after the latest update.

I have configured nginx proxy manager and can access through there via an internal only domain name.

This used to work fine but now it seems that I'm stuck in a login loop.
Login is successful when accessing directly via IP.

I have this running in docker in UNRAID.

only log I can see created on login attempt is the following.

`11-08-2024 11:53:00.949 INFO [https-jsse-nio-18043-exec-6] [] c.t.s.o.i.p.c.b.h(): saveSession. session idle timeout is invalid. idleTimeout: null, tenantId: 65f02f23340a0c0dcaf8ea8d.
11-08-2024 11:53:19.730 INFO [https-jsse-nio-18043-exec-5] [] c.t.s.o.i.p.c.b.h(): saveSession. session idle timeout is invalid. idleTimeout: null, tenantId: 65f02f23340a0c0dcaf8ea8d.
11-08-2024 11:53:55.110 INFO [https-jsse-nio-18043-exec-7] [] c.t.s.o.i.p.c.b.h(): saveSession. session idle timeout is invalid. idleTimeout: null, tenantId: 65f02f23340a0c0dcaf8ea8d.`

### Expected Behavior

Should be able to log in

### Steps to Reproduce

access portal via URL
try to log in

### How You're Launching the Container

```plain
Via Unraid
```

### Container Logs

```plain
`11-08-2024 11:53:00.949 INFO [https-jsse-nio-18043-exec-6] [] c.t.s.o.i.p.c.b.h(): saveSession. session idle timeout is invalid. idleTimeout: null, tenantId: 65f02f23340a0c0dcaf8ea8d.
11-08-2024 11:53:19.730 INFO [https-jsse-nio-18043-exec-5] [] c.t.s.o.i.p.c.b.h(): saveSession. session idle timeout is invalid. idleTimeout: null, tenantId: 65f02f23340a0c0dcaf8ea8d.
11-08-2024 11:53:55.110 INFO [https-jsse-nio-18043-exec-7] [] c.t.s.o.i.p.c.b.h(): saveSession. session idle timeout is invalid. idleTimeout: null, tenantId: 65f02f23340a0c0dcaf8ea8d.`
```

mbentley, 2024-11-08:
I do not use NPM so I don't have a way to simply test it and I don't see enough info here to know how to reproduce your configuration of the container & NPM.  I know there are a [number of people who use traefik](https://github.com/mbentley/docker-omada-controller/discussions/317) and I haven't heard of anyone else having issues so it's hard to say if TP-Link changed or broke something about how reverse proxying is configured.  But for me to help, I am going to need the equivalent of the `docker run` command or at least the `docker inspect <container>` output and whatever information you can provide about how NPM is configured.

morkyy, 2024-11-08:
Here's the docker run command:

`docker run
  -d
  --name='omada-controller'
  --net='host'
  -e TZ="Europe/London"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="Tower"
  -e HOST_CONTAINERNAME="omada-controller"
  -e 'MANAGE_HTTPS_PORT'='18043'
  -e 'MANAGE_HTTP_PORT'='18088'
  -e 'PORTAL_HTTPS_PORT'='18843'
  -e 'PORTAL_HTTP_PORT'='18088'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.webui='https://[IP]:[PORT:18043]/login'
  -l net.unraid.docker.icon='https://raw.githubusercontent.com/benhedrington/hedrington-unraid-docker-templates/main/hedrington-unraid-docker-templates/omada-controller.png'
  -v '/mnt/user/appdata/omada-controller/data/':'/opt/tplink/EAPController/data':'rw,slave'
  -v '/mnt/user/appdata/omada-controller/work/':'/opt/tplink/EAPController/work':'rw,slave'
  -v '/mnt/user/appdata/omada-controller/logs/':'/opt/tplink/EAPController/logs':'rw,slave'
  --ulimit nofile=4096:8192 'mbentley/omada-controller'`

Here's the output of docker inspect:

`[
    {
        "Id": "13c7253449efe71c2656c791001dbc32eb7201b7e37d9eae4ed35407d3664a3b",
        "Created": "2024-11-05T15:28:33.370384781Z",
        "Path": "/entrypoint.sh",
        "Args": [
            "java",
            "-server",
            "-Xms128m",
            "-Xmx1024m",
            "-XX:MaxHeapFreeRatio=60",
            "-XX:MinHeapFreeRatio=30",
            "-XX:+HeapDumpOnOutOfMemoryError",
            "-XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof",
            "-Djava.awt.headless=true",
            "--add-opens",
            "java.base/sun.security.x509=ALL-UNNAMED",
            "--add-opens",
            "java.base/sun.security.util=ALL-UNNAMED",
            "-cp",
            "/opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties:",
            "com.tplink.smb.omada.starter.OmadaLinuxMain"
        ],
        "State": {
            "Status": "running",
            "Running": true,
            "Paused": false,
            "Restarting": false,
            "OOMKilled": false,
            "Dead": false,
            "Pid": 23167,
            "ExitCode": 0,
            "Error": "",
            "StartedAt": "2024-11-08T14:29:02.872541065Z",
            "FinishedAt": "2024-11-08T14:26:12.848823849Z",
            "Health": {
                "Status": "healthy",
                "FailingStreak": 0,
                "Log": [
                    {
                        "Start": "2024-11-08T14:26:00.889576638Z",
                        "End": "2024-11-08T14:26:00.935454803Z",
                        "ExitCode": 0,
                        "Output": "  HTTP/1.1 200 \n  X-Frame-Options: SAMEORIGIN\n  X-Content-Type-Options: nosniff\n  Content-Security-Policy: default-src 'self' https://*.tplinkcloud.com/;script-src 'self' 'unsafe-eval' 'sha256-7W9UiBaYGlOHpT1aQBLegqffUVHbYq6/ZAb+ErjUb40=' 'sha256-VGQ8jNTL2g0e8wPwOgyCQJDqhuRgfV7gRYexcBkBe4Y=' 'sha256-x2jgB1zBLi30IsfY+VNgWjwBGeHPJxOSrzl+IdsT6k0=' 'sha256-0AHZXO4clnpdcxqdmASPBEp4JCIrtaxIX/mUuL1kzZw=' 'sha256-lfXlPY3+MCPOPb4mrw1Y961+745U3WlDQVcOXdchSQc=';style-src 'self' 'unsafe-inline';connect-src 'self' https://*.tplinkcloud.com/ https://*.tplinkcloud.com:8843/ wss://*.tplinkcloud.com/ https://*.tiles.mapbox.com https://api.mapbox.com https://events.mapbox.com ;frame-src 'self' data:;img-src 'self' https://*.tplinkcloud.com/ https://*.mzstatic.com/ https://play-lh.googleusercontent.com/ data: blob:;child-src blob: ;worker-src blob: ;object-src 'self' data: blob:\n  Strict-Transport-Security: max-age=0; includeSubDomains\n  Referrer-Policy: strict-origin-when-cross-origin\n  Content-Type: text/html;charset=utf-8\n  Content-Language: en-US\n  Transfer-Encoding: chunked\n  Date: Fri, 08 Nov 2024 14:26:00 GMT\n  Keep-Alive: timeout=60\n  Connection: keep-alive\n"
                    },
                    {
[... 274 lines cut ...]
                    "IPAddress": "",
                    "IPPrefixLen": 0,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "",
                    "DriverOpts": null
                }
            }
        }
    }
]`

During the image startup I can't see anything wrong except these two errors:

`11-08-2024 14:37:57.547 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
11-08-2024 14:37:57.548 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started`

My NPM config is very simple I just forward 443 traffic from the domain to https://192.168.0.14:18043 with no other config like custom locations etc. This used to work and nothing has changed really other than me updating the docker image.

[image]

And here are the controller settings in the UI.

[image]

morkyy, 2024-11-08:
Here's is the NPM config file as well

`map $scheme $hsts_header {
    https   "max-age=63072000; preload";
}

server {
  set $forward_scheme https;
  set $server         "192.168.0.14";
  set $port           18043;

  listen 80;
listen [::]:80;

  server_name omada.internal.serverakos.com;

# Asset Caching
  include conf.d/include/assets.conf;

  # Block Exploits
  include conf.d/include/block-exploits.conf;

proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection $http_connection;
proxy_http_version 1.1;

  access_log /data/logs/proxy-host-13_access.log proxy;
  error_log /data/logs/proxy-host-13_error.log warn;

  location / {

    

    # Access Rules: 2 total
    
    allow 192.168.0.0/24;
    
    allow 10.253.0.0/24;
    
    deny all;

    # Access checks must...
    
    satisfy all;
    

    
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;
    proxy_http_version 1.1;
    

    # Proxy!
    include conf.d/include/proxy.conf;
  }

  # Custom
  include /data/nginx/custom/server_proxy[.]conf;
}`

forgenator, 2024-11-17:
Just a pointer, I use traefik, and ill let traefik handle ssl certs, and point my instance to omada http, without https redirect. This makes certs and everything work really nice.

mbentley, 2024-11-21:
Sorry for the delay in looking at the configs that you provided; I will try to take a look a bit later today.

morkyy, 2024-12-09:
@mbentley any update on this issue? I'm still having this problem on v5.15.6.7

mbentley, 2024-12-10:
How do you have SSL configured using NPM?  Do your settings look like this?
[image]

[image]

[image]

[image]

Also, could you open your browser's developer tools and see if there are any errors in the console?  For example, here is what I get using the above settings:
[image]

morkyy, 2024-12-10:
My settings in NPM are exactly like the ones one your screenshots.

Weirdly though I'm not getting the same errors in the console.

[image]

I'm getting the same errors on loading the page and on logon attempt.

There seems to be an error with the password field in the form but I don't understand why that would be?

morkyy, 2024-12-10:
exact same log though shows up when accessing via the IP directly where logon works on that. 

So I guess nothing that points out to what the problem actually is.

morkyy, 2024-12-10:
That works and it's probably the best solution tbh and then I can let NPM handle all the cert stuff.

Only question is whether this will break any device updating?

I've had a recurring issue with a remote gateway that I've adopted, not updating it's software cause of IP and SSL issues.

I think last time I was looking into this it was to do with the way the update is pulled from the controller after it was downloaded and was relevant to the internal IP vs external IP since this is a remote gateway.

I think you were the one pointing out the issue in another old thread.

mbentley, 2024-12-10:
It's been a while since I've looked at that specific issue with how it works with the device updates - I honestly don't know or recall at the moment and would have to refer back to a previous thread about the update behavior, assuming none of that has changed in the app at this point.

morkyy, 2024-12-10:
Yeah so the updating issue I have which is slightly relevant with the reverse proxy usage is covered here:

https://community.tp-link.com/en/business/forum/topic/598378

https://github.com/mbentley/docker-omada-controller/discussions/316

It's even more complicated when the device you managing is remote. Played a bit with it now again and managed to find a solution but it's far from ideal.

For some reason for updating, gateways/switches require access to the Omada controller management port (18043).

this essentially means that if you want to update a device you need to get it to bypass the reverse proxy. It's very messy with remote hosts because you basically need to expose port 10843 to the internet without a proxy in between.

Just tested it and it works but not a great solution really.

smithj33, 2025-02-24:
I usually use Traefik, but have NPM running just for testing. I just added my controller to NPM to test it out.

On the controller I have enabled Redirect HTTP to HTTPS, Auto Refresh Portal IP, HTTP redirect to HTTPS for Portal.
On NPM I have https, controller IP, port 8043, and enabled Block Common Exploits, Websockets, Force SSL.

Using these settings works perfect every time.
Now, if in NPM is switch to http and port 8088, leaving everything else the same, I get the login screen, but cannot move forward.

So I would say NPM works fine on the latest version of the controller with the settings above.