mbentley/docker-omada-controller #509: [Bug]: 5.15 does not start

tr1plus, 2024-12-08:
### Controller Version

5.15

### Describe the Bug

This was my first time using this software. When starting with tag latest/5.15 the logs stop at the following:

```
##############################################################################
##############################################################################
WARNGING: autobackup directory not found! Please configure automatic backups!
  For instructions, see https://github.com/mbentley/docker-omada-controller#controller-backups
##############################################################################
##############################################################################

INFO: Starting Omada Controller as user omada
12-08-2024 21:43:47.526 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
12-08-2024 21:43:47.533 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
12-08-2024 21:43:47.562 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
12-08-2024 21:43:47.570 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
12-08-2024 21:43:47.590 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
12-08-2024 21:43:47.612 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
12-08-2024 21:43:47.613 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
12-08-2024 21:43:47.613 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
12-08-2024 21:43:47.696 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
12-08-2024 21:43:51.206 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 228
12-08-2024 21:43:51.207 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
12-08-2024 21:43:51.208 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
12-08-2024 21:43:51.970 WARN [main] [] c.t.s.o.s.a.u.X509CertificateUtils(): Valid radius server keystore is missing. Generating one ...
Exception in thread "main" java.lang.NoSuchFieldError: id_alg_zlibCompress
        at org.bouncycastle.asn1.cms.CMSObjectIdentifiers.<clinit>(Unknown Source)
        at org.bouncycastle.operator.DefaultSignatureNameFinder.<clinit>(Unknown Source)
        at org.bouncycastle.operator.jcajce.OperatorHelper.<clinit>(Unknown Source)
        at org.bouncycastle.operator.jcajce.JcaContentSignerBuilder.<init>(Unknown Source)
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:493)
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:609)
        at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:381)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:96)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:46)
```

I started up everything using version/tag 3.13 (as I saw a lot of issues with 3.14 I wanted to go to a stable version) and everything is working there. Once I had adopted my switch & EAP, I tried to re start the container with latest tag and I got the same error as above.

Currently rolling back to 3.13 (as I had backed up the data dir)

### Expected Behavior

5.15 should start without issue, same as with the 3.13 I  used

### Steps to Reproduce

1. Copy docker compose file from example
2. Run docker compose up
3. Change directories
4. Observer this fails

### How You're Launching the Container

```plain
docker compose up

version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.13 #or 5.15, or latest
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - PORT_TRANSFER_V2=29815
      - PORT_RTTY=29816
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Europe/Brussels
    volumes:
      - ${dir}/data:/opt/tplink/EAPController/data
      - ${dir}/logs:/opt/tplink/EAPController/logs
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'Europe/Brussels'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
WARN: Ownership not set correctly on '/opt/tplink/EAPController/properties'; setting correct ownership (omada:omada)
INFO: Version check passed; image version (5.15.6.7) >= the last version ran (5.13.30.8); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: output of 'java -version':
openjdk version "17.0.13" 2024-10-15
OpenJDK Runtime Environment (build 17.0.13+11-Ubuntu-2ubuntu120.04)
OpenJDK 64-Bit Server VM (build 17.0.13+11-Ubuntu-2ubuntu120.04, mixed mode, sharing)

##############################################################################
##############################################################################
WARNGING: autobackup directory not found! Please configure automatic backups!
  For instructions, see https://github.com/mbentley/docker-omada-controller#controller-backups
##############################################################################
##############################################################################

INFO: Starting Omada Controller as user omada
12-08-2024 21:43:47.526 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
12-08-2024 21:43:47.533 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
12-08-2024 21:43:47.562 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
12-08-2024 21:43:47.570 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
12-08-2024 21:43:47.590 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
12-08-2024 21:43:47.612 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
12-08-2024 21:43:47.613 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
12-08-2024 21:43:47.613 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
12-08-2024 21:43:47.696 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
12-08-2024 21:43:51.206 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 228
12-08-2024 21:43:51.207 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
12-08-2024 21:43:51.208 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
12-08-2024 21:43:51.970 WARN [main] [] c.t.s.o.s.a.u.X509CertificateUtils(): Valid radius server keystore is missing. Generating one ...
Exception in thread "main" java.lang.NoSuchFieldError: id_alg_zlibCompress
        at org.bouncycastle.asn1.cms.CMSObjectIdentifiers.<clinit>(Unknown Source)
        at org.bouncycastle.operator.DefaultSignatureNameFinder.<clinit>(Unknown Source)
        at org.bouncycastle.operator.jcajce.OperatorHelper.<clinit>(Unknown Source)
        at org.bouncycastle.operator.jcajce.JcaContentSignerBuilder.<init>(Unknown Source)
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:493)
        at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:609)
        at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:381)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:96)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:46)
```

mbentley, 2024-12-08:
I am assuming that instead of 3.13 and 3.14 that you mean 5.13 and 5.14.  I can't seem to reproduce this - are you customizing any settings related to RADIUS server config by chance?  Anything in particular that you have configured that might be related to what it is complaining about?

weak-head, 2024-12-08:
@mbentley it happens when you start from scratch and you don't migrate from previous version. 
The workaround is to deploy 5.14 and then use 'latest' or '5.15' tag. 

If you start from scratch and initializing the volumes using '5.15' this happens. To reproduce try to remove the existing docker volumes or use the empty folders with no data inside:
```
    volumes:
      - ./omada/data:/opt/tplink/EAPController/data
      - ./omada/logs:/opt/tplink/EAPController/logs
```

tr1plus, 2024-12-09:
> I am assuming that instead of 3.13 and 3.14 that you mean 5.13 and 5.14. I can't seem to reproduce this - are you customizing any settings related to RADIUS server config by chance? Anything in particular that you have configured that might be related to what it is complaining about?

Yes, I meant 5.xx (sorry I'm a bit sick so less concentrated ;))

As this is also happening on a fresh start (see comment from weak-head) - I think this has nothing to do with any specific configurations.

> @mbentley it happens when you start from scratch and you don't migrate from previous version. The workaround is to deploy 5.14 and then use 'latest' or '5.15' tag.
>
> If you start from scratch and [...]

Thanks for this analysis - I think migrating from 5.13 straight to 5.15 also breaks this. (but maybe I did something wrong with the back up my data dir)

Would you recommended upgrading from 5.13 to 5.14 to 5.15, or should I wait for a fix for now?

mbentley, 2024-12-09:
OK, that's odd then because I can create new controllers and have no issues with startup.  That's the one test that I always do before pushing changes on new versions.  Could you both provide some info for me:

- Output of `docker info`
- Output of `cat /etc/os-release`
- What filesystem type is your persistent data stored on?

I'm trying to think of anything else that would cause the controller to not start but just out of curiosity, could you try to use the `5.15-openj9` tag and see if you have the same behavior?

tr1plus, 2024-12-09:
This also happens when upgrading from 5.13 to 5.15, same behavior, I'll try upgrading from 5.13 to 5.14 to 5.15 now.

> * Output of `docker info`
```
Client: Docker Engine - Community
 Version:    27.3.1
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.17.1
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.29.7
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 22
  Running: 21
  Paused: 0
  Stopped: 1
 Images: 23
 Server Version: 27.3.1
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 88bf19b2105c8b17560993bee28a01ddc2f97182
 runc version: v1.2.2-0-g7cb3632
 init version: de40ad0
 Security Options:
  apparmor
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 5.10.0-33-amd64
 Operating System: Debian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: x86_64
 CPUs: 2
 Total Memory: 3.828GiB
 Name: docker
 ID: QYRD:6DY5:QKBG:DZPP:3TSO:75DE:2BRK:N5UE:J7YD:KHXV:FAKZ:JTQI
 Docker Root Dir: /mnt/docker/docker-root
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```

> * Output of `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
[... 51 lines cut ...]
omada-controller  | INFO: Starting Omada Controller as user omada
omada-controller  | 12-09-2024 13:32:22.756 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
omada-controller  | 12-09-2024 13:32:22.764 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
omada-controller  | 12-09-2024 13:32:22.843 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
omada-controller  | 12-09-2024 13:32:22.850 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
omada-controller  | 12-09-2024 13:32:22.866 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
omada-controller  | 12-09-2024 13:32:22.886 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
omada-controller  | 12-09-2024 13:32:22.887 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
omada-controller  | 12-09-2024 13:32:22.887 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
omada-controller  | 12-09-2024 13:32:22.949 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
omada-controller  | 12-09-2024 13:32:25.967 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 237
omada-controller  | 12-09-2024 13:32:25.969 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
omada-controller  | 12-09-2024 13:32:25.970 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
omada-controller  | 12-09-2024 13:32:26.679 WARN [main] [] c.t.s.o.s.a.u.X509CertificateUtils(): Valid radius server keystore is missing. Generating one ...
omada-controller  | Exception in thread "main" java.lang.NoSuchFieldError: org/bouncycastle/asn1/pkcs/PKCSObjectIdentifiers.id_alg_zlibCompress
omada-controller  |     at org.bouncycastle.asn1.cms.CMSObjectIdentifiers.<clinit>(Unknown Source)
omada-controller  |     at org.bouncycastle.operator.DefaultSignatureNameFinder.<clinit>(Unknown Source)
omada-controller  |     at org.bouncycastle.operator.jcajce.OperatorHelper.<clinit>(Unknown Source)
omada-controller  |     at org.bouncycastle.operator.jcajce.JcaContentSignerBuilder.<init>(Unknown Source)
omada-controller  |     at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:493)
omada-controller  |     at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:609)
omada-controller  |     at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:381)
omada-controller  |     at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:96)
omada-controller  |     at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:46)
```

tr1plus, 2024-12-09:
Going from 5.13 -> 5.14 -> 5.15 seems to work.

Starting a new controller on 5.13 works.
Starting a new controller on 5.14 works.
Starting a new controller on 5.15 does not work.

weak-head, 2024-12-09:
If it would be helpful, I have a little bit different error, though it's still related to `X509CertificateUtils`:

```
omada_controller  | INFO: Starting Omada Controller as user omada
omada_controller  | tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
omada_controller  | tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
omada_controller  | 12-09-2024 08:51:10.614 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
omada_controller  | 12-09-2024 08:51:10.616 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
omada_controller  | 12-09-2024 08:51:10.630 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
omada_controller  | 12-09-2024 08:51:10.633 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
omada_controller  | 12-09-2024 08:51:10.644 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
omada_controller  | 12-09-2024 08:51:10.657 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
omada_controller  | 12-09-2024 08:51:10.658 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
omada_controller  | 12-09-2024 08:51:10.658 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
omada_controller  | 12-09-2024 08:51:10.684 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
omada_controller  | Exception in thread "main" java.lang.NoSuchFieldError: xmss_SHAKE128_512ph
omada_controller  |       at org.bouncycastle.operator.DefaultSignatureAlgorithmIdentifierFinder.<clinit>(Unknown Source)
omada_controller  |       at org.bouncycastle.operator.jcajce.JcaContentSignerBuilder.build(Unknown Source)
omada_controller  |       at com.tplink.smb.omada.system.api.util.X509CertificateUtils.a(SourceFile:243)
omada_controller  |       at com.tplink.smb.omada.starter.OmadaBootstrap.b(SourceFile:935)
omada_controller  |       at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:823)
omada_controller  |       at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:371)
omada_controller  |       at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:96)
omada_controller  |       at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:46)
omada_controller  | 12-09-2024 08:51:15.193 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 241
```

weak-head, 2024-12-09:
@tr1plus Your container logs looks similar to the logs from an existing closed 5.14 bug: https://github.com/mbentley/docker-omada-controller/issues/418#issuecomment-2387994104

tr1plus, 2024-12-09:
> @tr1plus Your container logs looks similar to the logs from an existing closed 5.14 bug: [#418 (comment)](https://github.com/mbentley/docker-omada-controller/issues/418#issuecomment-2387994104)

I had looked into that one, but the exception itself is different (it happens at the same certificate generation tho).

`Exception in thread "main" java.lang.IllegalAccessError: class com.tplink.smb.omada.system.api.util.X509CertificateUtils (in unnamed module @0x313ac989) cannot access class sun.security.x509.X500Name (in module java.base) because module java.base does not export sun.security.x509 to unnamed module @0x313ac989`

vs

`Exception in thread "main" java.lang.NoSuchFieldError: org/bouncycastle/asn1/pkcs/PKCSObjectIdentifiers.id_alg_zlibCompress`

mbentley, 2024-12-09:
While I'll continue to see if there is anything I can do to replicate the issue, I would suggest starting a thread on the TP-Link community forums: https://community.tp-link.com/en/business/forum/582 to raise awareness to TP-Link.

I'm running the exact same version of Debian w/the same kernel, and same version of Docker, using the compose file provided and I am not seeing the issue which makes me think it is something going on in the controller software.

tr1plus, 2024-12-09:
Thanks!
In the meantime I will upgrade my existing install to 5.15 by going trough the versions one by one.

mbentley, 2024-12-10:
If you all can please post on the forum post at https://community.tp-link.com/en/business/forum/topic/721750?page=1,  that would be helpful!

mbentley, 2024-12-27:
A workaround was posted:

Delete the files `bcpkix-jdk15on-1.70.jar` and `bcprov-jdk15on-1.70.jar` from the controller and the controller will start.  I may update the entrypoint with a workaround that will be triggered by env var or something to delete them for the user as that's kind of tricky.

erkr, 2025-01-08:
Please help. I tried to setup this controller fresh in home assistant today and also get this MogoDB exception
Question: Where to apply the mentioned WORKAROUND_509 for this Home assistant add-on?

Thanks Eric

mbentley, 2025-01-08:
> Please help. I tried to setup this controller fresh in home assistant today and also get this keystore exception Question: Where to apply the mentioned WORKAROUND_509 for this Home assistant add-on?
> 
> Thanks Eric

I'm not immediately familiar with how it works with home assistant to deploy the container but what you're looking for is the ability to add or modify the default environment variables for the container.  You need to add an environment variable with the name/key `WORKAROUND_509` and a value of `true` for it to work.  Not sure if home assistant exposes the ability to manually specify an env var but I would expect it should somehow.

erkr, 2025-01-08:
Thx, a Developer of the add-on just responded and will look into it. Best Eric

erkr, 2025-02-11:
I managed to do a fresh install of 5.15.6.7 with the workaround enabled.
Later I updated to 5.15.8.2 and it failed to start again. Has to Disable the workaround to get it running

hyacin75, 2025-02-11:
> > I had this the first time I tried to go above 5.13 too and had to rollback, totally forgot and tried again today, rolled back to 5.14, and then to 5.13 which was working previously, and I'm still getting stuck in a 'failed to start MongoDB server' loop :'(
> > Can't even remember what I did before, but I'm guessing I need to restore the persistent volume contents from a backup. :-(
> > Edit: I have no backups ... forgot to reconfigure when I moved clusters :'( ... 5.13 stuck in a bootloop, MongoDB won't start ... up the creek without a paddle :'( ... tried the WORKAROUND_509 but Mongo still won't start :-(
> 
> I can look at logs if you provide them.
> 
> Something similar to `docker logs omada-controller >& output.log` for the controller itself.
> 
> For MongoDB, these can be copied from the container using `docker cp omada-controller:/opt/tplink/EAPController/logs/mongod.log .` or by copying them from the volume/bind mount specified for the logs directory.

Appreciated!

Those mongo logs helped me identify that the rollback is causing issues now, so I'll increment back up to 5.15 and see if I can get anything more useful out of them.

```
2025-02-11T15:55:48.332+0000 F -        [initandlisten] Fatal assertion 28579 UnsupportedFormat: Unable to find metadata for table:index-2095-964445191568798925 Index: {name: site_time, ns: omada_data.oswstat5min_20250209} - version too new for this mongod. See http://dochub.mongodb.org/core/3.4-index-downgrade for detailed instructions on how to handle this error. at src/mongo/db/storage/wiredtiger/wiredtiger_index.cpp 268
```

hyacin75, 2025-02-11:
5.14 kicked off a huge DB upgrade, I guess I was too impatient every time I tried to bump the version -

```
2025-02-11T16:14:51.763+0000 I INDEX    [initandlisten]          building index using bulk method; build may temporarily use up to 83 megabytes of RAM
2025-02-11T16:14:51.924+0000 I INDEX    [initandlisten] build index on: omada_data.widsdata_2025m03_t1s1 properties: { v: 2, key: { site_id: 1, detect_mac: 1, start_time: -1 }, name: "site_id_1_detect_mac_1_start_time_-1", ns: "omada_data.widsdata_2025m03_t1s1", background: true }
2025-02-11T16:14:51.924+0000 I INDEX    [initandlisten]          building index using bulk method; build may temporarily use up to 83 megabytes of RAM
2025-02-11T16:14:52.087+0000 I INDEX    [initandlisten] build index on: omada_data.widsdata_2025m03_t1s1 properties: { v: 2, key: { site_id: 1, level: 1, start_time: -1 }, name: "site_id_1_level_1_start_time_-1", ns: "omada_data.widsdata_2025m03_t1s1", background: true }
2025-02-11T16:14:52.087+0000 I INDEX    [initandlisten]          building index using bulk method; build may temporarily use up to 83 megabytes of RAM
2025-02-11T16:14:52.255+0000 I INDEX    [initandlisten] build index on: omada_data.widsdata_2025m03_t1s1 properties: { v: 2, key: { site_id: 1, channel: 1, start_time: -1 }, name: "site_id_1_channel_1_start_time_-1", ns: "omada_data.widsdata_2025m03_t1s1", background: true }
2025-02-11T16:14:52.255+0000 I INDEX    [initandlisten]          building index using bulk method; build may temporarily use up to 83 megabytes of RAM
2025-02-11T16:14:55.233+0000 I STORAGE  [initandlisten] finished checking dbs
```

When it finished after about 12 minutes, it started up just fine!!  Thanks @mbentley !!

mbentley, 2025-02-11:
That's pretty wild - I've heard of a few other people who had really slow experiences with the database migrations and I am not really sure what exactly is the cause because it seems like it's across a variety of different devices, many with plenty of horsepower and plenty fast storage but it's not always immediately obvious what's going on when there is something blocking in MongoDB.

donbowman, 2025-02-11:
Mine took over 15 minutes on a nanopi r5s w/ emmc flash (RK3568, quad-core A55, 4GB ram). I also stopped it a few times assuming it was dead. I'm assuming it relates to how long its been up and some activity logs inside?

mbentley, 2025-02-26:
I'm going to go ahead and close this.  If anyone has further issues related to this, let me know.

closed 2025-02-26 by mbentley

mbentley, 2025-11-06:
Ah ok, so in your case, you tried to start a clean install on a host where it doesn't meet the processor requirements with of armv8.2-a but apparently I must be writing the version executed before doing the armv8.2-a and AVX checks.  I will have to look into that because ultimately that should be a bug as it was never going to run the container.  Typically removing the last ran version file is a good way for people to break their controllers and not be able to recover without a backup but in your case, it was just a warning about something that must have never started in the first place.  If you really wanted to run 6.x you could [run MongoDB externally from the container](https://github.com/mbentley/docker-omada-controller/blob/master/external_mongodb/README.md).