mbentley/docker-omada-controller #516: [Bug]: "docker compose up -d" fails

mjsteckel, 2024-12-26:
### Controller Version

v5.15

### Describe the Bug

"docker compose up -d" fails with error 

when using https://raw.githubusercontent.com/mbentley/docker-omada-controller/master/docker-compose.yml

```
# docker -D compose up -d 
[+] Running 4/6
[+] Running 6/6ller [⣿⣿⣤⣿⣿] 222.8MB / 392.1MB Pulling                                                                                                                                    ...
[+] Running 1/1
 ✔ Container omada-controller  Started                                                                                                                                                 
**DEBU[0013] otel error                                    error="<nil>"
DEBU[0013] otel error                                    error="<nil>"**
```

Additional into

```
# cat /etc/os-release 
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
```

```
# docker compose version
Docker Compose version 2.27.1+ds1-0ubuntu1~24.04.1
```

```
# docker version
Client:
 Version:           26.1.3
 API version:       1.45
 Go version:        go1.22.2
 Git commit:        26.1.3-0ubuntu1~24.04.1
 Built:             Mon Oct 14 14:29:26 2024
 OS/Arch:           linux/amd64
 Context:           default

Server:
 Engine:
  Version:          26.1.3
  API version:      1.45 (minimum version 1.24)
  Go version:       go1.22.2
  Git commit:       26.1.3-0ubuntu1~24.04.1
  Built:            Mon Oct 14 14:29:26 2024
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.7.12
  GitCommit:        
 runc:
  Version:          1.1.12-0ubuntu3.1
  GitCommit:        
 docker-init:
  Version:          0.19.0
  GitCommit:     
```

### Expected Behavior

Container installs and starts

### Steps to Reproduce

See aove

### How You're Launching the Container

```plain
docker compose up -d
```

### Container Logs

```plain
There are no container logs as the container does not start
```

### Additional Context

The directory "/opt/tplink/" is not created by "docker compose up"

mjsteckel, 2024-12-27:
Turns out to be the same as issue #509. 

Per the reply on the TP-link Omada forum (link below) I deleted the files "bcpkix-jdk15on-1.70.jar" and "bcprov-jdk15on-1.70.jar" and the controller started.

https://community.tp-link.com/en/business/forum/topic/721750?replyId=1460064

The "otel error..." error msg initially misdirected me...

mbentley, 2024-12-27:
I added an env var `WORKAROUND_509` that when set to true, will auto delete those files during the entrypoint processing.

closed 2024-12-27 by mbentley