mbentley/docker-omada-controller #539: [Bug]: MongoDB will not start [5.15-chromium]

travisboss, 2025-02-20:
### Controller Version

5.15-chromium

### Describe the Bug

Updated to latest and tried to go back to 5.14-chromium and getting the same issue now.

I tried with the workaround 509 true or false and same issue.

### Expected Behavior

Container starts
Profit!

### Steps to Reproduce

1. Use docker compose above
2. Change to 5.15-chromium
3. docker compose up -d omada-controller

### How You're Launching the Container

```plain
services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.15-chromium
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=${PUID}
      - PGID=${PGID}
      - MANAGE_HTTP_PORT=${MANAGE_HTTP_PORT}
      - MANAGE_HTTPS_PORT=${MANAGE_HTTPS_PORT}
      - PORTAL_HTTP_PORT=${PORTAL_HTTP_PORT}
      - PORTAL_HTTPS_PORT=${PORTAL_HTTPS_PORT}
      - PORT_APP_DISCOVERY=${PORT_APP_DISCOVERY}
      - PORT_ADOPT_V1=${PORT_ADOPT_V1}
      - PORT_UPGRADE_V1=${PORT_UPGRADE_V1}
      - PORT_MANAGER_V1=${PORT_MANAGER_V1}
      - PORT_MANAGER_V2=${PORT_MANAGER_V2}
      - PORT_DISCOVERY=${PORT_DISCOVERY}
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=${TZ}
      - WORKAROUND_509=true **tried true and false, still will not start**
    volumes:
      - ./Omada/data:/opt/tplink/EAPController/data
      - ./Omada/logs:/opt/tplink/EAPController/logs
    security_opt:
      - no-new-privileges=true

volumes:
  data:
  logs:
```

### Container Logs

```plain
INFO: '/opt/tplink/EAPController/lib/bcprov-jdk15on-1.70.jar' isn't present, skipping
INFO: WORKAROUND_509 complete!
INFO: output of 'java -version':
openjdk version "17.0.14" 2025-01-21
OpenJDK Runtime Environment (build 17.0.14+7-Ubuntu-120.04)
OpenJDK 64-Bit Server VM (build 17.0.14+7-Ubuntu-120.04, mixed mode, sharing)

INFO: Starting Omada Controller as user omada
02-20-2025 11:08:40.794 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
02-20-2025 11:08:40.800 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
02-20-2025 11:08:40.823 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
02-20-2025 11:08:40.827 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
02-20-2025 11:08:40.840 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
02-20-2025 11:08:40.856 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
02-20-2025 11:08:40.856 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
02-20-2025 11:08:40.856 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
02-20-2025 11:08:40.884 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
02-20-2025 11:08:40.888 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 577
02-20-2025 11:08:40.890 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
02-20-2025 11:08:40.890 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
02-20-2025 11:08:42.900 INFO [main] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 577
02-20-2025 11:08:45.427 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 214
02-20-2025 11:08:45.427 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
02-20-2025 11:08:45.427 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
Exception in thread "main" java.lang.NoClassDefFoundError: org/bouncycastle/jce/provider/BouncyCastleProvider
	at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:822)
	at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:371)
	at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:96)
	at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:46)
Caused by: java.lang.ClassNotFoundException: org.bouncycastle.jce.provider.BouncyCastleProvider
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:525)
	... 4 more
02-20-2025 11:08:59.446 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
02-20-2025 11:08:59.456 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.
02-20-2025 11:08:59.456 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 214
02-20-2025 11:09:02.010 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): mongodb process id is 238
02-20-2025 11:09:02.010 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
02-20-2025 11:09:02.010 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Mongo DB server started
02-20-2025 11:09:16.028 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
02-20-2025 11:09:16.028 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.
02-20-2025 11:09:16.028 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 238
02-20-2025 11:09:18.558 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): mongodb process id is 281
02-20-2025 11:09:18.558 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
02-20-2025 11:09:18.558 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Mongo DB server started
02-20-2025 11:09:32.576 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
02-20-2025 11:09:32.577 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.
02-20-2025 11:09:32.577 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 281
02-20-2025 11:09:35.103 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): mongodb process id is 323
02-20-2025 11:09:35.103 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
02-20-2025 11:09:35.103 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Mongo DB server started
02-20-2025 11:09:49.121 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
02-20-2025 11:09:49.121 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.
02-20-2025 11:09:49.121 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 323
02-20-2025 11:09:51.655 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): mongodb process id is 365
02-20-2025 11:09:51.655 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
02-20-2025 11:09:51.655 INFO [mongo-check-thread] [] c.t.s.o.s.s.b(): Mongo DB server started
```

### MongoDB Logs

```plain
I do not have this enabled
```

watchingdogs, 2025-02-21:
I have the same system and I did `WORKAROUND_509` as well.

mbentley, 2025-02-21:
Are you certain that there isn't another container running, accessing the same volume where the persistent data is?  The logs from MongoDB seem to indicate there is another MongoDB process possibly:

```
5.576-0600 I CONTROL  [initandlisten] options: { repair: true, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-02-20T11:15:25.577-0600 I -        [initandlisten] Detected data files in ../data/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
2025-02-20T11:15:25.577-0600 I STORAGE  [initandlisten] Detected WT journal files.  Running recovery from last checkpoint.
2025-02-20T11:15:25.577-0600 I STORAGE  [initandlisten] journal to nojournal transition config: create,cache_size=15512M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2025-02-20T11:15:26.075-0600 E STORAGE  [initandlisten] WiredTiger error (11) [1740071726:75866][442:0x7f5f0f5ebac0], wiredtiger_open: ../data/db/WiredTiger.lock: handle-lock: fcntl: Resource temporarily unavailable
2025-02-20T11:15:26.075-0600 E STORAGE  [initandlisten] WiredTiger error (16) [1740071726:75931][442:0x7f5f0f5ebac0], wiredtiger_open: WiredTiger database is already being managed by another process: Device or resource busy
2025-02-20T11:15:26.076-0600 E -        [initandlisten] Assertion: 28718:16: Device or resource busy src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 397
2025-02-20T11:15:26.076-0600 I STORAGE  [initandlisten] exception in initAndListen: Location28718: 16: Device or resource busy, terminating
2025-02-20T11:15:26.076-0600 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
```

mbentley, 2025-02-21:
These are the MongoDB logs I get when I try to run a 2nd controller using the same persistent data:

```
2025-02-21T22:59:10.986+0000 I CONTROL  [main] ***** SERVER RESTARTED *****
2025-02-21T22:59:10.990+0000 I CONTROL  [initandlisten] MongoDB starting : pid=308 port=27217 dbpath=../data/db 64-bit host=e808260ffe82
2025-02-21T22:59:10.990+0000 I CONTROL  [initandlisten] db version v3.6.8
2025-02-21T22:59:10.990+0000 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
2025-02-21T22:59:10.990+0000 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
2025-02-21T22:59:10.990+0000 I CONTROL  [initandlisten] allocator: tcmalloc
2025-02-21T22:59:10.991+0000 I CONTROL  [initandlisten] modules: none
2025-02-21T22:59:10.991+0000 I CONTROL  [initandlisten] build environment:
2025-02-21T22:59:10.991+0000 I CONTROL  [initandlisten]     distarch: x86_64
2025-02-21T22:59:10.991+0000 I CONTROL  [initandlisten]     target_arch: x86_64
2025-02-21T22:59:10.991+0000 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-02-21T22:59:10.991+0000 I -        [initandlisten] Detected data files in ../data/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
2025-02-21T22:59:10.991+0000 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=63725M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2025-02-21T22:59:11.536+0000 E STORAGE  [initandlisten] WiredTiger error (11) [1740178751:536833][308:0x7fad336bcac0], wiredtiger_open: ../data/db/WiredTiger.lock: handle-lock: fcntl: Resource temporarily unavailable
2025-02-21T22:59:11.536+0000 E STORAGE  [initandlisten] WiredTiger error (16) [1740178751:536897][308:0x7fad336bcac0], wiredtiger_open: WiredTiger database is already being managed by another process: Device or resource busy
2025-02-21T22:59:11.537+0000 E -        [initandlisten] Assertion: 28595:16: Device or resource busy src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 421
2025-02-21T22:59:11.537+0000 I STORAGE  [initandlisten] exception in initAndListen: Location28595: 16: Device or resource busy, terminating
2025-02-21T22:59:11.537+0000 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
2025-02-21T22:59:11.537+0000 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27217.sock
2025-02-21T22:59:11.537+0000 I CONTROL  [initandlisten] now exiting
2025-02-21T22:59:11.537+0000 I CONTROL  [initandlisten] shutting down with code:100
```

travisboss, 2025-02-25:
@mbentley There was no second service using Mongo and I found that odd, a restart did fix it my issue it seems.  This server has not been restarted in a long time and the container was on 5.14 for quite some time too.

watchingdogs, 2025-03-10:
After upgrading from 5.14.32.2 (where it started) I still get this error:
```
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) exists; skipping creation
INFO: User (omada) exists; skipping creation
INFO: Time zone set to 'Europe/Budapest'
INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
INFO: Value of 'port.upgrade.v1' already set to 29813 in omada.properties
INFO: Value of 'port.manager.v1' already set to 29811 in omada.properties
INFO: Value of 'port.manager.v2' already set to 29814 in omada.properties
INFO: Value of 'port.discovery' already set to 29810 in omada.properties
INFO: Value of 'port.transfer.v2' already set to 29815 in omada.properties
INFO: Value of 'port.rtty' already set to 29816 in omada.properties
INFO: Value of 'mongo.external' already set to false in omada.properties
INFO: Value of 'eap.mongod.uri' already set to mongodb://127.0.0.1:27217/omada in omada.properties
INFO: Version check passed; image version (5.15.8.2) >= the last version ran (5.15.8.2); writing image version to last ran file...
INFO: userland/kernel check passed
INFO: WORKAROUND_509=true; deleting files that block controller startup, if present
INFO: '/opt/tplink/EAPController/lib/bcpkix-jdk15on-1.70.jar' isn't present, skipping
INFO: '/opt/tplink/EAPController/lib/bcprov-jdk15on-1.70.jar' isn't present, skipping
INFO: WORKAROUND_509 complete!
INFO: output of 'java -version':
openjdk version "17.0.14" 2025-01-21
OpenJDK Runtime Environment (build 17.0.14+7-Ubuntu-120.04)
OpenJDK 64-Bit Server VM (build 17.0.14+7-Ubuntu-120.04, mixed mode, sharing)

##############################################################################
##############################################################################
WARNING: autobackup directory not found! Please configure automatic backups!
  For instructions, see https://github.com/mbentley/docker-omada-controller#controller-backups
##############################################################################
##############################################################################

INFO: Starting Omada Controller as user omada
03-10-2025 17:12:03.622 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
03-10-2025 17:12:03.647 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
03-10-2025 17:12:03.778 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
03-10-2025 17:12:03.792 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
03-10-2025 17:12:03.840 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
03-10-2025 17:12:03.913 INFO [main] [] c.t.s.o.c.o.c.b(): success to load configuration omada.properties
03-10-2025 17:12:03.915 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
03-10-2025 17:12:03.915 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
03-10-2025 17:12:04.060 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
03-10-2025 17:12:13.098 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 217
03-10-2025 17:12:13.101 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
03-10-2025 17:12:13.102 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
Exception in thread "main" java.lang.NoClassDefFoundError: org/bouncycastle/jce/provider/BouncyCastleProvider
        at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:822)
        at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:371)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:96)
        at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:46)
Caused by: java.lang.ClassNotFoundException: org.bouncycastle.jce.provider.BouncyCastleProvider
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:525)
        ... 4 more
```
... but nothing more. It stops here. Mongodb logs seem normal.

mbentley, 2025-03-10:
@watchingdogs - remove `WORKAROUND_509=true` from your env vars.  That was for a very specific version hitting a very specific problem.

travisboss, 2025-03-14:
@mbentley my issue has now serviced again randomly.

I have moved to 5.15 NONE chromium version, the container is just active and is running 24/7 and just stops working, I cannot start it unless I restart the service, there is only one version of omada running and has been that way for years.

```
9b62df369444   omada-controller            --        -- / --             --        --        --          --
c5154ec6c808   pihole                      --        -- / --             --        --        --          --
eb1405f682ec   threadfin                   --        -- / --             --        --        --          --
e48fcc4922b6   plex                        --        -- / --             --        --        --          --
e4453394a66a   swag                        --        -- / --             --        --        --          --
54a4feb96c3c   diun                        --        -- / --             --        --        --          --
f107a960e11b   crowdsec                    --        -- / --             --        --        --          --
eb19308909d2   bitwarden-backup            --        -- / --             --        --        --          --
c80c565c14d5   bitwarden                   --        -- / --             --        --        --          --
f3cc8f2d3828   authentik-postgresql        --        -- / --             --        --        --          --
9eb4ae1de9dd   authentik-redis             --        -- / --             --        --        --          --
9989f74dcb3a   audiobookshelf              --        -- / --             --        --        --          --
7a8b16c3f6c0   adguardhome                 --        -- / --             --        --        --          --
3797d1f424f1   cloudflare-worker-bouncer   --        -- / --             --        --        --          --
6c80d1934888   jellyfin                    --        -- / --             --        --        --          --
3e5e45d3dfa2   socket-proxy                --        -- / --             --        --        --          --
c5867c0d39dc   pds                         --        -- / --             --        --        --          --
bc359dbcce0f   authentik-server            --        -- / --             --        --        --          --
ebf639bf95c5   authentik-worker            --        -- / --             --        --        --          --
```

```
a:619) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:542) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:335) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:333) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:208) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.config.DependencyDescriptor.resolveCandidate(DependencyDescriptor.java:276) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.doResolveDependency(DefaultListableBeanFactory.java:1391) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1311) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.resolveFieldValue(AutowiredAnnotationBeanPostProcessor.java:710) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:693) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.annotation.InjectionMetadata.inject(InjectionMetadata.java:119) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor.postProcessProperties(AutowiredAnnotationBeanPostProcessor.java:408) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1431) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:619) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:542) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:335) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:333) ~[spring-beans-5.3.31.jar:5.3.31]
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:208) ~[spring-beans-5.3.31.jar:5.3.31]
[... cut ...]
2025-03-14T10:43:42.700-0500 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-03-14T10:43:42.700-0500 E STORAGE  [initandlisten] Failed to set up listener: SocketException: Address already in use
2025-03-14T10:43:42.700-0500 I CONTROL  [initandlisten] now exiting
2025-03-14T10:43:42.700-0500 I CONTROL  [initandlisten] shutting down with code:48
```

travisboss, 2025-03-14:
Running `sudo lsof -i :27217` showing nothing using the port.

travisboss, 2025-03-14:
Service started.

```
travis@gryffindor:~$ sudo lsof -i :27217
COMMAND  PID   USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
java    3038 travis  467u  IPv6  32621      0t0  TCP localhost:52538->localhost:27217 (ESTABLISHED)
java    3038 travis  468u  IPv6  32624      0t0  TCP localhost:52548->localhost:27217 (ESTABLISHED)
java    3038 travis  470u  IPv6  36209      0t0  TCP localhost:32908->localhost:27217 (ESTABLISHED)
mongod  5083 travis   11u  IPv4  29911      0t0  TCP localhost:27217 (LISTEN)
mongod  5083 travis  616u  IPv4  32622      0t0  TCP localhost:27217->localhost:52538 (ESTABLISHED)
mongod  5083 travis  617u  IPv4  32625      0t0  TCP localhost:27217->localhost:52548 (ESTABLISHED)
mongod  5083 travis  618u  IPv4  36210      0t0  TCP localhost:27217->localhost:32908 (ESTABLISHED)
```

mbentley, 2025-03-14:
Odd.  It seems like something is holding or locking the files.  I'm wondering if the MongoDB is terminating suddenly for some reason.  Could you share the entire MongoDB log? If you don't know how to get them, see: https://github.com/mbentley/docker-omada-controller/blob/master/SUPPORT_INFO.md

travisboss, 2025-03-15:
@mbentley 

```
2025-03-14T10:53:49.184-0500 I CONTROL  [initandlisten] now exiting
2025-03-14T10:53:49.184-0500 I CONTROL  [initandlisten] shutting down with code:100
2025-03-14T10:53:49.198-0500 I CONTROL  [main] ***** SERVER RESTARTED *****
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] MongoDB starting : pid=901 port=27217 dbpath=../data/db 64-bit host=gryffindor
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] db version v3.6.8
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] allocator: tcmalloc
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] modules: none
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] build environment:
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten]     distarch: x86_64
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten]     target_arch: x86_64
2025-03-14T10:53:49.202-0500 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-03-14T10:53:49.202-0500 E STORAGE  [initandlisten] Failed to set up listener: SocketException: Address already in use
2025-03-14T10:53:49.203-0500 I CONTROL  [initandlisten] now exiting
2025-03-14T10:53:49.203-0500 I CONTROL  [initandlisten] shutting down with code:48
2025-03-14T10:54:05.228-0500 I CONTROL  [main] ***** SERVER RESTARTED *****
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] MongoDB starting : pid=930 port=27017 dbpath=../data/db 64-bit host=gryffindor
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] db version v3.6.8
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] allocator: tcmalloc
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] modules: none
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] build environment:
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten]     distarch: x86_64
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten]     target_arch: x86_64
2025-03-14T10:54:05.232-0500 I CONTROL  [initandlisten] options: { repair: true, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-03-14T10:54:05.232-0500 I -        [initandlisten] Detected data files in ../data/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
2025-03-14T10:54:05.232-0500 I STORAGE  [initandlisten] Detected WT journal files.  Running recovery from last checkpoint.
2025-03-14T10:54:05.232-0500 I STORAGE  [initandlisten] journal to nojournal transition config: create,cache_size=15512M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2025-03-14T10:54:05.719-0500 E STORAGE  [initandlisten] WiredTiger error (11) [1741967645:719327][930:0x7f1060cc6ac0], wiredtiger_open: ../data/db/WiredTiger.lock: handle-lock: fcntl: Resource temporarily unavailable
2025-03-14T10:54:05.719-0500 E STORAGE  [initandlisten] WiredTiger error (16) [1741967645:719392][930:0x7f1060cc6ac0], wiredtiger_open: WiredTiger database is already being managed by another process: Device or resource busy
2025-03-14T10:54:05.719-0500 E -        [initandlisten] Assertion: 28718:16: Device or resource busy src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 397
2025-03-14T10:54:05.719-0500 I STORAGE  [initandlisten] exception in initAndListen: Location28718: 16: Device or resource busy, terminating
2025-03-14T10:54:05.719-0500 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
2025-03-14T10:54:05.719-0500 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27017.sock
2025-03-14T10:54:05.719-0500 I CONTROL  [initandlisten] now exiting
2025-03-14T10:54:05.719-0500 I CONTROL  [initandlisten] shutting down with code:100
2025-03-14T10:54:05.732-0500 I CONTROL  [main] ***** SERVER RESTARTED *****
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] MongoDB starting : pid=934 port=27217 dbpath=../data/db 64-bit host=gryffindor
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] db version v3.6.8
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] allocator: tcmalloc
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] modules: none
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten] build environment:
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten]     distarch: x86_64
2025-03-14T10:54:05.736-0500 I CONTROL  [initandlisten]     target_arch: x86_64
[... cut ...]
2025-03-15T02:30:01.246-0500 I STORAGE  [conn20] Finishing collection drop for omada_data.dpiapplicationtrafficstat8daily_20250305 (c31565e1-21ae-4d02-ae18-825a8919037b).
2025-03-15T02:30:01.255-0500 I COMMAND  [conn20] CMD: drop omada_data.dpiapplicationtrafficstat8daily_20250306
2025-03-15T02:30:01.255-0500 I STORAGE  [conn20] Finishing collection drop for omada_data.dpiapplicationtrafficstat8daily_20250306 (bba5c28f-f818-419e-a973-8ddbe9b9ebcb).
```

It is working fine after a reboot, I am not sure if anything in there will help but if it does not and it fails again I will make sure to run that command.

jschanz, 2025-03-18:
I can start the container if I remove the environment variable:

WORKAROUND_509=true

Maybe this helps

watchingdogs, 2025-03-22:
> [@watchingdogs](https://github.com/watchingdogs) - remove `WORKAROUND_509=true` from your env vars. That was for a very specific version hitting a very specific problem.

Removing this solved the issue, and now everything is working for me as well with the newest version. Thank you so much.

mbentley, 2025-03-23:
I [added a version check](https://github.com/mbentley/docker-omada-controller/pull/556) so `WORKAROUND_509` no longer does anything if it is set to true and isn't on a specific version.

travisboss, 2025-05-15:
```
1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s6", background: true }
2025-05-15T01:00:00.073-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.074-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s5 with generated UUID: 6c2f6158-f072-4ea4-9ee3-6ab5f9ba5012
2025-05-15T01:00:00.081-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s5 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s5", background: true }
2025-05-15T01:00:00.081-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.082-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s8 with generated UUID: 939a07a1-898f-4ba6-8630-884e3a942955
2025-05-15T01:00:00.091-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s8 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s8", background: true }
2025-05-15T01:00:00.091-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.092-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s7 with generated UUID: b17b35a1-c1fa-444b-8fa6-49b11aa9c935
2025-05-15T01:00:00.101-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s7 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s7", background: true }
2025-05-15T01:00:00.101-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.101-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s9 with generated UUID: 58255f85-1159-4831-af2f-52942280a590
2025-05-15T01:00:00.110-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s9 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s9", background: true }
2025-05-15T01:00:00.110-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.111-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s2 with generated UUID: 45edaf1c-f772-4cbd-b9eb-28d67bd0887f
2025-05-15T01:00:00.118-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s2 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s2", background: true }
2025-05-15T01:00:00.118-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.119-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s1 with generated UUID: 46d72f5a-91f5-41a2-8220-49eddbdce15f
2025-05-15T01:00:00.127-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s1 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s1", background: true }
2025-05-15T01:00:00.127-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.128-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s16 with generated UUID: 4511f950-60e2-4ec5-a68c-b339f1fb6580
2025-05-15T01:00:00.136-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s16 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s16", background: true }
2025-05-15T01:00:00.136-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.137-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s15 with generated UUID: aab9cef0-8930-42ca-83c4-3b8d98088631
2025-05-15T01:00:00.145-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s15 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s15", background: true }
2025-05-15T01:00:00.145-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.145-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s14 with generated UUID: 74e51e71-e077-4b16-a5de-20132d9f8f55
2025-05-15T01:00:00.155-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s14 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s14", background: true }
2025-05-15T01:00:00.155-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.156-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s13 with generated UUID: 92ab8644-b5c9-4c1b-8cfb-f31e4ac887c8
2025-05-15T01:00:00.166-0500 I INDEX    [conn16] build index on: omada_data.upgradelog_2025w21_t16s13 properties: { v: 2, key: { omadac_id: 1, ctime: -1 }, name: "omadac_id_1_ctime_-1", ns: "omada_data.upgradelog_2025w21_t16s13", background: true }
2025-05-15T01:00:00.166-0500 I INDEX    [conn16] build index done.  scanned 0 total records. 0 secs
2025-05-15T01:00:00.166-0500 I STORAGE  [conn16] createCollection: omada_data.upgradelog_2025w21_t16s12 with generated UUID: bbd0a1bf-cfff-4dc6-807d-083e58d3e6e3
[... cut ...]
2025-05-15T16:07:33.195-0500 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-05-15T16:07:33.195-0500 E STORAGE  [initandlisten] Failed to set up listener: SocketException: Address already in use
2025-05-15T16:07:33.195-0500 I CONTROL  [initandlisten] now exiting
2025-05-15T16:07:33.195-0500 I CONTROL  [initandlisten] shutting down with code:48
```

Have this same issue again, the container just suddenly seems to stop and only a restart fixes this, I am on 5.15 that is not chromium.

travisboss, 2025-05-15:
I do get an error if I go back to 5.14 due to the version difference, would I be safe to try and go back to 5.14?

`The version from the image (5.14.32.4) is older than the last version executed (5.15.20.20)!`

mbentley, 2025-05-16:
No, you can never go backwards - it will 100% brick your controller.

travisboss, 2025-05-16:
@mbentley so I restarted and the biggest difference is I see ipv4 now not just ipv6 listeneing.

```
COMMAND   PID   USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
java    12542 travis  514u  IPv6 106647      0t0  TCP localhost:38294->localhost:27217 (ESTABLISHED)
java    12542 travis  515u  IPv6 105620      0t0  TCP localhost:38296->localhost:27217 (ESTABLISHED)
java    12542 travis  516u  IPv6  99995      0t0  TCP localhost:48104->localhost:27217 (ESTABLISHED)
java    12542 travis  554u  IPv6 102123      0t0  TCP localhost:32886->localhost:27217 (ESTABLISHED)
java    12542 travis 1036u  IPv6 100113      0t0  TCP localhost:35678->localhost:27217 (ESTABLISHED)
mongod  12792 travis   11u  IPv4 105550      0t0  TCP localhost:27217 (LISTEN)
mongod  12792 travis  610u  IPv4 104493      0t0  TCP localhost:27217->localhost:38294 (ESTABLISHED)
mongod  12792 travis  611u  IPv4 104494      0t0  TCP localhost:27217->localhost:38296 (ESTABLISHED)
mongod  12792 travis  612u  IPv4 104533      0t0  TCP localhost:27217->localhost:48104 (ESTABLISHED)
mongod  12792 travis  614u  IPv4 104556      0t0  TCP localhost:27217->localhost:32886 (ESTABLISHED)
mongod  12792 travis  658u  IPv4 110394      0t0  TCP localhost:27217->localhost:35678 (ESTABLISHED)
```

maqyg, 2025-07-07:
I try to run Omada Controller using docker-compose on a Synology, and I see the same issue.
There is already something running on the Synology on port 27217, so MongoDB fails to start.

Is it possible to change the port MongoDB is using within Omada Controller ?

mbentley, 2025-07-07:
The container may schedule but check the logs - there isn't anything handling `EAP_MONGOD_PORT` at the moment so it's probably not changing the port unless there is some other logic in the controller to change the value of the port if the URI is updated.

maqyg, 2025-07-07:
Ok, strange! At least it started up without MongoDB failing due to the "Address already in use" error. And it detected my devices. I'll check it out over the next days.

mbentley, 2025-07-07:
I just started up a test container and it looks like it is still listening on 27217:

```
$ docker top omada-controller
UID                 PID                 PPID                C                   STIME               TTY                 TIME                CMD
508                 434                 412                 99                  15:41               pts/0               00:01:17            java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true --add-opens java.base/sun.security.x509=ALL-UNNAMED --add-opens java.base/sun.security.util=ALL-UNNAMED -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain
508                 635                 434                 0                   15:41               pts/0               00:00:00            tail -F -n 0 /opt/tplink/EAPController/logs/server.log
508                 681                 434                 0                   15:41               pts/0               00:00:00            sh -c "/opt/tplink/EAPController/bin/mongod" --port 27217 --dbpath "../data/db" -pidfilepath "../data/mongo.pid" --logappend --logpath "../logs/mongod.log" --bind_ip 127.0.0.1
508                 682                 681                 2                   15:41               pts/0               00:00:00            /opt/tplink/EAPController/bin/mongod --port 27217 --dbpath ../data/db -pidfilepath ../data/mongo.pid --logappend --logpath ../logs/mongod.log --bind_ip 127.0.0.1

$ docker exec -it omada-controller netstat -ltnpu
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.11:37469        0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.1:27217         0.0.0.0:*               LISTEN      -
tcp6       0      0 :::2376                 :::*                    LISTEN      -
udp        0      0 127.0.0.11:53802        0.0.0.0:*                           -
```

Oddly enough though despite it throwing errors about MongoDB not being accessible, it does seem like PARTS of the controller does work - it's storing data but there are some odd errors showing.  It's like some parts of the controller use the URI parameter, others just use the port

kingp0dd, 2025-08-08:
fixed it by renaming omada/data/db/WiredTiger.lock to omada/data/db/WiredTiger.lock.bak

mbentley, 2025-08-09:
One thing to try would be to stop the controller container and then do a `lsof WiredTiger.wt` or `lsof mongod.lock` to see if there is some other process out there that is still somehow holding the files open.  Here's what I get:

```bash
# pwd
/zfs/storage/omada-controller/data/db
# lsof WiredTiger.wt mongod.lock
COMMAND    PID  USER   FD   TYPE DEVICE SIZE/OFF   NODE NAME
mongod  692823 omada    7uW  REG   0,46        4 117946 mongod.lock
mongod  692823 omada   14u   REG   0,46 12197888 123390 WiredTiger.wt
```