mbentley/docker-omada-controller #553: [Feature]: New Cluster Mode - Support?

mbentley, 2025-03-20:
### What problem are you looking to solve?

There is a new feature coming in one of the betas - [cluster mode](https://www.omadanetworks.com/us/support/faq/4347/). This was found in the `install.sh` of the latest beta (5.15.20.16)

### Describe the solution that you have in mind

Should this be supported?  What additional flags would be required?

### Additional Context

```diff
diff --git a/install-ga.sh b/install-beta.sh
index d258422..449b2ab 100755
--- a/install-ga.sh
+++ b/install-beta.sh
@@ -8,11 +8,21 @@ INSTALLDIR=${DEST_DIR}/${DEST_FOLDER}
 DATA_DIR="${INSTALLDIR}/data"
 LINK=/etc/init.d/tpeap
 LINK_CMD=/usr/bin/tpeap
+CLUSTER_LINK_CMD=/usr/bin/omadacluster
 
+CLUSTER_DIR="${DATA_DIR}/cluster"
 BACKUP_DIR=${INSTALLDIR}/../omada_db_backup
 DB_FILE_NAME=omada.db.tar.gz
 INSTALL_PARAM=$1
 CONSULT_IMPORT_DB=1
+INIT_CLUSTER=0
+
+help() {
+  echo -e "Usage: bash $0 [argument]
+Script argument:
+  help                                 Show this screen.
+  init-cluster-mode                    After installing the controller, it does not start and prompts to start in cluster mode"
+}
 
 # user confirm
 user_confirm() {
@@ -53,7 +63,7 @@ need_import_mongo_db() {
     done
 }
 
- data_is_empty() {
+data_is_empty() {
     # DIR db is existed and not empty.
     if [ -d ${DATA_DIR}/db -a `ls -A ${DATA_DIR}/db | wc -w` -gt 0 ]; then
        echo "data dir is not empty."
@@ -62,8 +72,6 @@ need_import_mongo_db() {
     return 1
 }
 
-
-
 # user confirm
 import_mongo_db() {
     data_is_empty
@@ -88,7 +96,7 @@ import_mongo_db() {
 
 # return: 0, compatible; 1, not compatible;
 version_compatible() {
-    to_be_installed_version="Omada Controller v5.15.8.2 for Linux (X64)"
+    to_be_installed_version="Omada Controller v5.15.20.16 for Linux (X64)"
     installed_version=$(${INSTALLDIR}/bin/control.sh version)
     if [[ "$to_be_installed_version" == "Omada Pro Controller"* ]]; then
         if [[ "$installed_version" != "Omada Pro Controller"* ]]; then
@@ -173,9 +181,63 @@ check_root_perms() {
     [ $(id -ru) != 0 ] && { echo "You must be root to install the ${DESC}. Exit." 1>&2; exit 1; }
 }
 
+version_gt() {
+  test "$(echo "$@" | tr " " "\n" | sort -V | head -n 1)" != "$1"
+}
+version_lt() {
+  test "$(echo "$@" | tr " " "\n" | sort -rV | head -n 1)" != "$1"
+}
+
+java_version_check() {
+  if ! type java >/dev/null 2>&1; then
+    echo "JRE is not installed. please install the dependency."
+    exit 1
+  fi
+  # jre 17
+  JAVA_VERSION=$(java -version 2>&1 |awk 'NR==1{gsub(/"/,"");print $3}')
+  if version_lt "${JAVA_VERSION}" "17.0.0"; then
+    echo "java17-runtime or java17-runtime-headless dependency not satisfied."
+    exit 1
+  fi
+}
+
+jsvc_check() {
+  if ! type jsvc >/dev/null 2>&1; then
+    echo "JSVC software is not installed, please install the dependency."
+    exit 1
+  fi
+  # jre > 8 and jsvc < 1.1.0 : "Cannot find any VM in Java Home /usr/lib/jvm/default-java/"
+  JAVA_VERSION=$(java -version 2>&1 |awk 'NR==1{gsub(/"/,"");print $3}')
+  JSVC_VERSION=$(jsvc -help | grep 'jsvc (Apache Commons Daemon)' | grep -oP '\d+\.\d+\.\d+')
+  JRE_HOME="$( readlink -f "$( which java )" | sed "s:bin/.*$::" )"
+  (version_gt "${JAVA_VERSION}" "1.9") && (version_lt "${JSVC_VERSION}" "1.1.0") && [ ! -d "${JRE_HOME}/lib/amd64" ] && {
+    echo "JRE ${JAVA_VERSION} is greater than 8 and JSVC ${JSVC_VERSION} is less than 1.1.0"
+    mkdir ${JRE_HOME}/lib/amd64
+    ln -s ${JRE_HOME}/lib/server ${JRE_HOME}/lib/amd64/
+  }
+}
+
+# parameter check
+if [[ $# != 0 ]] ; then
+    if [[ $# != 1 || $1 != "init-cluster-mode" || $1 == "help" ]] ; then
+        help
+        exit
+    fi
+    if [ $1 == "init-cluster-mode" ] ; then
+        INIT_CLUSTER=1
+        echo "Start controller distributed cluster mode."
+    fi
+fi
+
 # root permission check
 check_root_perms
 
+# check java depends of controller
+java_version_check
+
+# check jsvc depends of controller and deal with low version jsvc issues
+jsvc_check
+
 if ! user_confirm ; then
     exit
 fi
@@ -210,16 +272,19 @@ done
 if [ $count -gt 0 ]; then
     link_name=${LINK}${count}
     link_cmd_name=${LINK_CMD}${count}
+    cluster_link_cmd_name=${CLUSTER_LINK_CMD}${count}
 
 else
     link_name=${LINK}
     link_cmd_name=${LINK_CMD}
+    cluster_link_cmd_name=${CLUSTER_LINK_CMD}
 fi
 
 
 
 ln -s ${INSTALLDIR}/bin/control.sh ${link_name}
 ln -s ${INSTALLDIR}/bin/control.sh ${link_cmd_name}
+ln -s ${INSTALLDIR}/bin/cluster_control.sh ${cluster_link_cmd_name}
 ln -sf $(which mongod) ${INSTALLDIR}/bin/mongod
 
 
@@ -232,6 +297,7 @@ if test -x ${link_name}; then
     if [ $result -ne 0 ]; then
         chkconfig --add ${link_name}
         chkconfig --add ${link_cmd_name}
+        chkconfig --add ${cluster_link_cmd_name}
 #       echo "add service with chkconfig"
     fi
 
@@ -239,13 +305,17 @@ if test -x ${link_name}; then
     echo "=========================="
 
     if [ $CONSULT_IMPORT_DB == 1 ]; then
-    import_mongo_db
+        import_mongo_db
     fi
 
-    echo "${DESC} will start up with system boot. You can also control it by [${link_cmd_name}]. "
-
-    ${link_name} start
-
+    if [ $INIT_CLUSTER == 1 ]; then
+        echo "Before enabling the cluster mode, first configure the system environment by referring to the Cluster UG(https://www.omadanetworks.com/support/faq/4347/). Then please edit the properties file that specifies the cluster mode in directory ${CLUSTER_DIR} and use command 'sudo omadacluster -config <properties_path> -node <node_name> init' to initialize the cluster node."
+        echo "Controllers that have not completed cluster initialization will still start in normal mode."
+        ${cluster_link_cmd_name} help
+    else
+        echo "${DESC} will start up with system boot. You can also control it by [${link_cmd_name}]. "
+        ${link_name} start
+    fi
     echo "========================"
     exit
 fi
```

mbentley, 2025-03-20:
Here's the `cluster_control.sh` script:

```bash
#!/bin/bash
#
# Cluster startup script for TP-Link's Omada Controller.
#

NAME="omada"
DESC="Omada Controller"

OMADA_HOME=$(dirname $(dirname $(readlink -f $0)))
DATA_DIR="${OMADA_HOME}/data"
CLUSTER_DIR="${DATA_DIR}/cluster"
DB_DIR="${DATA_DIR}/db"
LOG_DIR="${OMADA_HOME}/logs"
MONGOD_LOG_PATH="${LOG_DIR}/mongod.log"
PROPERTY_DIR="${OMADA_HOME}/properties"
STARTUP_INFO_PATH="${DATA_DIR}/startupInfo"
MAIN_CLASS="com.tplink.smb.omada.starter.OmadaLinuxMain"

OMADA_USER=${OMADA_USER:-root}
OMADA_GROUP=$(id -gn ${OMADA_USER})
MONGODB_ORG_SHELL=mongo

CLUSTER_PROPERTIES_PATH="$2"
NODE_ID="$4"

declare -A cluster_config

help() {
    echo "Usage: $0 [<option> ...] <command>"
    cat <<EOF

Commands:
  help                                                   - this screen
  -config <properties_file> -node <node_name> init       - Initialize a specific cluster node using a specified configuration file

Options:
  -config                                                - cluster properties file
  -node                                                  - node name

EOF
}

# root permission check
check_root_perms() {
    [ $(id -ru) != 0 ] && { echo "You must be root to execute this script. Exit." 1>&2; exit 1; }
}

version_gt() {
  test "$(echo "$@" | tr " " "\n" | sort -V | head -n 1)" != "$1"
}

# check env
check_env() {
    if ! type mongod >/dev/null 2>&1; then
        echo "mongodb-org-server is not installed, please install it."
        exit
    fi
    # Obtain the mongoDB version and verify the mongo shell based on the mongoDB version
    MONGODB_VERSION=$(mongod -version | grep 'db version' | grep -oP '\d+\.\d+\.\d+')
    if version_gt "${MONGODB_VERSION}" "5.0.0"; then
        MONGODB_ORG_SHELL=mongosh
        if ! type ${MONGODB_ORG_SHELL} >/dev/null 2>&1; then
            echo "mongodb-mongosh is not installed, please install it."
            exit
        fi
    else
        if ! type ${MONGODB_ORG_SHELL} >/dev/null 2>&1; then
[... 525 lines cut ...]
}

# root permission check
check_root_perms
if [ $# != 5 ]; then
    help
    exit
elif [[ $1 != "-config" && $3 != "-node" && $5 != "init" ]]; then
    help
    exit
fi
[ ! -r $2 ] && {
    echo "$2 does not exist or is not readable."
    exit
}
CLUSTER_PROPERTIES_PATH=$(readlink -f $2)
chmod 666 ${CLUSTER_PROPERTIES_PATH}

# env check
check_env
# init cluster
init_cluster
```

It does have a mention of MongoDB 5.0 or greater so that will be something that will be new to support.

mbentley, 2026-08-27:
Sorry, I've yet to look into it and what all it entails for me to support it.  I'll add this to my priority list since it seems there are a few folks who would like it.

Sriracha02, 2026-08-28:
> Sorry, I've yet to look into it and what all it entails for me to support it. I'll add this to my priority list since it seems there are a few folks who would like it.

no need to apologise. just thought I'd ask since issue was open quite awhile ago.

Thanks anyways!