mbentley/docker-omada-controller #565: [Bug]: Unable to bind to cloud

ilexpl01, 2025-04-02:
### Controller Version

5.15.20.18

### Describe the Bug

Running docker version.

Below my compose file:

```
version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:latest
    restart: unless-stopped
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - PORT_TRANSFER_V2=29815
      - PORT_RTTY=29816
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false

    volumes:
      - /volume1/docker/omada-controller/omada-data:/opt/tplink/EAPController/data:rw
      - /volume1/docker/omada-controller/omada-logs:/opt/tplink/EAPController/logs:rw
      - /volume1/docker/localtime:/etc/localtime:ro
      - /volume1/docker/timezone:/etc/timezone:ro

``` 

I get the following error:

[image]

I also see the multiple errors in my log:

04-02-2025 20:39:12.846 INFO [client-inform-work-group-2] [] c.t.s.o.c.p.c.k(): com.tplink.smb.omada.client.port.cache.k$b: Lan network does not exist, OmadacId: cc966a046a1aedcd418248d7afa1655a, SiteId: 66158081904515713df785cb, vid:0

04-02-2025 20:39:15.592 INFO [Thread-9] [] c.t.e.c.c.q(): set connection status: DISCONNECTED_NORMAL

04-02-2025 20:39:42.959 INFO [client-inform-work-group-0] [] c.t.s.o.c.p.c.k(): com.tplink.smb.omada.client.port.cache.k$b: Lan network does not exist, OmadacId: cc966a046a1aedcd418248d7afa1655a, SiteId: 66158081904515713df785cb, vid:0

Not sure if related to my problem.

### Expected Behavior

connect to TP-Link cloud service

### Steps to Reproduce

....

### How You're Launching the Container

```plain
....
```

### Container Logs

```plain
...
```

### MongoDB Logs

```plain
...
```

mbentley, 2025-04-02:
My controller currently can not bind to the cloud controller myself.  When I try to do so, I get that exact same UI error but my error logs have a very specific error that shows up.  Can you tail the logs to see what shows up right when you hit the `Log In and Bind` button?

ilexpl01, 2025-04-02:
Could be this below:

2025-04-02T21:17:20.858720585Z 	at com.tplink.smb.omada.apigateway.dispatch.ApiController.apiHandler(SourceFile:162) ~[api-gateway-core-5.15.20.18.jar:5.15.20.18]
2025-04-02T21:17:20.858773502Z 	at jdk.internal.reflect.GeneratedMethodAccessor988.invoke(Unknown Source) ~[?:?]
2025-04-02T21:17:20.858837972Z 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
2025-04-02T21:17:20.858889970Z 	at java.base/java.lang.reflect.Method.invoke(Method.java:569) ~[?:?]
2025-04-02T21:17:20.858939216Z 	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:262) ~[spring-web-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.858994540Z 	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:190) ~[spring-web-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859089267Z 	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:118) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859153940Z 	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:917) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859211265Z 	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:829) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859266299Z 	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859319469Z 	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1089) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859397053Z 	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:979) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859449617Z 	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1014) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859502558Z 	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:914) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859562427Z 	at jakarta.servlet.http.HttpServlet.service(HttpServlet.java:547) ~[jakarta.servlet-api-6.0.0.jar:6.0.0]
2025-04-02T21:17:20.859615468Z 	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:885) ~[spring-webmvc-6.1.2.jar:6.1.2]
2025-04-02T21:17:20.859666272Z 	at jakarta.servlet.http.HttpServlet.service(HttpServlet.java:614) ~[jakarta.servlet-api-6.0.0.jar:6.0.0]
2025-04-02T21:17:20.859715781Z 	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:205) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.859771344Z 	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:149) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.859826108Z 	at org.apache.tomcat.websocket.server.WsFilter.doFilter(WsFilter.java:51) ~[tomcat-embed-websocket-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.859879305Z 	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:174) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.859930955Z 	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:149) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.859980440Z 	at org.apache.shiro.web.servlet.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:141) ~[shiro-web-1.13.0-jakarta.jar:1.13.0]
2025-04-02T21:17:20.860034332Z 	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:174) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.860085972Z 	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:149) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.860140377Z 	at com.tplink.smb.omada.apigateway.dispatch.filter.HttpTraceFilter.doFilter(SourceFile:40) ~[api-gateway-core-5.15.20.18.jar:5.15.20.18]
2025-04-02T21:17:20.860196100Z 	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:174) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.860249765Z 	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:149) ~[tomcat-embed-core-10.1.17.jar:10.1.17]
2025-04-02T21:17:20.860299327Z 	at org.apache.shiro.web.servlet.ProxiedFilterChain.doFilter(ProxiedFilterChain.java:61) ~[shiro-web-1.13.0-jakarta.jar:1.13.0]
2025-04-02T21:17:20.860360761Z 	at org.apache.shiro.web.servlet.AdviceFilter.executeChain(AdviceFilter.java:108) ~[shiro-web-1.13.0-jakarta.jar:1.13.0]
2025-04-02T21:17:20.860413788Z 	at org.apache.shiro.web.servlet.AdviceFilter.doFilterInternal(AdviceFilter.java:137) ~[shiro-web-1.13.0-jakarta.jar:1.13.0]
2025-04-02T21:17:20.860464916Z 	at org.apache.shiro.web.servlet.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:154) ~[shiro-web-1.13.0-jakarta.jar:1.13.0]
2025-04-02T21:17:20.860516976Z 	at org.apache.shiro.web.servlet.ProxiedFilterChain.doFilter(ProxiedFilterChain.java:66) ~[shiro-web-1.13.0-jakarta.jar:1.13.0]
[... cut ...]
2025-04-02T21:17:20.969305320Z 04-02-2025 21:17:20.968 WARN [Thread-9] [] c.t.e.c.c.e(): Request params from delegate is in invalid format.
2025-04-02T21:17:21.007011830Z 04-02-2025 21:17:21.006 INFO [Thread-47] [] c.t.e.c.c.q(): 'recvThread' exception:java.net.SocketException: There is no more data because the end of the stream has been reached.
2025-04-02T21:17:21.007208913Z 04-02-2025 21:17:21.006 INFO [Thread-47] [] c.t.e.c.c.q(): Thread 'recvThread' is stopped
2025-04-02T21:17:25.978252608Z 04-02-2025 21:17:25.977 INFO [Thread-9] [] c.t.e.c.c.q(): set connection status: DISCONNECTED_NORMAL

mbentley, 2025-04-02:
Slightly different than what I see but it's like it's not getting the response as expected from the cloud service - nothing specific to the containerization of the controller so it would be something that would need to be addressed by the TP-Link people.  Easiest is to [create a post on the TP-Link community forums](https://community.tp-link.com/en/business/forum/582).

closed 2025-04-02 by mbentley